hbldh/bleak #698: BleakScannerWinRT import fails in Jupyter Notebook - [WinError -2147417850] Cannot change thread mode after it is set

datoslabs, 2021-12-09:
* bleak version: 0.13.0
* Python version: 3.8.8
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: N/A

### Description
Hi,

New to bleak.  I encountered `[WinError -2147417850] Cannot change thread mode after it is set` when I attempted to import the scanner with the following:

```
from bleak.backends.winrt.scanner import (
        BleakScannerWinRT as BleakScanner,
    ) 
```

After some googling, it appears to be a known issue trying to initialize `winrt` with MTA in Jupyter Notebook.  I was able to avoid the error by:

1.  Install winrt
2.  Execute the following twice (the first execution resulted in the same error but the second time will run without errors) :
```
import importlib
_winrt=importlib.import_module("winrt._winrt")
_winrt.uninit_apartment()
import winrt
```
3.  Afterwards, running the following to import BleakScannerWinRT works without any error:
```
from bleak.backends.winrt.scanner import (
        BleakScannerWinRT as BleakScanner,
    ) 
```
Is this the right approach?  I seemed to be able to scan and discover my BLE device; I am still getting familiar with this library to read the characteristics so I am not sure if I would run into any problems down the road.

dlech, 2021-12-13:
Hmm... now that we have our own `bleak-winrt` package, we should probably set the default to be STA instead of MTA to avoid the need for such workarounds.

hbldh, 2021-12-13:
I ran into this trying to debug from VS Code. It is a shame if such a basic and not at all uncommon operation produces an error. I am way too uninitiated in why it is needed to say what the default should be, but in general I think Bleak should aim for making things as non-problematic for the non-experts as possible.

commit 89adf8ce (2022-01-10): Version 0.14.0

Added
-----

* Added ``service_uuids`` kwarg to  ``BleakScanner``. This can be used to work around issue of scanning not working on macOS 12. Fixes #230. Works around #635.
* Added UUIDs for LEGO Powered Up Smart Hubs.

Changed
-------

* Changed WinRT backend to use GATT session status instead of actual device connection status.
* Changed handling of scan response data on WinRT backend. Advertising data and scan response data is now combined in callbacks like other platforms.
* Updated ``bleak-winrt`` dependency to v1.1.0. Fixes #698.

Fixed
-----

* Fixed ``InvalidStateError` [... cut]
PR #727 (2022-01-10): Release/0.14.0

Added
-----

* Added ``service_uuids`` kwarg to  ``BleakScanner``. This can be used to work around issue of scanning not working on macOS 12. Fixes #230. Works around #635.
* Added UUIDs for LEGO Powered Up Smart Hubs.

Changed
-------

* Changed WinRT backend to use GATT session status instead of actual device connection status.
* Changed handling of scan response data on WinRT backend. Advertising data and scan response data is now combined in callbacks like other platforms.
* Updated ``bleak-winrt`` dependency to v1.1.0. Fixes #698.

Fixed
-----

* Fixed ``InvalidStateError` [... cut]
closed 2021-12-31 by dlech