hbldh/bleak #701: Unable to connect to device on Ubuntu

xsaahil, 2021-12-28:
* bleak version: 0.11.0
* Python version: 3.8.10
* Operating System: Ubuntu 20.04.3
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

I am trying to connect to a BLE Server to do some GATT operations such as reads and writes. When I am trying to connect to the device I am always getting this error:

`bleak.exc.BleakDBusError: org.bluez.Error.NotAvailable`

The laptop connects to the device and disconnects immediately and returns error.

### What I Did

I tried connecting directly with `bluetoothctl connect XX:XX:XX:XX:XX` and this command is able to connect to the device and the connection is stable until I disconnect.

Currently, this things happen:

- When I run the script, the laptops BLE connects to the device
- After like 500 to 600 ms the laptop, this error shows up.
```
File "/home/user/venv/lib/python3.8/site-packages/bleak/backends/bluezdbus/utils.py", line 24, in assert_reply raise BleakDBusError(reply.error_name)
bleak.exc.BleakDBusError: org.bluez.Error.NotAvailable
```

- This is the how the object looks like that hits the exception and disconnects:
``` 
{'destination': ':1.368', 'path': None, 'interface': None, 'member': None, 'message_type': <MessageType.ERROR: 3>, 'flags': <MessageFlag.NO_REPLY_EXPECTED: 1>, 'error_name': 'org.bluez.Error.NotAvailable', 'reply_serial': 6, 'sender': ':1.303', 'unix_fds': [], 'signature': 's', 'signature_tree': <dbus_next.signature.SignatureTree object at 0x7fc57f7e9df0>, 'body': ['Operation currently not available'], 'serial': 3227}
 ```

I have tried reproducing this on a different laptop and the same thing hapens, I have also tried changing the bluez version from **5.53** to **5.45** and it still results the same.

dlech, 2021-12-28:
> * bleak version: 0.11.0

Please try with Bleak 0.13.0. This should include more information in the error message.

xsaahil, 2021-12-29:
I have updated the version to 0.13.0 and this the error message:

```{'destination': ':1.449', 'path': None, 'interface': None, 'member': None, 'message_type': <MessageType.ERROR: 3>, 'flags': <MessageFlag.NO_REPLY_EXPECTED: 1>, 'error_name': 'org.bluez.Error.NotAvailable', 'reply_serial': 6, 'sender': ':1.441', 'unix_fds': [], 'signature': 's', 'signature_tree': <dbus_next.signature.SignatureTree object at 0x7fde31e9ef40>, 'body': ['Operation currently not available'], 'serial': 326}```

I have tried using pygatt and it works and keeps the connection, but the connection does not stay with bleak.

dlech, 2021-12-29:
Looking at the BlueZ source code, it seems like [this error](https://github.com/bluez/bluez/search?q=btd_error_not_available) comes from is an audio profile. Is this a dual-mode device (low energy + classic)?

There [isn't a BlueZ API to only connect to low energy](https://github.com/bluez/bluez/blob/9be85f867856195e16c9b94b605f65f6389eda33/doc/device-api.txt#L24-L36), so I'm afraid we probably can't fix this in Bleak.

You could try opening an issue in the [BlueZ](https://github.com/bluez/bluez) repository to see if they can offer more insight. Including logs from `btmon` there would probably be helpful.