hbldh/bleak #707: OSError: [WinError -2147418113] when connecting to a same device for the second time

xixiaxixi, 2021-12-30:
* bleak version: ```0.13.0```
* Python version: `3.9.7`
* Operating System: `Windows 10 20H1`
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

When connecting to a same device  for the second time in a short time(10 min or more), error occured.

### What I Did

1. Connected to a device, transfered some data(or do nothing), then disconnected (python exited)
2. Connected to the same device again, got error below:

```
Traceback (most recent call last):
  File "E:\bmc1025\bmc\learnlua\bletest3.py", line 107, in <module>
    sys.exit(asyncio.run(main()))
  File "D:\Programmes\anaconda\envs\bleenv\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "D:\Programmes\anaconda\envs\bleenv\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "E:\bmc1025\bmc\learnlua\bletest3.py", line 67, in main
    raise e
  File "E:\bmc1025\bmc\learnlua\bletest3.py", line 64, in main
    await client.connect()
  File "D:\Programmes\anaconda\envs\bleenv\lib\site-packages\bleak\backends\winrt\client.py", line 257, in connect
    await self.get_services()
  File "D:\Programmes\anaconda\envs\bleenv\lib\site-packages\bleak\backends\winrt\client.py", line 426, in get_services
    await self._requester.get_gatt_services_async(
OSError: [WinError -2147418113] 灾难性故障
```

灾难性故障: maybe 'Fatal error' in English

### More

Restarting OS, or waiting for a long time solves. Maybe windows does some clean up.

xixiaxixi, 2021-12-30:
After debugging for hours, I found if I click "Step over" slowly, the code is more likely to run as expected.

After trying `sleep` in many places, I found the key.

...\backends\winrt\client.py, about line 258
```
...
await asyncio.sleep(1) # +
# Obtain services, which also leads to connection being established.
await self.get_services()
```

dlech, 2021-12-31:
Interesting. It would be nice if you could do a [Bluetooth packet capture](https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10) for both the working (sleep workaround) and non-working case so we can see what the difference is.

There is also #695 which adds the possibility to use cached services on the second connection which may help work around the issue.

Lukwrt, 2022-03-30:
I've had the same issue on one computer. And despite using this modification of the client.py script the OS error still remains : 

> ...\backends\winrt\client.py, about line 258
> 
> ```
> ...
> await asyncio.sleep(1) # +
> # Obtain services, which also leads to connection being established.
> await self.get_services()
> ```

Do you have any ideas on how to fix this ? 

As mentioned by OP when I restart the computer it works fine sometimes. The issue is probably related to the winrt backend (I have the same logs as OP except in French for Fatal Error / Catastrophic Failure : Défaillance Irrémédiable), is https://github.com/hbldh/bleak/pull/695 this feature going to be integrated in bleak package as the issues seem connected.

closed 2025-05-16 by dlech