hbldh/bleak #712: Enhancement: Subclass BleakException to better handle expected (not found) vs. failure conditions

jeffsf, 2022-01-01:
* bleak version: 0.13.0
* Linux `bluetoothctl: 5.55`

Application has either through configuration or previous discovery the MAC address of the target device. The device may either be online or not. Failure to connect raises a `BleakError` along with many other conditions. BleakError is not subclassed, past BleakDBusError that I could find. The arguably ugly hack of matching the exception's args seems required to determine if it is an "expected" condition, or something else has gone wrong.  

```
                    if e.args[0].startswith(
                                'Device with address') \
                            or e.args[0].startswith(
                                'Connection was not successful!'):
```

It would be preferred if this could be done without relying on the message as an "unsupported API" detail.

dlech, 2022-01-01:
Duplicate of #527

dlech, 2022-01-01:
If you have specific conditions that you handle, please list them in #527 so we know what to prioritize.

closed 2022-01-01 by dlech