hbldh/bleak #714: OSError: [WinError -2147023836] The specified service does not exist as an installed service

vuslat-ozel, 2022-01-03:
* bleak version: 0.13.0
* Python version: 3.9.6 - pip 21.3.1
* Operating System: Windows 10 Pro 19043.1415
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
scanner doesn't works, sample discovery also doesn't work. application.py is:
```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())
```

I was developing an application using bleak, it was fine. But we need to stop working on this project due to an emergency on another project for a month. I'm not sure if this issue caused by a windows update or sth else but now it stopped working. As a workaround, I'm testing with the sample codes from Bleak and still crashes with the same error which I posted below.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Traceback (most recent call last):
  File "c:\Users\vozel\Desktop\Development\Des.BaseStation\application.py", line 9, in <module>
    asyncio.run(main())
  File "C:\Python39\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Python39\lib\asyncio\base_events.py", line 642, in run_until_complete
    return future.result()
  File "c:\Users\vozel\Desktop\Development\Des.BaseStation\application.py", line 5, in main
    devices = await BleakScanner.discover()
  File "C:\Python39\lib\site-packages\bleak\backends\scanner.py", line 99, in discover
    async with cls(**kwargs) as scanner:
  File "C:\Python39\lib\site-packages\bleak\backends\scanner.py", line 79, in __aenter__
    await self.start()
  File "C:\Python39\lib\site-packages\bleak\backends\winrt\scanner.py", line 142, in start
    self.watcher.start()
OSError: [WinError -2147023836] The specified service does not exist as an installed service
```

dlech, 2022-01-03:
This error comes from Windows. So I would try logging Bluetooth packets as described in our troubleshooting guide to try to gain more insight.

dlech, 2022-01-03:
Or could it be that Bluetooth is just turned off?

vuslat-ozel, 2022-01-04:
I've tried every troubleshooting steps but still at the same page. Even there are no additional logs on our application.

dlech, 2022-01-04:
Reboot the computer? Try a different computer?

JamesYangLim, 2022-09-01:
Students tried it and it works on Windows 10 but not in Windows 11.
```
Traceback (most recent call last):
  File "c:\Users\james\source\repos\bluetooth\run.py", line 26, in <module>
    x.run_quaternions(sensors)
  File "c:\Users\james\source\repos\bluetooth\dot.py", line 232, in run_quaternions
    loop.run_until_complete(tasks)
  File "C:\Python310\lib\asyncio\base_events.py", line 646, in run_until_complete
    return future.result()
  File "c:\Users\james\source\repos\bluetooth\dot.py", line 201, in async_run
    async with IMU(dot) as d:
  File "c:\Users\james\source\repos\bluetooth\dot.py", line 82, in __aenter__
    await self.client.__aenter__()
  File "C:\Python310\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 314, in connect
    await self.get_services()
  File "C:\Python310\lib\site-packages\bleak\backends\winrt\client.py", line 521, in get_services
    await service.get_characteristics_async(*args),
OSError: [WinError -2147483634] A method was called at an unexpected time
```

dlech, 2022-09-01:
@JamesYangLim This is a different error. Please start a new discussion and provide a minimal example code to reproduce the problem.

closed 2025-11-30 by github-actions[bot]