hbldh/bleak #716: example/detection_callback.py doesn't work with all on bluez

mkoderer, 2022-01-03:
* bleak version:
develop branch
* Python version:
3.9.2
* Operating System:
Linux raspberrypi 5.10.89-v7+ #1506 SMP Fri Dec 31 14:38:59 GMT 2021 armv7l GNU/Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:
bluetoothctl: 5.55

### Description

Registering all known uuids leads to an exception:

```
> python examples/detection_callback.py all
2022-01-03 19:38:50,872 __main__ INFO: Adding all known service uuids
Traceback (most recent call last):
  File "/home/pi/git/bleak/examples/detection_callback.py", line 55, in <module>
    asyncio.run(main(service_uuids))
  File "/usr/lib/python3.9/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 642, in run_until_complete
    return future.result()
  File "/home/pi/git/bleak/examples/detection_callback.py", line 44, in main
    await scanner.start()
  File "/home/pi/git/bleak/.venv/lib/python3.9/site-packages/bleak-0.14.0a1-py3.9.egg/bleak/backends/bluezdbus/scanner.py", line 166, in start
    assert_reply(reply)
  File "/home/pi/git/bleak/.venv/lib/python3.9/site-packages/bleak-0.14.0a1-py3.9.egg/bleak/backends/bluezdbus/utils.py", line 23, in assert_reply
    raise BleakDBusError(reply.error_name, reply.body)
bleak.exc.BleakDBusError: [org.bluez.Error.InProgress] Operation already in progress
```

dlech, 2022-01-03:
Interesting. This seems to trigger a bug in BlueZ. After running the program with `all`, then even using a single UUID triggers the error until `bluetoothd` is restarted. I'm guessing that sending it so many UUIDs overwhelms BlueZ somehow.

I'm not too worried about this since `all` is just a hack to work around a macOS 12 bug and should never be used otherwise.

mkoderer, 2022-01-04:
@dlech Should we active the "all" behavior by default for Mac (without any parameter)?

The bluez bug is interesting it starts when you specify more than 64 service_uuids.
I will try to reproduce it  directly and update the version

mkoderer, 2022-01-04:
It's an clear bluez bug/limitation:

```
$ bluetoothctl
[bluetooth]# menu scan
[bluetooth]# uuids 0001 0003 0005 0007 0008 000f 0010 0011 0012 0014 0016 0017 0019 001b 001e 001f 0100 1000 1001 1002 1101 1102 1103 1104 1105 1106 1107 1108 1109 110a 110b 110c 110d 110e 110f 1110 1111 1112 1113 1114 1115 1116 1117 1118 1119 111a 111b 111c 111d 111e 111f 1120 1121 1122 1123 1124 1125 1126 1127 1128 112d 112e 112f 1130 1131
[bluetooth]# scan on
SetDiscoveryFilter success
Failed to start discovery: org.bluez.Error.InProgress
```

dlech, 2022-01-04:
> @dlech Should we active the "all" behavior by default for Mac (without any parameter)?

As long as it prints a message explaining why some devices might not be detected (not in list of known services or no advertised service at all) I think that would be fine.