hbldh/bleak #720: Devices that do not include a service UUID in advertising data cannot be detected on macOS 12

dlech, 2022-01-05:
* bleak version: develop
* Python version: all
* Operating System: macOS 12.x

### Description

This is basically the same issue as #635. That issue was closed because there is now a workaround available that works in most cases. However, there is one case that doesn't have a workaround. If a device does not include a service UUID in the advertising data, then the advertising data from that device cannot be received in Bleak.

### Call to action

This is a regression from previous versions of macOS. Please everyone report this issue to Apple using the Feedback Assistant app. If enough people report it, hopefully they will prioritize fixing it.

hbldh, 2022-01-05:
Very good to keep an issue alive for this.

pauloesteban, 2022-01-18:
Hi everyone,

Anyone tried a similar workaround with PyInstaller?

Cheers

kdewald, 2022-01-24:
@dlech have you have any updates on your research on this topic?

I've been able to reproduce it with a very minimalist console application on XCode (https://github.com/kdewald/MontereyBluetoothError). My theory is that there is some problem between the TCC and the Bluetooth subsystems on Monterey, as looking at the logs shows that certain authorization steps do not happen when running console applications, even though they have Bluetooth authorization.

dlech, 2022-01-25:
I haven't done any further research. Reiterating what I have already said, the `NSBluetoothAlwaysUsageDescription` (I assume this is what you mean when you say "they have Bluetooth authorization") only prevents the app from crashing with `SIGABRT`. It is completely independent of the "no scan results" problem. The "no scan results" problem has only to do with the ability of the process to be considered the "foreground" app which consists of having an icon in the dock and being properly signed.

dlech, 2022-01-30:
Good news! It looks like Apple has fixed the issue in macOS 12.3 Beta (21E5196i).

PR #818 (2022-04-29): Release/0.14.3

Changed
-------

* Suppress macOS 12 scanner bug error message for macOS 12.3 and higher. Fixes #720.
* Added filters ``Discoverable`` and ``Pattern`` to BlueZ D-Bus scanner. Fixes #790.

Fixed
-----

* Fixed reading the battery level returns a zero-filled ``bytearray`` on BlueZ >= 5.48. Fixes #750.
* Fixed unpairing does not work on windows with WinRT. Fixes #699
* Fixed leak of ``_disconnect_futures`` in ``CentralManagerDelegate``.
* Fixed callback not removed from ``_disconnect_callbacks`` on disconnect in ``CentralManagerDelegate``.
closed 2022-03-05 by dlech