hbldh/bleak #729: Cannot scan for connected devices

MagnusModig, 2022-01-11:
* bleak version: 0.13.0
* Python version: 3.9
* Operating System: Win 11
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I try to scan connected devices.

### What I Did

Ran this code...```import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())``

But I get this error..

`Traceback (most recent call last):
  File "D:\scan.py", line 2, in <module>
    from bleak import BleakScanner
  File "D:\bleak.py", line 6, in <module>
    from bleak import BleakClient, BleakScanner
ImportError: cannot import name 'BleakClient' from partially initialized module 'bleak' (most likely due to a circular import) (D:\bleak.py)`

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

dlech, 2022-01-11:
You cannot name your script `bleak.py`. This causes the circular import.

MagnusModig, 2022-01-11:
Thanks, that helped to rename the file

commit 53a48d31 (2022-01-11): docs: warn against bleak.py

We frequently get bug reports about circular import errors. This is
due to naming your script bleak.py. So put warnings everywhere so
hopefully people won't miss it.

Fixes: #729
Fixes: #704
Fixes: #515
PR #730 (2022-01-12): docs: warn against bleak.py
We frequently get bug reports about circular import errors. This is due to naming your script `bleak.py`. So put warnings  everywhere so hopefully people won't miss it.

Fixes: #729
Fixes: #704
Fixes: #515
closed 2022-01-12 by dlech