hbldh/bleak #736: Devices showing up as Unknown

quasi635, 2022-01-13:
* bleak version: 0.14.1
* Python version: 3.7.9
* Operating System: Windows 10

### Description

After upgrading to bleak 0.14.1 from 0.13.0, my devices no longer show up when doing `discover()`

### What I Did

I ran the example code
```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())
```

On 0.13.0 I see a number of devices, including:
```
A0:E6:F8:3A:84:A1: Temperature 835-041>19
E0:7D:EA:6B:1E:5A: //code.Node 020-122>G0
7E:68:78:BB:3F:32: Apple, Inc. (...)
69:BB:ED:8E:0A:97: Microsoft (...)
34:3C:5B:F3:EC:52: Microsoft (...)
75:59:95:91:07:00: Apple, Inc. (...)
7C:BE:1C:6C:9E:C1: Apple, Inc. (...)
A0:9E:1A:34:DB:6D: Polar H7 34DB6D24
```

On 0.14.1 those same devices show up as:
```
A0:E6:F8:3A:84:A1: Unknown
E0:7D:EA:6B:1E:5A: Unknown
7E:68:78:BB:3F:32: Apple, Inc. (...)
69:BB:ED:8E:0A:97: Microsoft (...)
34:3C:5B:F3:EC:52: Microsoft (...)
75:59:95:91:07:00: Apple, Inc. (...)
7C:BE:1C:6C:9E:C1: Apple, Inc. (...)
A0:9E:1A:34:DB:6D: Polar H7 34DB6D24
```

Any help? I can't figure out what changed between the two versions to cause this. Both version show all of the Apple/Microsoft/Polar devices around me (I removed the byte value and replaced with three dots).

dlech, 2022-01-14:
Can you enable logging and share the results?

The change that broke this is probably https://github.com/hbldh/bleak/commit/3f26be60c3e69d61c5327a0fe0bbf0aee20f1456. In particular, the logic that determines the name is at: https://github.com/hbldh/bleak/blob/3f26be60c3e69d61c5327a0fe0bbf0aee20f1456/bleak/backends/winrt/scanner.py#L211-L232

dlech, 2022-01-14:
I could see how this might fail if `args.advertisement.local_name` returns an empty string instead of `None`.

quasi635, 2022-01-14:
> 

Thank you for pointing out the source. Yes, this seems like the likely culprit. I will set my project to use 0.13.0 for now. Hopefully this can be tackled for a future release. I would be happy to make a PR to try and address this fix if your team is busy.

dlech, 2022-01-14:
Yes, we would be happy to review a pull request.

kva85, 2022-01-20:
I had the same issue, digging a bit further when printing the device I get also the following for bleak 0.14. Not sure if I need to log this here

C7:40:B2:32:46:18: This value has special meaning depending on the context in which it used. Link Manager Protocol (LMP): This value may be used in the internal and interoperability tests before a Company ID has been assigned. This value shall not be used in shipping end products. Device ID Profile: This value is reserved as the default vendor ID when no Device ID service record is present in a remote device. 

I tracked it down to the fact that we specified a company UUID in the ble manufacturing data, removing this removed the above warning. With bleak 0.13, all worked fine.

kva85, 2022-01-26:
I print the following:
`
...
        for args in filter(lambda d: d is not None, raw_data):
            print('adv', raw_data.adv.advertisement.local_name)
            print('scan', raw_data.scan.advertisement.local_name)
`

and get

adv SENSOR_0x0000043D
scan

dlech, 2022-01-26:
Thanks for looking into this. Can you please test #748 to see if it fixes the issue?

closed 2022-01-26 by dlech

quasi635, 2022-01-28:
The 0.14.2 fix worked perfectly, thank you @dlech!!!

PR #749 (2022-01-26): v0.14.2

Changed
-------

* Updated ``bleak-winrt`` dependency to v1.1.1. Fixes #741.

Fixed
-----

* Fixed ``name`` is ``'Unknown'`` in WinRT backend. Fixes #736.