hbldh/bleak #739: Windows 10: Device does not support this function - OSError: [WinError -2147024580]

WouterJD, 2022-01-14:
* bleak version: `0.14.1`_ (2022-01-12)
* Python version: Python 3.8.1
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux: not applicable

### Description
I'm trying to get kevinkar/bless running, starting with his example server.py

Purpose: I want to use bless [bleak] to create a fitness-machine-service service to extend [FortiusAnt](https://github.com/WouterJD/FortiusANT) expection that native winrt BLE is better than the current nodejs implementation.

### What I Did
I am running the test on a Thinkpad T430,
- using the builtin bluetooth adaptor
OR
- after disabling the builtin adapter, using an ASUS BT400 BLE dongle
Both give same result.

server.py creates a server, a service and one attribute and then starts the service.

in starting the service, an abend occurs at `service.service_provider.start_advertising(adv_parameters)`
**OSError: [WinError -2147024580] Het apparaat ondersteunt de opdrachtfunctie niet**
"The device does not support the command-function"

### What I Did
What can be the cause for the **OSError: [WinError -2147024580]** error

dlech, 2022-01-14:
> kevinkar/bless

This repository is https://github.com/hbldh/bleak, not https://github.com/kevincar/bless so I'm afraid we can't help with this problem here.

dlech, 2022-01-14:
Although I would suspect that the error message is telling you that the hardware is not capable of being a peripheral.

closed 2022-01-14 by dlech

WouterJD, 2022-01-14:
Hi @dlech I understand; bless uses bleak to access the BLE-devices; service_provider.start_advertising(adv_parameters) 
bless -> bleak -> bleak_winRT

Therefore I thought you might be so kind to assist in understanding the issue.

I'm looking for the reason why my Thinkpad T430 standard bluettoth device (or the ASUS BT400 BLE dongle) produces this error...
What would make the hardware not being capable to be a peripheral?

dlech, 2022-01-14:
This isn't a Bleak API, it is a Windows API. 

https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattserviceprovider.startadvertising?view=winrt-22000

There is a feedback link on that page, so maybe Microsoft could clarify what exactly that error means?

WouterJD, 2022-01-14:
Thanks for the link, very helpful.
Now find the error... Hoping that Google, @kevincar, @dlech, @SomeOneElse can assist🙂