hbldh/bleak #740: GATT Services: Unreachable, when connecting on windows10

pawelm87, 2022-01-16:
* bleak version: 0.12.1
* Python version: 3.8
* Operating System: Windows10
* BlueZ version (`bluetoothctl -v`) in case of Linux: -

### Description

I am trying to establish a connection to a device using one of the examples.
I have a debug of the device, I get a connect then a disconnect message, and a few seconds later I get a reconnect message. The python debug indicates nothing, then I get the error message GATT service: unreachable

### What I Did

The code snippet that is responsible for the connection. The callback function is responsible for sending/reading data by client characteristics 
```
async def _connect(device: BLEDevice, timeout: float, callback):
    async with BleakClient(address_or_ble_device = device, timeout = 20.0) as client:
        result = await callback(client=client)
        print(result)
        return result
```
I also used your example
```
async def main(device: BLEDevice):
    async with BleakClient(device) as client:
        svcs = await client.get_services()
        print("Services:")
        for service in svcs:
            print(service)
```

After a few seconds I get this error "raise BleakDotNetTaskError( bleak.exc.BleakDotNetTaskError: Could not get GATT services: Unreachable"

part of the console error:
```
File "C:\Users\P\Desktop\tool\firmware-loader\src\Ble\bleHelper.py", line 113, in connectByBleDevice
     return await _connect(bleDeviceOrAddr=bleDevice, timeoutConnection=timeoutConnection, callback=callback,
   File "C:\Users\P\Desktop\tool\firmware-loader\src\Ble\bleHelper.py", line 52, in _connect     async with BleakClient(
   File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
     await self.connect()
   File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\client.py", line 227, in connect
     await self.get_services(use_cached=use_cached)   File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\client.py", line 417, in get_services
     raise BleakDotNetTaskError( bleak.exc.BleakDotNetTaskError: Could not get GATT services: Unreachable
```

dlech, 2022-01-16:
> * bleak version: 0.12.1

Does the error still happen with Bleak v0.14.1?

dlech, 2022-01-16:
Did you try logging Bluetooth packets to see what is going on? https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

pawelm87, 2022-01-17:
```
PS C:\Users\P\Desktop\tool-production\firmware-loader> pip.exe show bleak
Name: bleak
Version: 0.14.1
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: c:\users\p\appdata\local\programs\python\python38\lib\site-packages
Requires: bleak-winrt
Required-by:
PS C:\Users\P\Desktop\tool-production\firmware-loader> pip.exe show winrt
Name: winrt
Version: 1.0.21033.1
Summary: Access Windows Runtime APIs from Python
Home-page: https://github.com/Microsoft/xlang/tree/master/src/package/pywinrt/projection
Author: Microsoft Corporation
Author-email: None
License: MIT
Location: c:\users\p\appdata\local\programs\python\python38\lib\site-packages
Requires:
Required-by:

PS C:\Users\P\Desktop\tool-production\firmware-loader> python.exe .\main_v2.py
Traceback (most recent call last):
  File ".\main_v2.py", line 13, in <module>
    from source.CLIToolHandler.bleHandler import downloadUploadConfigFile, LKey
  File "C:\Users\P\Desktop\tool-production\firmware-loader\source\CLIToolHandler\bleHandler.py", line 3, in <module>
    from bleak import BleakClient
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\__init__.py", line 78, in <module>
    from bleak.backends.winrt.scanner import (
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\scanner.py", line 7, in <module>
    from bleak_winrt.windows.devices.bluetooth.advertisement import (
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak_winrt\__init__.py", line 3, in <module>
    from . import _winrt
ImportError: DLL load failed while importing _winrt: The specified module could not be found.
```

I upgraded the bleak library from 0.12.1 to 0.14.1 and while it worked before - it connected - now I get this error immediately after running the script.

On another computer I run the same script with 0.14.1 library and I do not get such error - the library versions are the same.

I did not go to test the GATT services error - I stopped at this error

pawelm87, 2022-01-17:
I checked, on the computer where I have the winrt error the bleak-winrt 1.1.0 package is installed

dlech, 2022-01-17:
I didn't think that bleak-winrt had any dependencies that didn't ship with Windows but maybe it requires MS Visual C++ 2022 redistributable? See https://stackoverflow.com/a/64620736/1976323.

pawelm87, 2022-01-18:
I installed Microsoft Visual C++ Redistributable and this solved my problem with the import error. I will now test the new lib version.

Has anyone so far reported a similar problem - GATT service unreachable, or can the new released versions 0.13.0 and 0.14.1 relate to this problem?

I think the exception flies from here:
https://github.com/hbldh/bleak/blob/develop/bleak/backends/winrt/client.py#L268
https://docs.microsoft.com/en-us/uwp/api/windows.devices.bluetooth.genericattributeprofile.gattsession?view=winrt-22000

dlech, 2022-01-18:
> Did you try logging Bluetooth packets to see what is going on? https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

pawelm87, 2022-01-19:
Not yet, but I plan to

allrobot, 2022-03-25:
> 您是否尝试记录蓝牙数据包以查看发生了什么？https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

Hello, I meet some error:
```
bleak.exc.BleakError: Could not get GATT services: Unreachable
```
[image]

This is bluetooth bags:
[ERROR.txt](https://github.com/hbldh/bleak/files/8348459/ERROR.txt)

This is ESP32 Code:
[BLE UART.txt](https://github.com/hbldh/bleak/files/8348469/BLE.UART.txt)

allrobot, 2022-03-25:
> > 您是否尝试记录蓝牙数据包以查看发生了什么？https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

I'm going to try to read the TX UUID, but it doesn't work.

mjhoefer, 2022-04-19:
> Not yet, but I plan to

@pawelm87 did you end up finding a solution to this? Thank you in advance.

pawelm87, 2022-04-19:
I have not yet found a solution. 

It occurs randomly in my opinion. I'm testing it on different computers and on one I don't see the error, on another the error occurs once every hundred runs and on yet another it occurs frequently but also randomly.

One thing I have noticed is that resetting the computer (windows 10) clears the error for a while. 

I am trying to catch bluetooth frames but due to randomness it is quite difficult :(

dlech, 2022-07-25:
We have just merged #695 which could make a difference with this issue. Can everyone experiencing this problem please test? If there are still problems, Bluetooth packet captures would be helpful.

hacihasansavan, 2022-10-13:
Hi, I'm getting same error with a ble device. I guess there is no solution for now right?