hbldh/bleak #741: Need to statically link bleak-winrt.

dlech, 2022-01-18:
```
PS C:\Users\P\Desktop\tool-production\firmware-loader> pip.exe show bleak
Name: bleak
Version: 0.14.1
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: c:\users\p\appdata\local\programs\python\python38\lib\site-packages
Requires: bleak-winrt
Required-by:
PS C:\Users\P\Desktop\tool-production\firmware-loader> pip.exe show winrt
Name: winrt
Version: 1.0.21033.1
Summary: Access Windows Runtime APIs from Python
Home-page: https://github.com/Microsoft/xlang/tree/master/src/package/pywinrt/projection
Author: Microsoft Corporation
Author-email: None
License: MIT
Location: c:\users\p\appdata\local\programs\python\python38\lib\site-packages
Requires:
Required-by:

PS C:\Users\P\Desktop\tool-production\firmware-loader> python.exe .\main_v2.py
Traceback (most recent call last):
  File ".\main_v2.py", line 13, in <module>
    from source.CLIToolHandler.bleHandler import downloadUploadConfigFile, LKey
  File "C:\Users\P\Desktop\tool-production\firmware-loader\source\CLIToolHandler\bleHandler.py", line 3, in <module>
    from bleak import BleakClient
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\__init__.py", line 78, in <module>
    from bleak.backends.winrt.scanner import (
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\scanner.py", line 7, in <module>
    from bleak_winrt.windows.devices.bluetooth.advertisement import (
  File "C:\Users\P\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak_winrt\__init__.py", line 3, in <module>
    from . import _winrt
ImportError: DLL load failed while importing _winrt: The specified module could not be found.
```

I upgraded the bleak library from 0.12.1 to 0.14.1 and while it worked before - it connected - now I get this error immediately after running the script (python3.8).

On another computer I update lib from 0.12.1 to 0.14.1 and run the same script and I do not get such error - the library versions are the same (python 3.7).

I did not go to test the GATT services error - I stopped at this error

_Originally posted by @pawelm87 in https://github.com/hbldh/bleak/issues/740#issuecomment-1014347974_

dlech, 2022-01-18:
This was solved by installing the Microsoft Visual C++ 2022 redistributable. However we should not require users to do this.

I think we can statically link the `_winrt` module by using the `/MT` compiler option.

hbldh, 2022-01-18:
If you are able to include this in the `bleak-winrt` package, then we save the users from a potentially disheartening situation. I am all for it; it is a good find.

I was continually surprised that the `pythonnet` backend never needed any such package to be installed, but now it seems that we have come to the end of the road. If you are able to wrap in a dependency you control, then it is handled as good as it can be in my book.

PR #749 (2022-01-26): v0.14.2

Changed
-------

* Updated ``bleak-winrt`` dependency to v1.1.1. Fixes #741.

Fixed
-----

* Fixed ``name`` is ``'Unknown'`` in WinRT backend. Fixes #736.
closed 2022-01-18 by dlech