hbldh/bleak #743: TypeError: 'objc_meta_class' object is not iterable

Aborgh, 2022-01-20:
* bleak version: 0.14.1
* Python version: 3.10.0b4
* Operating System: macOS 12.1
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Trying to run the example script on mac(m1) results in `TypeError: 'objc_meta_class' object is not iterable`.

### What I Did

I tried to run the example code from the README 
```
import asyncio
from bleak import BleakScanner

async def main():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)

asyncio.run(main())

```
but this resulted in getting  the `TypeError: 'objc_meta_class' object is not iterable` error. As well as `raise BleakError("Bleak requires the CoreBluetooth Framework") from ex
bleak.exc.BleakError: Bleak requires the CoreBluetooth Framework`

I've tried adding `import CoreBluetooth` to the script but this just got me the `object is not iterable` error again.

I've also added my terminal (iTerm) to System Preferences -> Security & Privacy -> Bluetooth as suggested here #438 but with no luck.

Am I missing something?

dlech, 2022-01-20:
You need to install the required packages. How did you install Bleak?

Aborgh, 2022-01-20:
With pip `pip install bleak` and `pip install pyobjc` for good measure

dlech, 2022-01-20:
Sounds like a bug in PyObjC? What is the output of `pip show pyobjc-core`?

dlech, 2022-01-20:
It also looks like you are still running a beta version of Python 3.10 which could have bugs.

Aborgh, 2022-01-20:
> It also looks like you are still running a beta version of Python 3.10 which could have bugs.

That's it! Can't believe I didn't think of that. Thank you and sorry for taking your time!

Aborgh, 2022-01-20:
To clarify, running with python 3.9 solved it.

closed 2022-01-20 by Aborgh