hbldh/bleak #744: Bleak on raspberry is stuck in endless loop when scanning /pairing

TheAppService, 2022-01-23:
* bleak version: Version: 0.14.1
Summary: Bluetooth Low Energy platform Agnostic Klient
* Python version: Python 3.9.2
* Operating System: Raspbian Linux raspberrypi 5.10.63+ #1488 Thu Nov 18 16:14:04 GMT 2021 armv6l GNU/Linux
* BlueZ version (`bluetoothctl -v`) in case of Linux:
bluetoothctl: 5.55
### Description

Tried to start a Device Scan with Bleak and /or connecting to a bluetooth device, doesn't really matter. Sometimes in very rare cases it works (I absolutely don't know why.) But every other time, bleak/the bluetooth adapter, idk is stuck in an "endless looü
### What I Did

If I run this script without BLEAK_logging=1 then they run forever.
If I have bluetoothctl open at the same time, I see the following output:

```import asyncio
from bleak import discover

async def run():
    devices = await discover()
    for d in devices:
        print(d)

loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
Output till interrupted after about 10 seconds:
Console with python3 test.py (Does not matter if executed with sudo, pi user is in the bluetooth group)
```
python3 test.py 
2022-01-23 17:37:24,941 bleak.backends.bluezdbus.scanner DEBUG: cached devices: {}
2022-01-23 17:37:25,495 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.Properties.PropertiesChanged (/org/bluez/hci0): ['org.bluez.Adapter1', {'Discovering': <dbus_next.signature.Variant ('b', True)>}, []]
2022-01-23 17:37:25,821 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_60_42_37_25_A9_C8', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 60:42:37:25:A9:C8)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 60-42-37-25-A9-C8)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -52)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {76: <dbus_next.signature.Variant ('ay', b'\x0c\x0e\x00\xf4 \x16jM\xec,\xc3\xc5>0\xd8\x91\x10\x06N\x1dD\xbf|x')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-01-23 17:37:26,080 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_32_65_C6_CB_95_BF', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 32:65:C6:CB:95:BF)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 32-65-C6-CB-95-BF)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -74)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {76: <dbus_next.signature.Variant ('ay', b'\t\x06\x03V\xc0\xa8\x00\xbc')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-01-23 17:37:26,295 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_51_D3_75_5D_A2_00', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 51:D3:75:5D:A2:00)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 51-D3-75-5D-A2-00)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -85)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', [])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {76: <dbus_next.signature.Variant ('ay', b'\x10\x06\x15\x1e\xdb\x91\xc3\xed')>})>, 'TxPower': <dbus_next.signature.Variant ('n', 12)>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-01-23 17:37:26,565 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_48_E6_31_2A_C8_B3', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 48:E6:31:2A:C8:B3)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 48-E6-31-2A-C8-B3)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -69)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', ['0000fe9f-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {224: <dbus_next.signature.Variant ('ay', b'\x00#\xca\x9dO8')>})>, 'ServiceData': <dbus_next.signature.Variant ('a{sv}', {'0000fe9f-0000-1000-8000-00805f9b34fb': <dbus_next.signature.Variant ('ay', b'\x02PkgB_mFP9mc\x00\x00\x01~\x87\xcf\x0c\xab')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-01-23 17:37:26,663 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_5E_5A_18_DE_9D_5C', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 5E:5A:18:DE:9D:5C)>, 'AddressType': <dbus_next.signature.Variant ('s', random)>, 'Alias': <dbus_next.signature.Variant ('s', 5E-5A-18-DE-9D-5C)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -62)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_next.signature.Variant ('as', ['0000fe9f-0000-1000-8000-00805f9b34fb'])>, 'Adapter': <dbus_next.signature.Variant ('o', /org/bluez/hci0)>, 'ManufacturerData': <dbus_next.signature.Variant ('a{qv}', {224: <dbus_next.signature.Variant ('ay', b'\x00\x06\xcaik\xd1')>})>, 'ServiceData': <dbus_next.signature.Variant ('a{sv}', {'0000fe9f-0000-1000-8000-00805f9b34fb': <dbus_next.signature.Variant ('ay', b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00')>})>, 'ServicesResolved': <dbus_next.signature.Variant ('b', False)>}, 'org.freedesktop.DBus.Properties': {}}]
2022-01-23 17:37:26,715 bleak.backends.bluezdbus.scanner DEBUG: received D-Bus signal: org.freedesktop.DBus.ObjectManager.InterfacesAdded (/): ['/org/bluez/hci0/dev_00_12_6F_B4_EB_CB', {'org.freedesktop.DBus.Introspectable': {}, 'org.bluez.Device1': {'Address': <dbus_next.signature.Variant ('s', 00:12:6F:B4:EB:CB)>, 'AddressType': <dbus_next.signature.Variant ('s', public)>, 'Name': <dbus_next.signature.Variant ('s', BeoPlay A1)>, 'Alias': <dbus_next.signature.Variant ('s', BeoPlay A1)>, 'Paired': <dbus_next.signature.Variant ('b', False)>, 'Trusted': <dbus_next.signature.Variant ('b', False)>, 'Blocked': <dbus_next.signature.Variant ('b', False)>, 'LegacyPairing': <dbus_next.signature.Variant ('b', False)>, 'RSSI': <dbus_next.signature.Variant ('n', -68)>, 'Connected': <dbus_next.signature.Variant ('b', False)>, 'UUIDs': <dbus_ne [... cut ...]
[DEL] Device 32:65:C6:CB:95:BF 32-65-C6-CB-95-BF
[DEL] Device 51:D3:75:5D:A2:00 51-D3-75-5D-A2-00
[DEL] Device 5A:79:A2:41:9E:72 5A-79-A2-41-9E-72
[DEL] Device 6B:1A:F9:2D:87:4E 6B-1A-F9-2D-87-4E
[DEL] Device 00:12:6F:B4:EB:CB BeoPlay A1
[DEL] Device 4A:23:AF:46:D2:A8 4A-23-AF-46-D2-A8
[DEL] Device 58:32:85:08:84:5E 58-32-85-08-84-5E
[DEL] Device 65:8D:63:95:C0:6C 65-8D-63-95-C0-6C
[DEL] Device 16:E0:EE:EC:1A:4D FallDetection
[DEL] Device 60:42:37:25:A9:C8 60-42-37-25-A9-C8
´´´

Maybe someone can help

dlech, 2022-01-26:
I was not able to reproduce the issue. It works for me both on Ubuntu desktop and Raspberry Pi.

Daniel-Ebert, 2022-02-21:
Thank you for trying that out! Interestingly it worked with bluepy then on my pi without any issues... 
I think it probably won't be worth to figure it out further, but maybe this issue might help if someone else has this problem in the future.

closed 2022-02-21 by Daniel-Ebert