hbldh/bleak #745: Connecting to Client inside a Docker Container

bconrad98, 2022-01-24:
* bleak version: latest
* Python version: 3.8
* Operating System: Windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I am attempting to use bleak to connect to a bluetooth device, and I want to put my Python app in a Docker container using docker-compose. 
### What I Did

My application involves docker-compose with an mqtt service and a python app service, but the part that produces an error is when trying to connect to a device using the bleak library while running from a container.

```
ble_client = BleakClient(address)
await ble_client.connect()
```
### Intial Error output
```
polar_1  | Traceback (most recent call last):
polar_1  |   File "/brm/brm/ble_sensor.py", line 131, in connection
polar_1  |     await self.ble_client.connect()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 112, in connect
polar_1  |     device = await BleakScannerBlueZDBus.find_device_by_address(
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
polar_1  |     return await cls.find_device_by_filter(
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 249, in find_device_by_filter
polar_1  |     async with cls(detection_callback=apply_filter, **kwargs):
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 95, in __aenter__
polar_1  |     await self.start()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/bluezdbus/scanner.py", line 88, in start
polar_1  |     self._bus = await MessageBus(bus_type=BusType.SYSTEM).connect()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/aio/message_bus.py", line 122, in __init__
polar_1  |     super().__init__(bus_address, bus_type, ProxyObject)
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 85, in __init__
polar_1  |     self._setup_socket()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 575, in _setup_socket
polar_1  |     raise err
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 548, in _setup_socket
polar_1  |     self._sock.connect(filename)
polar_1  | FileNotFoundError: [Errno 2] No such file or directory
```
This first error came because dbus could not find the filename to connect to a socket. After searching online, I found that you could set the DBUS_SYSTEM_BUS_ADDRESS environment variable. I set the variable to `tcp:host=localhost` in the docker-compose.yml file and ended up with a different error.
### Second Error output
```
polar_1  | Traceback (most recent call last):
polar_1  |   File "/brm/brm/ble_sensor.py", line 131, in connection
polar_1  |     await self.ble_client.connect()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/bluezdbus/client.py", line 112, in connect
polar_1  |     device = await BleakScannerBlueZDBus.find_device_by_address(
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 220, in find_device_by_address
polar_1  |     return await cls.find_device_by_filter(
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 249, in find_device_by_filter
polar_1  |     async with cls(detection_callback=apply_filter, **kwargs):
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/scanner.py", line 95, in __aenter__
polar_1  |     await self.start()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/bleak/backends/bluezdbus/scanner.py", line 88, in start
polar_1  |     self._bus = await MessageBus(bus_type=BusType.SYSTEM).connect()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/aio/message_bus.py", line 122, in __init__
polar_1  |     super().__init__(bus_address, bus_type, ProxyObject)
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 85, in __init__
polar_1  |     self._setup_socket()
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 575, in _setup_socket
polar_1  |     raise err
polar_1  |   File "/usr/local/lib/python3.8/site-packages/dbus_next/message_bus.py", line 565, in _setup_socket
polar_1  |     self._sock.connect((ip_addr, ip_port))
polar_1  | ConnectionRefusedError: [Errno 111] Connection refused
```

The connection is refused. Any help is appreciated.

bconrad98, 2022-01-24:
Okay could this be because I'm using a Windows machine but the container is running a backend meant for linux?

dlech, 2022-01-25:
I'm not aware of any technology that bridges OS Bluetooth stacks like this.

closed 2022-01-25 by bconrad98

alexk-1998, 2022-04-09:
Did you ever find a solution to this problem?

builder555, 2023-02-12:
having the same problem on a Mac. did you ever figure it out @bconrad98