hbldh/bleak #746: Quick question about memoryview() in client.write_gatt_char (WinRT)

lakshjaisinghani, 2022-01-25:
Hey bleak team, 

**OS:** Windows 10
**bleak:** 0.14.1
**python:** 3.7.9

I am trying to learn about how your library works by building a basic version on windows. IN the code snippet linked below, I don't understand how a `winrt.windows.storage.streams.Buffer` can be used in `memoryview()` as it only takes byte arrays as arguments. 

https://github.com/hbldh/bleak/blob/77a9300a3be94911d4e2c603d7a023791d96d53a/bleak/backends/winrt/client.py#L607-L615

If you implement a simple data writer from scratch:
```python
from bleak_winrt.windows.storage.streams import Buffer # I'm using  from winrt.windows.storage.streams import Buffer
data =  b'\xEA\x38\x00\x01\x01'

buf = Buffer(len(data))
buf.length = buf.capacity
with memoryview(buf) as mv:
    mv[:] = data

result = await y.write_value_with_result_async(buf)
```

You get a `TypeError`:
 ```
TypeError: memoryview: a bytes-like object is required, not '_winrt_Windows_Storage_Streams.Buffer'
```

How do you guys make it work in the bleak backend? Is there a way I could write directly to the buffer without using memoryview?

dlech, 2022-01-25:
We forked the `pywinrt` code generation tool from Microsoft and made many bug fixes and improvements. See https://github.com/dlech/xlang. That fork contains CI build artifacts that contain a full build of the WinRT projection if you need more than what the `bleak-winrt` package provides.

dlech, 2022-01-25:
In particular, https://github.com/dlech/xlang/commit/9f00705a639adfdc340e728f73f364832fa58f14 makes `IBuffer` a bytes-like object. This can be used with more than just `memoryview`. For example, using `struct.pack_into()` can make the code much more readable if the data contains discrete values.

```python
buf = Buffer(5)
buf.length = 5
struct.pack_into("<BHBB", buf, 0, 0xEA, 0x0038, 0x01, 0x01)

result = await descriptor.obj.write_value_with_result_async(buf)
```

lakshjaisinghani, 2022-01-25:
That's prretty cool!

Thanks for helping out @dlech.

closed 2022-01-25 by lakshjaisinghani