hbldh/bleak #747: there are no api to read characteristic name

jinxiangwzh, 2022-01-26:
* bleak version:
* Python version:
* Operating System:
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

there are no api to read characteristic name.

[image]

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

It is also preferable if that example is not dependent on a specific peripheral device, but can be run and
reproduced with other BLE peripherals as well.

```
Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

hbldh, 2022-01-26:
There is the `characteristic.description` that provides this information, if available.

dlech, 2023-07-19:
https://bleak.readthedocs.io/en/latest/api/index.html#bleak.backends.service.BleakGATTService.description

closed 2023-07-19 by dlech