hbldh/bleak #752: kwargs on macOS 12

4193883-eng, 2022-01-28:
* bleak version: 0.14.2
* Python version: 3.9.10
* Operating System: MacOS
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

This error happened: 

> ERROR:bleak.backends.corebluetooth.scanner:macOS 12 requires non-empty service_uuids kwarg, otherwise no advertisement data will be received

### What I Did

Using the pylgbst library

hbldh, 2022-01-28:
A recent change in bleak, in response to a new requirement in macOS 12, requires a list of service uuids to scan for to be sent in when initiating scanning. 

Raise this as an issue in the pylgbst repo instead and they should be able to fix it.

dlech, 2022-01-28:
Also see #635 and #720.

closed 2022-01-28 by dlech