hbldh/bleak #754: Collect notify data from two devices (different MACs) with same UUIDs + Identify devices

cooperlees, 2022-01-30:
* bleak version: 0.14.2
* Python version: 3.9.7
* Operating System: Ubuntu 21.10
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.60

### Description

*I apologize in advance, I am sure the library is working fine here, this is just my limited BLE knowledge causing issues*
- I will hopefully repay any help here with an example / documentation update

#### My use case / setup

I have two Lithionics Batteries that each have different Bluetooth MAC addresses that I wish to pull the telemetry from each. I wish to store it in a way so I can create counters in prometheus for each respective device. They send ascii strings that I can split into a data class and then I will export to Prometheus (time series DB) via it's exporter API.

When I try to create a client per device via `BleakClient(mac_address)` the second device always fails to connect / stream data. It seems to just sit there blocked. So I then thought the one notify callback must get the data from both batteries, but that does not seem the case either as I always get the same "sender id" sent to the handler:

```
[2022-01-30 17:09:10,294] INFO: Received Li3 telementary from 17 (li3.py:50)
```
- Always client 17

My code connecting to the two device can be seen here: https://github.com/cooperlees/vanD/blob/main/src/vand/li3.py#L71
- The two devices UUIDs etc. can be found in the config: https://github.com/cooperlees/vanD/blob/main/vand.json#L9

What's the right way to get data and identify each battery (device) here so I can collect respective statistics? Do I need separate processes for each device?

[image]

### What I Did

I run a daemon that listens and converts the ascii bluetooth data into a data class and then exposes via the Prometheus Export API. For one device this is working, but I can't get it to work for two.

Any tips / tricks / info here would be appreciated. I thank you for this library and again, apaologize as I bet this is just all my limited knowledge of Bluetooth LE. This is my first time ever working with it.

dlech, 2022-01-30:
This is a commonly reported problem:  #724, #467 (it feels like it has come up more than this, but that is all I could find).

The solutions is to use `functools.partial` to wrap the callback function to include the required extra info.

But I'm wondering if it is worth making a breaking change to make it easier and more obvious. For example, we could replace the `handle` argument in the callback function with the `BleakClient` object itself. This would only break programs that are actually using the `handle` argument. I suspect that is not many because unless you are attaching the same callback to multiple characteristics, this information is not needed.

cooperlees, 2022-01-30:
The difference I see with my code vs. other examples you've shared is I that I have multiple (two in the case) class instances representing each "Battery". My callback has `self` available, so I was hoping each instance would just set the correct `self.stats`. The problem I hit, is the async context handler on the second instance seems to get stuck creating a `BleakClient`. Is this expected?

Should I just share the `BleakClient` + notifier coroutine maybe between instances of my Battery class and then just use self attributes to identify each battery?

So is this correct: If you `start_notify` for a specific service / characteristic, does that callback get all messages from all devices (MAC addresses)? So I don't need to `start_notify` for each MAC address?

dlech, 2022-01-30:
Sorry, I guess I didn't read close enough. Another common issue is that you need to use `BleakScanner` to scan for devices when you have multiple devices. Using the implicit scan in `BleakClient`, and trying to connect while still scanning in general, tends to confuse OS Bluetooth stacks.

cooperlees, 2022-01-30:
Ok, thanks for the pointers. With Ubuntu / Bluez - Doing a pre-scan for the *Service UUID** and passing the device to create the notify was the winner.

Here is the commit: https://github.com/cooperlees/vanD/commit/90e0e18197d7e34b21a84c42b795274217fb4b95

I'll do a PR adding this information to the "Troubleshooting" and maybe an example for this scenario you can possibly refer people to in the future.

PR #755 (2022-02-10): Add a Multiple Device Troubleshooting Section
- Add information / tips to help with:
  - Connection issues to multiple devices
  - Pass more parameters to notify callback function

Fixes #754
commit feee9d61 (2022-03-05): Add a Multiple Device Troubleshooting Sections (#755)

* Add a Multiple Device Troubleshooting Sections

- Add information / tips to help with:
  - Connection issues to multiple devices
  - Pass more parameters to notify callback function

Fixes #754
closed 2022-01-30 by cooperlees