hbldh/bleak #756: kivy bleak example, didn't find PythonScanCallback

pylogger-html, 2022-02-04:
* bleak version: 0.14.2
* Python version: 3.8.10
* Operating System: ubuntu 20.04 LTS
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.53

### Description

I'm trying to run the Kivy example on an android phone and final apk fails to launch with the error: "Didn't find class com.github.hbldh.bleak.PythonScanCallback". "buildozer android debug" completed successfully without errors.

### What I Did

I created a VM with ubuntu 20.04 LTS, made a virtual environment with the following packages:
appdirs==1.4.4
bleak==0.14.2
buildozer==1.2.0
colorama==0.4.4
Cython==0.29.27
dbus-next==0.2.3
distlib==0.3.4
filelock==3.4.2
importlib-metadata==4.10.1
Jinja2==3.0.3
MarkupSafe==2.0.1
pep517==0.6.0
pexpect==4.8.0
pkg_resources==0.0.0
platformdirs==2.4.1
ptyprocess==0.7.0
pytoml==0.1.21
sh==1.14.2
six==1.16.0
toml==0.10.2
virtualenv==20.13.0
zipp==3.7.0

I ran "buildozer android debug" using the included buildozer.spec file (no edits from GIT revision) and the command completed without any errors. Attached is the logcat from running the application on my phone.

# Check configuration tokens
# Ensure build layout
# Check configuration tokens
# Read available permissions from api-versions.xml
# Preparing build
# Check requirements for android
# Run 'dpkg --version'
# Cwd None
Debian 'dpkg' package management program version 1.19.7 (amd64).
This is free software; see the GNU General Public License version 2 or
later for copying conditions. There is NO warranty.
# Search for Git (git)
#  -> found at /usr/bin/git
# Search for Cython (cython)
#  -> found at /home/user/sirens/venv-sirens/bin/cython
# Search for Java compiler (javac)
#  -> found at /usr/lib/jvm/java-13-openjdk-amd64/bin/javac
# Search for Java keytool (keytool)
#  -> found at /usr/lib/jvm/java-13-openjdk-amd64/bin/keytool
# Install platform
# Run 'git config --get remote.origin.url'
# Cwd /home/user/sirens/.buildozer/android/platform/python-for-android
https://github.com/kivy/python-for-android.git
# Run 'git branch -vv'
# Cwd /home/user/sirens/.buildozer/android/platform/python-for-android
* develop 8b173d99 [origin/develop] Update recipe.py (#2544) (#2546)
# Run '/home/user/sirens/venv-sirens/bin/python -m pip install -q  \'appdirs\' \'colorama>=0.3.3\' \'jinja2\' \'six\' \'enum34; python_version<"3.4"\' \'sh>=1.10; sys_platform!="nt"\' \'pep517<0.7.0\' \'toml\''
# Cwd None
# Apache ANT found at /home/user/.buildozer/android/platform/apache-ant-1.9.4
# Android SDK found at /home/user/.buildozer/android/platform/android-sdk
# Recommended android's NDK version by p4a is: 19c
# Android NDK found at /home/user/.buildozer/android/platform/android-ndk-r19c
# Read available permissions from api-versions.xml
[... 101 lines cut ...]
02-04 09:14:19.556 16983 24570 I python  : testing python print redirection
02-04 09:14:19.556 16983 24570 I python  : Android path ['.', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/stdlib.zip', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/modules', '/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages']
02-04 09:14:19.557 16983 24570 I python  : os.environ is environ({'PATH': '/sbin:/system/sbin:/product/bin:/apex/com.android.runtime/bin:/system/bin:/system/xbin:/odm/bin:/vendor/bin:/vendor/xbin', 'MOUNT_SYSTEM_OK': 'true', 'ANDROID_BOOTLOGO': '1', 'ANDROID_ROOT': '/system', 'ANDROID_ASSETS': '/system/app', 'ANDROID_DATA': '/data', 'ANDROID_STORAGE': '/storage', 'ANDROID_RUNTIME_ROOT': '/apex/com.android.runtime', 'ANDROID_TZDATA_ROOT': '/apex/com.android.tzdata', 'EXTERNAL_STORAGE': '/sdcard', 'ASEC_MOUNTPOINT': '/mnt/asec', 'BOOTCLASSPATH': '/apex/com.android.runtime/javalib/core-oj.jar:/apex/com.android.runtime/javalib/core-libart.jar:/apex/com.android.runtime/javalib/okhttp.jar:/apex/com.android.runtime/javalib/bouncycastle.jar:/apex/com.android.runtime/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/knoxsdk.jar:/system/framework/knoxanalyticssdk.jar:/system/framework/smartbondingservice.jar:/system/framework/securetimersdk.jar:/system/framework/fipstimakeystore.jar:/system/framework/timakeystore.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/drutils.jar:/system/framework/android.test.base.jar:/system/framework/ucmopensslenginehelper.jar:/system/framework/esecomm.jar:/system/framework/tcmiface.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar:/apex/com.android.conscrypt/javalib/conscrypt.jar:/apex/com.android.media/javalib/updatable-media.jar', 'DEX2OATBOOTCLASSPATH': '/apex/com.android.runtime/javalib/core-oj.jar:/apex/com.android.runtime/javalib/core-libart.jar:/apex/com.android.runtime/javalib/okhttp.jar:/apex/com.android.runtime/javalib/bouncycastle.jar:/apex/com.android.runtime/javalib/apache-xml.jar:/system/framework/framework.jar:/system/framework/ext.jar:/system/framework/telephony-common.jar:/system/framework/voip-common.jar:/system/framework/ims-common.jar:/system/framework/knoxsdk.jar:/system/framework/knoxanalyticssdk.jar:/system/framework/smartbondingservice.jar:/system/framework/securetimersdk.jar:/system/framework/fipstimakeystore.jar:/system/framework/timakeystore.jar:/system/framework/sec_sdp_sdk.jar:/system/framework/sec_sdp_hidden_sdk.jar:/system/framework/drutils.jar:/system/framework/android.test.base.jar:/system/framework/ucmopensslenginehelper.jar:/system/framework/esecomm.jar:/system/framework/tcmiface.jar:/system/framework/QPerformance.jar:/system/framework/UxPerformance.jar', 'SYSTEMSERVERCLASSPATH': '/system/framework/services.jar:/system/framework/ethernet-service.jar:/system/framework/wifi-service.jar:/system/framework/com.android.location.provider.jar:/system/framework/uibc_java.jar:/system/framework/ssrm.jar', 'DOWNLOAD_CACHE': '/data/cache', 'MC_AUTH_TOKEN_PATH': '/efs/TEE', 'KNOX_STORAGE': '/data/knox/ext_sdcard', 'ENC_EMULATED_STORAGE_TARGET': '/storage/enc_emulated', 'ANDROID_SOCKET_zygote_secondary': '21', 'ANDROID_SOCKET_usap_pool_secondary': '22', 'ANDROID_ENTRYPOINT': 'main.pyc', 'ANDROID_ARGUMENT': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'ANDROID_APP_PATH': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'ANDROID_PRIVATE': '/data/user/0/com.github.hbldh.bleakdemo/files', 'ANDROID_UNPACK': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'PYTHONHOME': '/data/user/0/com.github.hbldh.bleakdemo/files/app', 'PYTHONPATH': '/data/user/0/com.github.hbldh.bleakdemo/files/app:/data/user/0/com.github.hbldh.bleakdemo/files/app/lib', 'PYTHONOPTIMIZE': '2', 'P4A_BOOTSTRAP': 'SDL2', 'PYTHON_NAME': 'python', 'P4A_IS_WINDOWED': 'True', 'P4A_ORIENTATION': 'portrait', 'P4A_NUMERIC_VERSION': 'None', 'P4A_MINSDK': '21', 'LC_CTYPE': 'C.UTF-8'})
02-04 09:14:19.557 16983 24570 I python  : Android kivy bootstrap done. __name__ is __main__
02-04 09:14:19.557 16983 24570 I python  : AND: Ran string
02-04 09:14:19.557 16983 24570 I python  : Run user program, change dir and execute entrypoint
02-04 09:14:19.643 16983 24570 I python  : [INFO   ] [Logger      ] Record log in /data/user/0/com.github.hbldh.bleakdemo/files/app/.kivy/logs/kivy_22-02-04_1.txt
02-04 09:14:19.644 16983 24570 I python  : [INFO   ] [Kivy        ] v2.0.0
02-04 09:14:19.644 16983 24570 I python  : [INFO   ] [Kivy        ] Installed at "/data/user/0/com.github.hbldh.bleakdemo/files/app/_python_bundle/site-packages/kivy/__init__.pyc"
02-04 09:14:19.644 16983 24570 I python  : [INFO   ] [Python      ] v3.8.9 (default, Feb  4 2022, 09:07:38) 
02-04 09:14:19.644 16983 24570 I python  : [Clang 8.0.2 (https://android.googlesource.com/toolchain/clang 40173bab62ec7462
02-04 09:14:19.644 16983 24570 I python  : [INFO   ] [Python      ] Interpreter at ""
02-04 09:14:20.539 16983 24570 I python  : [INFO   ] [Factory     ] 186 symbols loaded
02-04 09:14:20.737 16983 24570 I python  : [INFO   ] [Image       ] Providers: img_tex, img_dds, img_sdl2 (img_pil, img_ffpyplayer ignored)
02-04 09:14:20.760 16983 24570 I python  : [INFO   ] [Text        ] Provider: sdl2
02-04 09:14:21.215 16983 24570 I python  :  Traceback (most recent call last):
02-04 09:14:21.215 16983 24570 I python  :    File "/home/user/sirens/.buildozer/android/app/main.py", line 14, in <module>
[... cut ...]
02-04 09:14:21.217 16983 24570 I python  :  jnius.jnius.JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[zip file "/data/app/com.github.hbldh.bleakdemo-z21-1eW84jd7w4EncN4hEw==/base.apk"],nativeLibraryDirectories=[/data/app/com.github.hbldh.bleakdemo-z21-1eW84jd7w4EncN4hEw==/lib/arm, /data/app/com.github.hbldh.bleakdemo-z21-1eW84jd7w4EncN4hEw==/base.apk!/lib/armeabi-v7a, /system/lib, /system/product/lib]] java.lang.ClassNotFoundException
02-04 09:14:21.217 16983 24570 I python  : Python for android ended.

pylogger-html, 2022-02-07:
After some trial and error, I was able to get this example to run on my Android phone. To do this I had to copy the PythonScanCallback.java & PythonBluetoothGattCallback.java files from Bleak to my local directory and explicitly add them as source files in the buildozer.spec (android.add_src). This worked and the application finally ran.

It would be nice to not have to do that or to have a better way of doing things, but any working solution is better than nothing.

Lukwrt, 2022-02-11:
Had a similar issue that @pylogger-html had. I tried to run the kivy example with the buildozer.spec and the main.py in the develop branch and had an issue with PythonScanCallBack. I'm trying the fix suggested but it's true at first it's a bit destabilizing that the example is not working right away. I'll edit this to tell if the fix worked or not.

dlech, 2022-02-11:
CI tests for the Android build started failing a few months ago, so we disabled them. Something must have changed in the tooling. You could compare https://github.com/hbldh/bleak/runs/4450281250?check_suite_focus=true and https://github.com/hbldh/bleak/runs/4668307926?check_suite_focus=true to see if there is a difference in the installed dependencies. That should provide a clue of where to look for what changed.

navr32, 2022-07-22:
Hello ! same problem. 

07-22 13:12:01.635 17915 18525 I python  : jnius.jnius.JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[zip file "/data/app/org.test.blueble-1/base.apk"],nativeLibraryDirectories=[/data/app/org.test.blueble-1/lib/arm64, /data/app/org.test.blueble-1/base.apk!/lib/arm64-v8a, /system/lib64, /vendor/lib64, /system/vendor/lib64]] java.lang.ClassNotFoundException

Someone have better fix ? Have a nice days. many thanks for all the work.

navr32, 2022-07-24:
After some search i have reinstall and clean my virtualenv , reinstall dependcies cython,pyjnius python for android..and now this build clean.. many thanks ..and i don't know where the error exactly come from.  I build the 0.14.3 or the master and this build.

avzh72, 2022-12-07:
Hello ! same problem. 
"I/python  (10027):  jnius.jnius.JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[zip file "/data/app/com.github.hbldh.bleakdemo-2/base.apk"],nativeLibraryDirectories=[/data/app/com.github.hbldh.bleakdemo-2/lib/arm, /vendor/lib, /system/lib]] java.lang.ClassNotFoundException"
you reinstalled all packages for buildozer, or only virtualenv, cython, pyjnius?

navr32, 2022-12-07:
Hi ! I think just the package i have tell in previous post. 
In my memory buildozer was not the problem..i hope..because this is very long time i have done this..apologized..

navr32, 2022-12-07:
On which system do you run ? If linux i have detect that for python package support official by the distribution i have to install them by the package manager..and for package not in build support but just alternative package they give errors...
So finally install package from official build binaries distribution ..and install others by python package manager pip or others..and after all was good...

avzh72, 2022-12-07:
Thanks. it didn't work for me. The advice above helped me...
> After some trial and error, I was able to get this example to run on my Android phone. To do this I had to copy the PythonScanCallback.java & PythonBluetoothGattCallback.java files from Bleak to my [...]

navr32, 2022-12-07:
Yes curious thing...perhaps clean all buildozer and reinstall clean ..How do you install buildozer ? Pip or package distrib ?

avzh72, 2022-12-08:
pip3 install  buildozer
I'll try to install it again. I use virtualbox with lubuntu 20.04

obzerving, 2023-08-22:
So, it's August 2023 and pylogger-html's solution is still the workaround for this issue. Thank you, pylogger-html.

robgar2001, 2023-08-29:
@obzerving could you try again using the latest development version of the kivy example? I had a look at the bleak recipe in #1398. This issue might be fixed now. See the readme, since some stuff has changed.

The previous version of the recipe was quite directory dependent. Which might have caused the recipe to not be able to copy the Java files correctly for you.

Let me know if you find anything

Kind regards,

robgar2001

DemonBhaiii, 2024-02-15:
hi, i know i am a little late but i have figured it out. this error is coming because the script could not find the jar file. to fix this issue 
find download these java files
1. [https://github.com/hbldh/bleak/blob/develop/bleak/backends/p4android/java/com/github/hbldh/bleak/PythonBluetoothGattCallback.java](url)
2. [https://github.com/hbldh/bleak/blob/develop/bleak/backends/p4android/java/com/github/hbldh/bleak/PythonBluetoothGattCallback.java](url)

now make some folders in your main app directory
com/github/hbldh/bleak
in this bleak folder paste both the files with same name 
now 
compile these java files by using these commands

1. javac -classpath /home/<user name>/.buildozer/android/platform/android-sdk/platforms/android-31/android.jar PythonBluetoothGattCallback.java
2. javac -classpath /home/<user name>/.buildozer/android/platform/android-sdk/platforms/android-31/android.jar PythonBluetoothGattCallback.java

make sure this is the path where your .buildozer is or you could just use android SDK for it also

now convert them to jar file
open terminal and go to this directory with it 
com/github/hbldh/bleak

run this command
jar cf bluetooth_function.jar *.class

now finally in your buildozer.spec file
add this line
android.add_jars =  %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar

and you will be good to go ;)

dlech, 2024-02-20:
> Copy and paste the above links for some reason they are not working when you click them

fixed

abofgames, 2024-06-24:
hi there

I tried the Demonbhai's solution and just adding the java files using android.add_src But I keep getting the same error 

'''06-24 20:31:14.156 15242 16486 I python  :  Exception in thread Thread-3 (connect_to_device_async):
06-24 20:31:14.156 15242 16486 I python  :  Traceback (most recent call last):
06-24 20:31:14.156 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/other_builds/python3/arm64-v8a__ndk_target_24/python3/Lib/threading.py", line 1038, in _bootstrap_inner
06-24 20:31:14.157 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/jnius/__init__.py", line 85, in jnius_thread_hook
06-24 20:31:14.157 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/other_builds/python3/arm64-v8a__ndk_target_24/python3/Lib/threading.py", line 975, in run
06-24 20:31:14.157 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/app/main.py", line 592, in connect_to_device_async
06-24 20:31:14.158 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/bleak/__init__.py", line 521, in __init__
06-24 20:31:14.158 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/bleak/backends/client.py", line 255, in get_platform_client_backend_type
06-24 20:31:14.158 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/bleak/backends/p4android/client.py", line 19, in <module>
06-24 20:31:14.158 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/bleak/backends/p4android/defs.py", line 34, in <module>
06-24 20:31:14.158 15242 16486 I python  :    File "/home/abof/PycharmProjects/mytibai/.buildozer/android/platform/build-arm64-v8a/build/python-installs/mytibai/arm64-v8a/jnius/reflect.py", line 209, in autoclass
06-24 20:31:14.159 15242 16486 I python  :    File "jnius/jnius_export_func.pxi", line 22, in jnius.jnius.find_javaclass
06-24 20:31:14.159 15242 16486 I python  :    File "jnius/jnius_utils.pxi", line 79, in jnius.jnius.check_exception
06-24 20:31:14.159 15242 16486 I python  :  jnius.jnius.JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[directory "."],nativeLibraryDirectories=[/system/lib64, /system/system_ext/lib64, /system/lib64, /system/system_ext/lib64]] java.lang.ClassNotFoundException
'''
and I also used pip to uninstall and reinstall the py modules such as bleak, buildozer and pyjnius
and ofc do an appclean whenever I change anything in the the buildozer file.

VirajCEO, 2024-06-27:
Hi abofgames can you show me your project directory so that i could have a clear understanding what exactly is the issue here?

abofgames, 2024-06-27:
> Hi abofgames can you show me your project directory so that i could have a clear understanding what exactly is the issue here?

Hi there , here is both ways I used to add the java files 
[image]
and I used this in the buldozer file : 
android.add_jars = %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar
and this android.add_src = java
and they both gave the same result

abofgames, 2024-06-27:
[image]
`[app]

# (str) Title of your application
title = tib ai technology

# (str) Package name
package.name = mytibai

# (str) Package domain (needed for android/ios packaging)
package.domain = org.test

# (str) Source code where the main.py live
source.dir = .

# (list) Source files to include (let empty to include all the files)
source.include_exts = py,png,jpg,kv,atlas

# (list) List of inclusions using pattern matching
#source.include_patterns = assets/*,images/*.png

# (list) Source files to exclude (let empty to not exclude anything)
#source.exclude_exts = spec

# (list) List of directory to exclude (let empty to not exclude anything)
#source.exclude_dirs = /tests

# (list) List of exclusions using pattern matching
# Do not prefix with './'
#source.exclude_patterns = license,images/*/*.jpg

# (str) Application versioning (method 1)
version = 0.1

# (str) Application versioning (method 2)
# version.regex = __version__ = ['"](.*)['"]
# version.filename = %(source.dir)s/main.py

# (list) Application requirements
# comma separated e.g. requirements = sqlite3,kivy      materialyoucolor,exceptiongroup,asyncgui,asynckivy,
requirements = python3,
    kivy==2.3.0,
    kivymd,
    mysql.connector,
    numpy,
    pillow,
    android,
    typing_extensions,
    materialyoucolor,
    exceptiongroup,
    asyncgui,
    asynckivy,
    plyer,
    pyjnius,
    Kivy-Garden,
    pygments,
    bleak==0.22.2,
    async_to_sync,
    async-timeout,
    android.permissions

# (str) Custom source folders for requirements
# Sets custom source for any requirements with recipes
# requirements.source.kivy = ../../kivy

# (str) Presplash of the application
#presplash.filename = %(source.dir)s/data/presplash.png

# (str) Icon of the application
#icon.filename = %(source.dir)s/data/icon.png
[... 387 lines cut ...]
#
#[app:source.exclude_patterns]
#license
#data/audio/*.wav
#data/images/original/*
#

#    -----------------------------------------------------------------------------
#    Profiles
#
#    You can extend section / key with a profile
#    For example, you want to deploy a demo version of your application without
#    HD content. You could first change the title to add "(demo)" in the name
#    and extend the excluded directories to remove the HD content.
#
#[app@demo]
#title = My Application (demo)
#
#[app:source.exclude_patterns@demo]
#images/hd/*
#
#    Then, invoke the command line with the "demo" profile:
#
#buildozer --profile demo android debug
`

VirajCEO, 2024-06-27:
where is your main.py file? here in the screenshot of the directory you shared? is this your source directory?

abofgames, 2024-06-27:
here is a screenshot of the whole directory 
[image]

abofgames, 2024-06-27:
here I did what what you told me in a different env I was testing in :
[image]

setting source.dir to ./myapp tried compiling both using the jar file and by adding the java directory and I still get the same error 
[image]

VirajCEO, 2024-06-27:
Do one thing try making a basic java class that would return hello world and then call it with pyjnius in python script and try changing the path again and again by doing hit and trial 

Java code:
package com.github.hbldh.bleak

public class HelloWorld {
    public String sayHello() {
        return "Hello, World!";
    }

    public static void main(String[] args) {
        HelloWorld hw = new HelloWorld();
        System.out.println(hw.sayHello());
    }
}

now compile it whit this command

VirajCEO, 2024-06-27:
javac HelloWorld.java

then lastly try to call it with pyjnius

import os
from jnius import autoclass
# Import the HelloWorld Java class
HelloWorld = autoclass('com.source.bleak.HelloWorld')

# Create an instance of the HelloWorld class
hw_instance = HelloWorld()

# Call the sayHello method
print(hw_instance.sayHello())

if you are able to call it then it means at least your directory placement is good then we will try fixing it with some other method

VirajCEO, 2024-06-27:
For your convenience here is the java code
`package com.github.hbldh.bleak;
public class HelloWorld {
    public String sayHello() {
        return "Hello, World!";
    }

    public static void main(String[] args) {
        HelloWorld hw = new HelloWorld();
        System.out.println(hw.sayHello());
    }
}
`

and here is the python code 
`import os
from jnius import autoclass

# Import the HelloWorld Java class
HelloWorld = autoclass('com.github.hbldh.bleak.HelloWorld')

# Create an instance of the HelloWorld class
hw_instance = HelloWorld()

# Call the sayHello method
print(hw_instance.sayHello())

`

Sorry i didn't do it in one go just new in all this github issues thing

abofgames, 2024-06-27:
which method should I use , android.add_jars or android.add_src ?
and thank you for providing the code ^^

abofgames, 2024-06-27:
yup I got the same error :

[image]
here is my env :

[image]

and I have this in my buldozer.spec file : 

android.add_src = java

VirajCEO, 2024-06-28:
no no you don't need to compile the app with buildozer just run it on your pc to test weather the locaation of class file we are using is valid or not

remember to install pyjnius
with pip install pyjnius

just make test.py script in the project directory with the code i have provided to you earlier and then run it

Shubhamm33, 2025-01-19:
Can anyone help me with this?

bleak version: 0.22.3
Python version: 3.12.3
Operating System: ubuntu 24 LTS
BlueZ version (bluetoothctl -v) in case of Linux: 5.72

same issue of PythonScanCallback when building with buildozer
tried all options mentioned above its not working
jnius is not able to load the .java files while building

its a simple application in kivy which just scans and displays devices nearby

[image]

`
# main.py
from kivy.app import App
from kivy.uix.boxlayout import BoxLayout
from kivy.uix.button import Button
from kivy.uix.label import Label
from bleak import BleakScanner
import asyncio 

# from jnius import autoclass
# autoclass('org.jnius.NativeInvocationHandler')
# autoclass('com.github.hbldh.bleak.PythonScanCallback')
# autoclass('com.github.hbldh.bleak.PythonBluetoothGattCallback')
# autoclass('com.github.hbldh.bleak.PythonScanCallback$Interface')

class ScanButton(Button):
    def run_scan(self, label):
        label.text = "Scanning please wait"
        asyncio.create_task(self.async_run_scan(label))
        return
    
    async def async_run_scan(self, label):
        devices = await BleakScanner.discover()
        device_names = [device.name for device in devices]
        print(device_names)
        label.text = "\n".join(device_names) if device_names else "No devices found"

class MyApp(App):
    pass

if __name__ == '__main__':
    asyncio.run(MyApp().async_run())
`

`BoxLayout:
    orientation: "vertical"

    Label:
        id: _text
        text: "Scan for devices"
    
    ScanButton:
        text: "Scan"
        on_press: self.run_scan(_text)`

`
[app]

# (str) Title of your application
title = BLE Test

# (str) Package name
package.name = bletest

# (str) Package domain (needed for android/ios packaging)
[... 428 lines cut ...]
#license
#data/audio/*.wav
#data/images/original/*
#

#    -----------------------------------------------------------------------------
#    Profiles
#
#    You can extend section / key with a profile
#    For example, you want to deploy a demo version of your application without
#    HD content. You could first change the title to add "(demo)" in the name
#    and extend the excluded directories to remove the HD content.
#
#[app@demo]
#title = My Application (demo)
#
#[app:source.exclude_patterns@demo]
#images/hd/*
#
#    Then, invoke the command line with the "demo" profile:
#
#buildozer --profile demo android debug
`

not able to crack this , please help ?

allofmeng, 2025-01-21:
> After some trial and error, I was able to get this example to run on my Android phone. To do this I had to copy the PythonScanCallback.java & PythonBluetoothGattCallback.java files from Bleak to my [...]

Hey I have similar issue , and I tried to put these 2 java files into my directory. 
I want to know how much in detail i should specified in spec file ? 
Can i write 
android.add_src = src 
or i should write like below ? 
android.add_src  = /Users/markc/Documents/decentkivy/scaleandroid/.buildozer/android/app/build/libffi/iphoneos-arm64/libffi-3.4.4/src
Thank you

Shubhamm33, 2025-01-21:
> > After some trial and error, I was able to get this example to run on my Android phone. To do this I had to copy the PythonScanCallback.java & PythonBluetoothGattCallback.java files from Bleak to my local directory and explicitly add them as source files in the buildozer.spec (android.add_src). This worked and the application finally ran.
> > It would be nice to not have to do that or to have a better way of doing things, but any working solution is better than nothing.
> 
> Hey I have similar issue (
> 
> error: Reconnection Error: JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[directory "."],nativeLibraryDirectories=[/system/lib64, /system_ext/lib64, /system/lib64, /system_ext/lib64]] java.lang.ClassNotFoundException) ,
> 
> and I tried to put these 2 java files into my directory. I want to know how much in detail i should specified in spec file ? Can i write android.add_src = src or i should write like below ? android.add_src = /Users/markc/Documents/decentkivy/scaleandroid/.buildozer/android/app/build/libffi/iphoneos-arm64/libffi-3.4.4/src Thank you

@allofmeng 
I was able to fix this issue, but getting another issue at the moment for permissions
coming back to this issue

The solution provided by @DemonBhaiii worked for me

1. Create some folders in your working directory
_com/github/hbldh/bleak_

2. Download the following files and keep it in new folder create inside "bleak" folder (keep the filename as it, dont change it)
- [PythonBluetoothGattCallback.java](https://github.com/hbldh/bleak/blob/develop/bleak/backends/p4android/java/com/github/hbldh/bleak/PythonBluetoothGattCallback.java)
- [PythonScanCallback.java](https://github.com/hbldh/bleak/blob/develop/bleak/backends/p4android/java/com/github/hbldh/bleak/PythonScanCallback.java)

3. Compile these java files using the android.jar which would be present in hidden folder of ".buildozer" in your user directory, basically it uses the "android-sdk" present to compile these java file
- `javac -classpath /home/{YOUR-USER}/.buildozer/android/platform/android-sdk/platforms/android-31/android.jar PythonBluetoothGattCallback.java`
- `javac -classpath /home/{YOUR-USER}/.buildozer/android/platform/android-sdk/platforms/android-31/android.jar PythonBluetoothGattCallback.java`

4. Once it completes successfully, there will be 4 new files created with extension as _".class"_

5. Now convert them to jar file 
- open another new terminal and go to this directory with it _com/github/hbldh/bleak_
- run this command `jar cf bluetooth_function.jar *.class`

6.  Now finally in your buildozer.spec file add this line 
`android.add_jars = %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar`

7. Then do your `buildozer -v android debug` the application would start

_Coming back to my new issue it shows User denied permissions, but since thats another issue, that i'll check if anyone has posted it and continue there_

allofmeng, 2025-01-21:
Hey 👋 Shubhamm33
Thanks for the tip I will try it out.
Somehow I can't see android 31 under my . buildozer directory, but I will find other ways to compile. 

#i had similar permission denied issue , I had to add permission in my main.py and also spec file in order to grant access to APK.

Shubhamm33, 2025-01-21:
> Hey 👋 Shubhamm33 Thanks for the tip I will try it out. Somehow I can't see android 31 under my . buildozer directory, but I will find other ways to compile.
> 
> #i had similar permission denied issue , I had to add permission in my main.py and also spec file in order to grant access to APK.

Yes i tried adding them in both _py_ and _spec_ files, but its still showing denied
so i'm checking that currently

and regarding this
the .buildozer directory is present in your user directory, if you check with `ls -la` in new terminal it should show the buildozer directory

the one which is created in project-working directory is not the one
[image]

allofmeng, 2025-01-23:
Hey @Shubhamm33 
Thanks for the advice, I tried to follow through. 
I was able to build the apk , but when I ran in my emulator. 
It gave me below error message , it seems it still can not find the right .jar
Reconnection Error: JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[directory "."],nativeLibraryDirectories=[/system/lib64, /system_ext/lib64, /system/lib64, /system_ext/lib64]] java.lang.ClassNotFoundException

It seems to me that the java command compile the file ok , but somehow it still can't find them according to paths. 
Much appreciated your help.  

Here are my spec file path. 
android.add_jars = %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar
android.add_src =bleak

My directory : 

[image]

Shubhamm33, 2025-01-23:
Hey @allofmeng
Good to hear that the compilation went smooth

- you can keep this in spec file
`android.add_jars = %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar`
- and remove the `add_src = bleak`
- Also Please note that, there are no spelling errors in spec file, because that also breaks the apk building.
- Clean the build `buildozer android clean` 
- Then start new build `buildozer -v android debug`
- It will create apk in "bin" folder

(Also please check if your `source.dir` is in the `scaleandroid` folder, or outside it, maybe the directory structure is causing it not being to find the .jar)
is it failing when it reaches "bluetooth.jar" not found or something like that ?

Let me know.

allofmeng, 2025-01-27:
> Hey [@allofmeng](https://github.com/allofmeng) Good to hear that the compilation went smooth
> 
> * you can keep this in spec file
>   `android.add_jars = %(source.dir)s/com/github/hbldh/bleak/bluetooth_function.jar`
> * and remove the `add_src = bleak`
> * Also Please note that, there are no spelling errors in spec file, because that also breaks the apk building.
> * Clean the build `buildozer android clean`
> * Then start new build `buildozer -v android debug`
> * It will create apk in "bin" folder
> 
> (Also please check if your `source.dir` is in the `scaleandroid` folder, or outside it, maybe the directory structure is causing it not being able to find the .jar) is it failing when it reaches "bluetooth.jar" not found or something like that ?
> 
> Let me know.

Hey @Shubhamm33 
I tried your suggestion , it is able to compile into APK, and it can run on my android device. However, whenever I try to use bluetooth in the app, it will show me below code. 
Reconnection Error: JavaException: JVM exception occurred: Didn't find class "com.github.hbldh.bleak.PythonScanCallback" on path: DexPathList[[directory "."],nativeLibraryDirectories=[/system/lib64, /system_ext/lib64, /system/lib64, /system_ext/lib64]] java.lang.ClassNotFoundException

I am trying different ways to specify my java path. But the same code keeps coming up. 
Thanks for your tips so far, Do you know how should I fix this ?

Shubhamm33, 2025-01-30:
Hello @allofmeng 
its compiling into apk that means there is no problem in building process
but since its not able to pick the _blueooth.jar_ file , that means there is something wrong related to the jar file
because even after mentioning _"jar"_ path , its compiling,
but when running, its showing error (that means the _"jar"_ file is not being added into the apk )
are you sure you're specifying it correctly ?
can you share you .spec file and folder structure here ?

allofmeng, 2025-02-05:
> Hello [@allofmeng](https://github.com/allofmeng) its compiling into apk that means there is no problem in building process but since its not able to pick the _blueooth.jar_ file , that means there [...]

Hello @Shubhamm33 
Attached pic is my directory structure.  and .txt file for spec. 

[spec.txt](https://github.com/user-attachments/files/18665532/spec.txt)

[image]

Shubhamm33, 2025-02-05:
Hey @allofmeng 
Your directory structure seems correct,
but your spec file has minor problem,
adding "#" means its commenting that line and not implementing that line
also you've added wrong path for jar

[image]

Here, line 182 should be above without the "#" in beginning
so that would fix this ,
attaching the spec file below
please use this spec file and try building it again, that should do it 
let me know
- [spec.txt](https://github.com/user-attachments/files/18676323/spec.txt)