hbldh/bleak #759: proposed breaking change: change handle argument type in notification callbacks

dlech, 2022-02-06:
This is a commonly reported problem:  #724, #467 (it feels like it has come up more than this, but that is all I could find).

The solutions is to use `functools.partial` to wrap the callback function to include the required extra info.

But I'm wondering if it is worth making a breaking change to make it easier and more obvious. For example, we could replace the `handle` argument in the callback function with the `BleakClient` object itself. This would only break programs that are actually using the `handle` argument. I suspect that is not many because unless you are attaching the same callback to multiple characteristics, this information is not needed.

_Originally posted by @dlech in https://github.com/hbldh/bleak/issues/754#issuecomment-1025196739_

dlech, 2022-02-06:
Having thought about this a bit more, instead of replacing the `handle` argument with the `BleakClient` object, it would make more sense to replace it with a `BleakGATTCharacteristic` object. Then anyone using the `handle` argument would only have to make a small change (basically replace `handle` with `char` in the declaration and `char.handle` where it is read). Then we could add "parent" properties to be able to get the parent service of the characteristic and the parent `BleakClient` of the service.

koenvervloesem, 2022-02-07:
I like your second suggestion, passing the characteristic instead of the handle to the callback function. This also feels more natural from a general point of view, as notifications are about the characteristic anyway.

cooperlees, 2022-02-08:
+1 on the second option. Would be great to automatically get the Client + Characteristic in a callback by default. At the moment my code has it's own class to do this, but am happy for the breaking change and small refactor.

chill-git, 2022-02-20:
Why not let the user provide an arbitrary reference?
When they install the callback, they can specify what it is they want to get back.

commit f20705f0 (2022-09-22): BREAKING: change notification callback argument

This is a breaking change that changes the first argument of the
notification callback from the handle to the BleakGATTCharacteristic
object.

This has been a commonly reported problem by users (so much so it is
in our troubleshooting guide which can now be removed) and #759 has
received positive feedback for the breaking change.

It is likely that most users don't use the first argument anyway, so
in those cases, this won't actually be a breaking change.

Fixes: #759
PR #1020 (2022-09-22): Notify callback rework
Fixes #759. 

See commit messages and the issue for details.
commit be97338d (2022-09-22): BREAKING: change notification callback argument

This is a breaking change that changes the first argument of the
notification callback from the handle to the BleakGATTCharacteristic
object.

This has been a commonly reported problem by users (so much so it is
in our troubleshooting guide which can now be removed) and #759 has
received positive feedback for the breaking change.

It is likely that most users don't use the first argument anyway, so
in those cases, this won't actually be a breaking change.

Fixes: #759
PR #1027 (2022-09-23): Release/0.18.0

Changed
-------
* Relaxed ``async-timeout`` dependency version to support different installations. Merged #1009.
* ``BleakClient.unpair()`` in WinRT backend can be called without being connected first. Merged #1012.
* Use relative imports internally. Merged #1007.
* ``BleakScanner`` and ``BleakClient`` are now concrete classes. Fixes #582.
* Deprecated ``BleakScanner.register_detection_callback()``.
* Deprecated ``BleakScanner.set_scanning_filter()``.
* Deprecated ``BleakClient.set_disconnected_callback()``.
* Deprecated ``BleakClient.get_services()``.
* Refactored common code  [... cut]
closed 2022-09-22 by dlech