hbldh/bleak #761: BT discovery crashes on MacOS 12.2, "attempted to access privacy-sensitive data without a usage description"

jseppanen, 2022-02-11:
* bleak version: 0.14.2
* Python version:  3.10.2
* Operating System: MacOS 12.2
* Hardware: Apple Silicon/arm64

### Description

Bluetooth discovery crashes (segfaults).

### What I Did

```
% cat x.py
import asyncio
from bleak import BleakScanner

hub_uuid = "00001623-1212-EFDE-1623-785FEABCD123"

async def main():
    devices = await BleakScanner.discover(service_uuids=[hub_uuid])
    for d in devices:
        print(d)

asyncio.run(main())

% BLEAK_LOGGING=1 python x.py
zsh: abort      BLEAK_LOGGING=1 python x.py
```

Traceback:
```
Crashed Thread:        1  Dispatch queue: com.apple.root.default-qos

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Termination Reason:    Namespace TCC, Code 0 
This app has crashed because it attempted to access privacy-sensitive data without a usage description. The app's Info.plist must contain an NSBluetoothAlwaysUsageDescription key with a string value explaining to the user how the app uses this data.

Thread 0::  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	       0x1b424d0c0 __psynch_cvwait + 8
1   libsystem_pthread.dylib       	       0x1b4285808 _pthread_cond_wait + 1228
2   Python                        	       0x100b5d5d4 PyThread_acquire_lock_timed + 396
3   Python                        	       0x100bba2ac acquire_timed + 256
4   Python                        	       0x100bba4e8 lock_PyThread_acquire_lock + 56
5   Python                        	       0x100a12a34 method_vectorcall_VARARGS_KEYWORDS + 156
6   Python                        	       0x100b01854 call_function + 128
7   Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
8   Python                        	       0x100af30ec _PyEval_Vector + 328
9   Python                        	       0x100b01854 call_function + 128
10  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
11  Python                        	       0x100af30ec _PyEval_Vector + 328
12  Python                        	       0x100b01854 call_function + 128
13  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
14  Python                        	       0x100af30ec _PyEval_Vector + 328
15  _objc.cpython-310-darwin.so   	       0x1015f5054 _PyObject_VectorcallTstate + 120
16  _objc.cpython-310-darwin.so   	       0x1015f4fd0 PyObject_Vectorcall + 60
17  _objc.cpython-310-darwin.so   	       0x1015f2f90 pysel_vectorcall + 440
18  Python                        	       0x100b01854 call_function + 128
19  Python                        	       0x100afc404 _PyEval_EvalFrameDefault + 32836
20  Python                        	       0x100af30ec _PyEval_Vector + 328
21  Python                        	       0x100a05c24 _PyObject_FastCallDictTstate + 208
22  Python                        	       0x100a7c630 slot_tp_init + 196
23  Python                        	       0x100a74968 type_call + 288
24  Python                        	       0x100a0636c _PyObject_Call + 128
25  Python                        	       0x100afc5ec _PyEval_EvalFrameDefault + 33324
26  Python                        	       0x100a1d5e0 gen_send_ex2 + 224
27  Python                        	       0x100af7888 _PyEval_EvalFrameDefault + 13512
28  Python                        	       0x100a1d5e0 gen_send_ex2 + 224
29  _asyncio.cpython-310-darwin.so	       0x100ee0a74 task_step_impl + 440
30  _asyncio.cpython-310-darwin.so	       0x100ee0848 task_step + 52
31  Python                        	       0x100a0594c _PyObject_MakeTpCall + 136
32  Python                        	       0x100b17d50 context_run + 92
33  Python                        	       0x100a58de8 cfunction_vectorcall_FASTCALL_KEYWORDS + 84
34  Python                        	       0x100afc5ec _PyEval_EvalFrameDefault + 33324
35  Python                        	       0x100af30ec _PyEval_Vector + 328
36  Python                        	       0x100b01854 call_function + 128
37  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
38  Python                        	       0x100af30ec _PyEval_Vector + 328
39  Python                        	       0x100b01854 call_function + 128
40  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
41  Python                        	       0x100af30ec _PyEval_Vector + 328
42  Python                        	       0x100b01854 call_function + 128
43  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
44  Python                        	       0x100af30ec _PyEval_Vector + 328
45  Python                        	       0x100b01854 call_function + 128
46  Python                        	       0x100afc384 _PyEval_EvalFrameDefault + 32708
47  Python                        	       0x100af30ec _PyEval_Vector + 328
48  Python                        	       0x100b01854 call_function + 128
49  Python                        	       0x100afc404 _PyEval_EvalFrameDefault + 32836
50  Python                        	       0x100af30ec _PyEval_Vector + 328
51  Python                        	       0x100af2f90 PyEval_EvalCode + 104
52  Python                        	       0x100b4c6cc run_eval_code_obj + 84
53  Python                        	       0x100b4c614 run_mod + 112
54  Python                        	       0x100b4c280 pyrun_file + 148
55  Python                        	       0x100b4bb94 _PyRun_SimpleFileObject + 268
56  Python                        	       0x100b4b1d4 _PyRun_AnyFileObject + 232
57  Python                        	       0x100b6d40c pymain_run_file_obj + 220
58  Python                        	       0x100b6cb5c pymain_run_file + 72
59  Python                        	       0x100b6c3cc Py_RunMain + 868
60  Python                        	       0x100b6d578 pymain_main + 36
61  Python                        	       0x100b6d7ec Py_BytesMain + 40
62  dyld                          	       0x1005650f4 start + 520

Thread 1 Crashed::  Dispatch queue: com.apple.root.default-qos
0   libsystem_kernel.dylib        	       0x1b4273eb8 __abort_with_payload + 8
1   libsystem_kernel.dylib        	       0x1b4276864 abort_with_payload_wrapper_internal + 104
2   libsystem_kernel.dylib        	       0x1b4276898 abort_with_payload + 16
3   TCC                           	       0x1b943a874 __TCC_CRASHING_DUE_TO_PRIVACY_VIOLATION__ + 172
4   TCC                           	       0x1b943b19c __TCCAccessRequest_block_invoke.194 + 600
5   TCC                           	       0x1b9438794 __tccd_send_message_block_invoke + 632
6   libxpc.dylib                  	       0x1b3fd99e8 _xpc_connection_reply_callout + 116
7   libxpc.dylib                  	       0x1b3fd98e0 _xpc_connection_call_reply_async + 88
8   libdispatch.dylib             	       0x1b40c6c2c _dispatch_client_callout3 + 20
9   libdispatch.dylib             	       0x1b40e4698 _dispatch_mach_msg_async_reply_invoke + 348
10  libdispatch.dylib             	       0x1b40d90c0 _dispatch_kevent_worker_thread + 1316
11  libsystem_pthread.dylib       	       0x1b428133c _pthread_wqthread + 344
12  libsystem_pthread.dylib       	       0x1b4280018 start_wqthread + 8

Thread 2:
0   libsystem_pthread.dylib       	       0x1b4280010 start_wqthread + 0
```

dlech, 2022-02-11:
This is not a problem in Bleak but but rather a problem with your terminal app. It does not have permission to use Bluetooth. 

To work around the problem, you can manually add the terminal app to the list of apps that allow Bluetooth in *System Preferences*. Screenshot: https://github.com/pybricks/support/discussions/583#discussioncomment-1933666. 

To fix the app itself, it needs to add the permission request to add `NSBluetoothAlwaysUsageDescription` to it's `info.plist` like this: https://github.com/gnachman/iTerm2/pull/457.

jseppanen, 2022-02-14:
Thanks so much for your help, yeah that was it, and now it works like a charm!

It's funny that Apple decided to send SIGABORT instead of a more user friendly error code/message that could have been  printed to the console

closed 2022-02-14 by jseppanen

dlech, 2022-02-14:
What terminal application are you using?