hbldh/bleak #768: macOS 12 Arm M1 MAX script aborts

pawelm87, 2022-02-17:
* bleak version: 0.12.1
* Python version: 3.9
* Operating System: macOS 12 Arm M1 MAX and macOS (probably version 12) standard
* BlueZ version (`bluetoothctl -v`) in case of Linux: -

### Description

I would like to run the program on the new architecture - macOS 12 ARM M1, unfortunately the program crashes. It works correctly on WIN10 and macOS Intel.

### What I Did

In order to locate the error I try to run an example from the repository - disconnect_callback.py. In the example I changed only the debug level (line 7):
```
import asyncio
from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
​
import logging
logging.basicConfig(format='[%(name)s - %(levelname)s] %(message)s', level=logging.DEBUG)
​
​
def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    print(device.address, "RSSI:", device.rssi, advertisement_data)
​
​
async def run():
    scanner = BleakScanner()
    scanner.register_detection_callback(simple_callback)
​
    while True:
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()
​
​
loop = asyncio.get_event_loop()
loop.run_until_complete(run())
```
Running the example on macOS ARM M1 (standard, not MAX) then I get the message:
```
username@user-name-osx Downloads % python3 /Users/username/Downloads/detect_callback.py
[asyncio - DEBUG] Using selector: KqueueSelector
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] centralManagerDidUpdateState_
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] Bluetooth powered on
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
...
```
The program supposedly works, but it does not find any device, even though there are a lot of devices with BLE around. I have seen similar problems that are solved by migrating to the latest version of the library, tomorrow i will update library, but...

I am now trying to run the same example on macOS 12 ARM M1 MAX:

[image]

In this case, the program crashes immediately. You can see that the program starts because the DEBUG log is displayed, but after a while it is aborted without any message. And it's strange, different behavior than before - I couldn't find a similar issues on the forum. 

I found on the forum that macOS 12 needs a library update - fix in version 0.14.0 and higher but script running on MAX version is aborted immediately ?

If I am duplicating the problem I apologize but I have limited access to these two machines for testing so I would like to make sure if it is the fault of the library version possibly try something new. Tomorrow I will update the library to the latest version and see if it worked.

dlech, 2022-02-18:
> * bleak version: 0.12.1

Does this problem happen with Bleak 0.14.2?

dlech, 2022-02-18:
> The program supposedly works, but it does not find any device,

Duplicate of #635 and #720.

dlech, 2022-02-18:
> In this case, the program crashes immediately.

Duplicate of #761. I suppose we must not have this in the troubleshooting section of the docs.

pawelm87, 2022-02-18:
```username@user-name-osx ~ % python3 /Users/username/Downloads/detection_callback.py 
[asyncio - DEBUG] Using selector: KqueueSelector
[root - WARNING] Scanning using all known service UUIDs to work around a macOS 12 bug. Some devices may not be detected. Please report this to Apple using the Feedback Assistant app and reference <https://github.com/hbldh/bleak/issues/635>.
Scanning using all known service UUIDs to work around a macOS 12 bug. Some devices may not be detected. Please report this to Apple using the Feedback Assistant app and reference <https://github.com/hbldh/bleak/issues/635>.
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] centralManagerDidUpdateState_
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] Bluetooth powered on
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] centralManager_didDiscoverPeripheral_advertisementData_RSSI_
[root - INFO] 2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -77, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -77, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] Discovered device 2E650724-81A3-D9EB-0155-ECBE570D8F8D: Amazfit GTR 2e @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataServiceData', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600002848180>
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
ć[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
^R
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] centralManager_didDiscoverPeripheral_advertisementData_RSSI_
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] centralManager_didDiscoverPeripheral_advertisementData_RSSI_
[root - INFO] 2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -77, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -77, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] Discovered device 2E650724-81A3-D9EB-0155-ECBE570D8F8D: Amazfit GTR 2e @ RSSI: -77 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataServiceData', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600002848180>
[root - INFO] 2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -76, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
2E650724-81A3-D9EB-0155-ECBE570D8F8D RSSI: -76, AdvertisementData(local_name='Amazfit GTR 2e', manufacturer_data={343: b'\x02\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\xff\x02\xc0\x0c\x0b\x13\x1bF'}, service_data={'0000fee0-0000-1000-8000-00805f9b34fb': b'\xf9\x04\x00\x00'}, service_uuids=['0000fee0-0000-1000-8000-00805f9b34fb'])
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] Discovered device 2E650724-81A3-D9EB-0155-ECBE570D8F8D: Amazfit GTR 2e @ RSSI: -76 (kCBAdvData <nsdict_keys(['kCBAdvDataManufacturerData', 'kCBAdvDataTimestamp', 'kCBAdvDataServiceData', 'kCBAdvDataRxPrimaryPHY', 'kCBAdvDataIsConnectable', 'kCBAdvDataRxSecondaryPHY', 'kCBAdvDataLocalName', 'kCBAdvDataServiceUUIDs'])>) and Central: <CBCentralManager: 0x600002848180>
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[bleak.backends.corebluetooth.CentralManagerDelegate - DEBUG] 'isScanning' changed
[... cut ...]
username@user-name-osx ~ % pip3 show bleak
Name: bleak
Version: 0.14.2
Summary: Bluetooth Low Energy platform Agnostic Klient
Home-page: https://github.com/hbldh/bleak
Author: Henrik Blidh
Author-email: henrik.blidh@nedomkull.com
License: MIT
Location: /Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages
Requires: pyobjc-framework-libdispatch, pyobjc-core, pyobjc-framework-CoreBluetooth
Required-by: 
```

macOS 12.2.1, i see that macos12.3 is a beta version not official released :(

pawelm87, 2022-02-18:
```
"""
Detection callback w/ scanner
--------------
Example showing what is returned using the callback upon detection functionality
Updated on 2020-10-11 by bernstern <bernie@allthenticate.net>
"""

import asyncio
import logging
import platform
import sys

from bleak import BleakScanner
from bleak.backends.device import BLEDevice
from bleak.backends.scanner import AdvertisementData
from bleak.uuids import uuid16_dict, uuid128_dict

# logger = logging.getLogger(__name__)
logging.basicConfig(format='[%(name)s - %(levelname)s] %(message)s', level=logging.DEBUG)

def simple_callback(device: BLEDevice, advertisement_data: AdvertisementData):
    logging.info(f"{device.address} RSSI: {device.rssi}, {advertisement_data}")
    print(f"{device.address} RSSI: {device.rssi}, {advertisement_data}")

async def main(service_uuids):
    mac_ver = platform.mac_ver()[0].split(".")
    if mac_ver[0] and int(mac_ver[0]) >= 12 and not service_uuids:
        # In macOS 12 Monterey the service_uuids need to be specified. As a
        # workaround for this example program, we scan for all known UUIDs to
        # increse the chance of at least something showing up. However, in a
        # "real" program, only the device-specific advertised UUID should be
        # used. Devices that don't advertize at least one service UUID cannot
        # currently be detected.
        logging.warning(
            "Scanning using all known service UUIDs to work around a macOS 12 bug. Some devices may not be detected. Please report this to Apple using the Feedback Assistant app and reference <https://github.com/hbldh/bleak/issues/635>."
        )
        print("Scanning using all known service UUIDs to work around a macOS 12 bug. Some devices may not be detected. Please report this to Apple using the Feedback Assistant app and reference <https://github.com/hbldh/bleak/issues/635>.")
        for item in uuid16_dict:
            service_uuids.append("{0:04x}".format(item))
        service_uuids.extend(uuid128_dict.keys())
    scanner = BleakScanner(service_uuids=service_uuids)
    scanner.register_detection_callback(simple_callback)

    while True:
        await scanner.start()
        await asyncio.sleep(5.0)
        await scanner.stop()

if __name__ == "__main__":
    logging.basicConfig(
        level=logging.INFO,
        format="%(asctime)-15s %(name)-8s %(levelname)s: %(message)s",
    )
    service_uuids = sys.argv[1:]
    asyncio.run(main(service_uuids))
```

closed 2022-03-05 by dlech

tmack8001, 2023-02-02:
I have this exact same issue, but am running with latest iTerm on the latest MacOS M1 13.2 AND have granted permissions to iTerm to access bluetooth... Similarly tests that were running in Github Actions on a MacOS image are now failing (no code change, likely an OS update as I have it configured to run against MacOS-latest).

Commenting out `devices = await BleakScanner.discover(10.0)` obviously makes it so this `illegal hardware instruction` go away, but that isn't ideal...

Have you noticed this with anyone else?

[image]

imirzadeh, 2023-02-11:
> I have this exact same issue, but am running with latest iTerm on the latest MacOS M1 13.2 AND have granted permissions to iTerm to access bluetooth... Similarly tests that were running in Github [...]

I'm using Python 3.11.1 on M1 (Ventura 13.1) and the sample code runs for me.

commit ee597d3a (2022-03-05): docs/troubleshooting: common macOS issues

This documents a couple of common macOS issues.

Fixes #768
PR #779 (2022-03-05): docs/troubleshooting: common macOS issues
This documents a couple of common macOS issues.

Fixes #768