hbldh/bleak #774: Get empty array when using "read_gatt_char"

MariaCrespoEchevarria, 2022-02-28:
* bleak version: 0.14.2
* Python version: 3.8.5
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description
Hello,
I am using the bleak library to make a connection with a device. The aim is to write a command using write_gatt_char to the device and get the response using read_gatt_char. I already manage to write the commands and it works. However, when I tried to read the answer , even if I dont get an error, I obtain an empty array as a response (model_number).
### What I Did
Here you have my code

```
import asyncio
import struct
from bleak import BleakScanner
from bleak import BleakClient
from bleak import BleakError
import nest_asyncio
nest_asyncio.apply()

async def run():
    devices = await BleakScanner.discover()
    for d in devices:
        print(d)
asyncio.run(run())

#CONNECT TO DEVICE AND
#ASK CHARACTERISTICS AND WRITE THEM
adress  = "CF:34:4E:3D:D9:64"
read_characteristic = "6E400003-B5A3-F393-E0A9-E50E24DCCA9E"
write_characteristic = "6E400002-B5A3-F393-E0A9-E50E24DCCA9E"

output_list = [";0"]
final = bytearray(''.join(output_list), encoding='utf8')
 
async def main(address):
    print("Connecting to device...")
    async with BleakClient(address) as client:
        print("Connected")
        for x in range(10):
            await client.write_gatt_char(write_characteristic, final)
            print(f"Sent: {output_list}")
            model_number = await client.read_gatt_char(read_characteristic)
asyncio.run(main(adress))
```

dlech, 2022-02-28:
I would try logging Bluetooth packets to see if the device really is sending an empty response.

ps2229, 2022-04-07:
were you able to solve the read issue?

dlech, 2022-10-03:
closing due to lack of feedback (logs)

closed 2022-10-03 by dlech