hbldh/bleak #776: [Packaging] conda-forge Package

pietakio, 2022-03-03:
Greetings from the snow-encrusted ice fields of Canada, wonderful `bleak` devs! This is a packaging request for a [conda-forge](https://conda-forge.org)-hosted `bleak` package.

Although [`bleak` is *technically* installable under (Ana|Mini)conda by deferring to `pip`](https://halfdonebutinthemaking.com/lego-boost-with-python), `conda`'s `pip` integration is [second-class at best and promotes installation-time issues with dependency resolution](https://stackoverflow.com/a/65098284/5049231). First-class `conda` support via conda-forge would be vastly preferable.

Thankfully, doing so is trivial! Our team maintains a suite of open-source conda-forge recipes (e.g., [beartype](https://github.com/conda-forge/beartype-feedstock), [svgpathtools](https://github.com/conda-forge/svgpathtools-feedstock)). We'd be happy to get `bleak` up and running with conda-forge as well if you'd prefer that. Alternately, we'd be happy to supply you with a trivial `recipe.yml` configuration that *should* suffice.

In any case, thanks for all the tremendous volunteerism! :1st_place_medal:

dlech, 2022-03-03:
Is it just a matter of adding a `recipe.yml` file to this repo or do we have to publish the package to conda-forge like we do to PyPI?

leycec, 2022-03-04:
Excellent questions one and all, @dlech. I'm our conda-forge go-to guy, so let's tackle this one deliverable at a time.

> ...or do we have to publish the package to conda-forge like we do to PyPI?

More this – except "publish the package to conda-forge" basically means:

1. [Fork the GitHub-hosted `conda-forge/staged-recipes` repo](https://github.com/conda-forge/staged-recipes).
2. Add your `recipe.yml` to that fork.
3. Open a PR that's usually merged within hours by a dedicated cadre of volunteers.

GitHub Actions automation then does everything else for you on your behalf – including the heavy lifting of actually:

* Publishing a `bleak` package to [Anaconda's `conda-forge` channel](https://anaconda.org/conda-forge).
* Bumping that package whenever it detects a new version of the `bleak` package pushed to PyPI.

Your PyPI release process continues to run the show. `conda-forge` parasitically feeds on that process in the background by silently pushing out new Anaconda versions of your package for each new PyPI version of your package.

What could possibly go wrong? <sup>:wink:</sup>

## A Positive Vision of the Future

If you've read this far, you're at least mildly interested. Before someone (who might be me) gets that ball rolling, however, someone (who might be me) will need to get the same ball rolling for your mandatory dependencies. Makes sense, right? Can't add a `bleak` package without first adding the packages that `bleak` depends on.

I'm afraid we can't help with your macOS dependencies (e.g., `pyobjc-*`) or Windows dependencies (i.e., `bleak-winrt`), because we're Linux all the way down. We'd be happy to help with your sole Linux dependency (i.e., `dbus-next`), however.

Specifically, we'd be happy to package `dbus-next` onto conda-forge with the same process as outlined above. Once `dbus-next` is up, we march onto `bleak` itself.

I should note that I've [already packaged both `dbus-next` *and* `bleak` for Gentoo Linux](https://github.com/leycec/raiagent). I'm intimately familiar with your packaging needs. Thankfully, they're minimal! Both packages were trivial to get up and running. I foresee no significant roadblocks to progress, except...

## There's Always a Catch

Since macOS and Windows is a bit out-of-scope for us, `bleak` will only initially be installable for Linux via conda-forge. That's the catch.

Still... something's better than nothing. The perfect is the enemy of the good, etcetera. MacOS and Windows dependencies can be incrementally packaged by someone more familiar with those platforms at a later date.

If you're amenable, I'll leap straight-away onto packaging `dbus-next` for conda-forge. We can rendezvous from there.

dlech, 2022-03-04:
I don't have any objections if someone else is going to set it all up and the maintenance is largely automated.

leycec, 2022-03-05:
Our thoughts exactly. :wink: 

We'll get this party started, then. I'll ping you onto the eventual `conda-forge` PR that officially packages `bleak` when it lands in a week or two. Thanks for being so considerate, @dlech.

dlech, 2023-07-19:
I don't think we need to leave this issue open, but happy to accept a PR if there is still interest here.

closed 2023-07-19 by dlech