hbldh/bleak #781: Bluetooth Classic support

JJTech0130, 2022-03-07:
### Description

[PyBluez](https://github.com/pybluez/pybluez) has been unmaintained for a while now. In addition, it uses C bindings that require Visual Studio to compile on Windows.

I'm asking if it might be possible to implement (as a part of `bleak`, or separately) a cross-platform, pure Python method for connecting to Bluetooth Classic, especially Bluetooth HID, devices.

This would aid me in one of my projects, in which I would like to use Nintendo Wiimotes in a Python application. I have not found a solution that would let me do this.

If I've misunderstood something, feel free to tell me. (I'm pretty sure Bluetooth HID is a part of Bluetooth Classic... I didn't see them in a `bleak` scan...) I've seen a mention of Bluetooth Classic in #405, but I think that ended up being unrelated?

I'd be willing to help work on this if someone could give me some direction.

dlech, 2022-03-07:
Bleak is explicitly BLE-only, no classic support.

Python has native support for Bluetooth Classic RFCOMM sockets on Windows, Linux and Mac (Windows was added in Python 3.9). 

For Bluetooth HID devices, the OS/BlueZ should handle the Bluetooth side of things and present it as an input device, so you would want something like evdev on Linux rather than using a Bluetooth library. According to https://wiki.archlinux.org/title/XWiimote, the `hid-wiimote` driver has existed since Linux kernel 3.1.

> You can connect to your Wii Remote like any other [Bluetooth](https://wiki.archlinux.org/title/Bluetooth) device. See Bluetooth for information on pairing Bluetooth devices. The Wii Remote does not need special handling anymore. The BlueZ wiimote plugin handles all peculiarities in the background for you.

JJTech0130, 2022-03-07:
That doesn't handle the Windows side of things. And while I use Arch on my main development machine, this is really a project for casual users, most of which will be on Windows.

JJTech0130, 2022-03-07:
I didn't know about Python's native Bluetooth support. I don't see a way to scan for devices, though— do you know if this is possible?

dlech, 2022-03-07:
> That doesn't handle the Windows side of things. And while I use Arch on my main development machine, this is really a project for casual users, most of which will be on Windows.

It looks like someone made a wiimote driver for Windows as well. https://github.com/jloehr/HID-Wiimote

> I didn't know about Python's native Bluetooth support. I don't see a way to scan for devices, though— do you know if this is possible?

The native support is only for RFCOMM sockets. So the Bluetooth device must be paired and connected before the socket can be created.

JJTech0130, 2022-03-07:
> It looks like someone made a wiimote driver for Windows as well. https://github.com/jloehr/HID-Wiimote

I tried that, but I couldn't get it to work. I also feel a like it adds a lot of complication to add a system driver for a simple project, a lot seems like it could go wrong 😄

JJTech0130, 2022-03-07:
> The native support is only for RFCOMM sockets. So the Bluetooth device must be paired and connected before the socket can be created.

Ah well. That's a bit disappointing ☹️.

closed 2022-04-29 by dlech