hbldh/bleak #785: Using  BleakClient's unpair() causes crash

andraskalmar, 2022-03-15:
* bleak version: 0.14.2
* Python version: 3.10.1
* Operating System: Windows 11 Pro, version: 21H2
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

My program connects to a device writes/reads some characteristics and then disconnects. After some seconds the procedure repeats. 
I am getting the following error message after running the application for some minutes, if I am using the BleakClient's unpair() function before disconnection - actually this function causes disconnection itself. If the the unpair() function is not called just the disconnect() the issue does not pop up.

```
Exception has occurred: CancelledError
exception: no description
  File "C:\Users\andras.kalmar\BLE_assistant\ccis-assistant-pythonble-33867\src\main.py", line 90, in scanner_loop
    success = await bleClient.connect(unit_id=key)
  File "C:\Users\andras.kalmar\BLE_assistant\ccis-assistant-pythonble-33867\src\main\tssbleclient\tss_ble_client.py", line 634, in connect
    await self.ble_client.connect(timeout=self.CONNECT_TIMEOUT)
  File "C:\Users\andras.kalmar\AppData\Roaming\Python\Python310\site-packages\bleak\backends\winrt\client.py", line 276, in connect
    await self.get_services()
  File "C:\Users\andras.kalmar\AppData\Roaming\Python\Python310\site-packages\bleak\backends\winrt\client.py", line 467, in get_services
    await self._requester.get_gatt_services_async(

During handling of the above exception, another exception occurred:

  File "C:\Program Files\Python_3_1_0\Lib\asyncio\base_events.py", line 641, in run_until_complete
    return future.result()
  File "C:\Program Files\Python_3_1_0\Lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\andras.kalmar\BLE_assistant\ccis-assistant-pythonble-33867\src\main.py", line 57, in scanner_loop_async
    asyncio.run(scanner_loop())
  File "C:\Users\andras.kalmar\BLE_assistant\ccis-assistant-pythonble-33867\src\main.py", line 287, in <lambda>
    threading.Thread(target=lambda: scanner_loop_async()).start()
  File "C:\Program Files\Python_3_1_0\Lib\threading.py", line 946, in run
    self._target(*self._args, **self._kwargs)
  File "C:\Program Files\Python_3_1_0\Lib\threading.py", line 1009, in _bootstrap_inner
    self.run()
  File "C:\Program Files\Python_3_1_0\Lib\threading.py", line 966, in _bootstrap
    self._bootstrap_inner()
```

andraskalmar, 2022-03-15:
Here is a stripped of example code.

The issue pops up typically after some minutes, however it might take 10-20 mins.
My device advertising with an 200 ms interval.

```
import bleak
from bleak import *
import asyncio
import threading
import time

class BleClient:

    ble_client = None
    INDICATION_CHARACTERISTIC_UUID = "69b5bbbb-a47f-4b31-9a4d-fd459169dd0f"

    def disconnect_callback(self, client):
            logging.info(f"Client with address {client.address} got disconnected!")
            print(f"Client with address {client.address} got disconnected!")
            self.is_disconnected = True

    def indication_handler(self, sender, data):
        logging.info(f"Indication received: {sender}: {bytes(data).hex()}")
        print(f"Indication received: {sender}: {bytes(data).hex()}")
        self.indicated_ble_message += data.hex()

    async def connect(self, address):
            try:
                self.ble_client = BleakClient(address)

            except Exception as e:
                    logging.error(f"Error: {repr(e)}")
                    print(str(e))
                    return False

            tic1 = time.perf_counter()

            for i in range(5):

                tic = time.perf_counter()
                logging.info(f"Connection attempt no: {i+1}")

                try:
                    await self.ble_client.connect(timeout=100)

                except asyncio.TimeoutError as e:
                    logging.info(f"Timeout when trying to connect: {repr(e)}")
                    toc = time.perf_counter()
                    logging.info(f"Attempted connection took {toc - tic:0.4f} seconds")
                    continue

                except BleakError as e:
                    logging.error(f"BleakError: {repr(e)}")
                    continue

                except Exception as e:
                    logging.error(f"Error: {repr(e)}")
                    print(str(e))
                    return False
                
                if self.ble_client.is_connected():
                    break

            
            try:
                logging.info("Connected to logger.")
                print("Connected to logger.")

                toc2 = time.perf_counter()
                logging.info(f"Connection took {toc2 - tic1:0.4f} seconds")
                print(f"Connection took {toc2 - tic1:0.4f} seconds")

                await self.pair_device(self.ble_client)

                self.is_disconnected = False
                self.ble_client.set_disconnected_callback(self.disconnect_callback)

                await self.start_notify_single(self.ble_client)
                print("Notification started")

                
            except asyncio.TimeoutError as e:
                logging.info(f"Timeout when trying to connect: {repr(e)}")
                toc = time.perf_counter()
                logging.info(f"Attempted connection took {toc - tic:0.4f} seconds")
                return False

            except BleakError as e:
                logging.error(f"BleakError: {repr(e)}")
                return False
                
            except Exception as e:
                    logging.error(f"Error: {repr(e)}")
                    print(str(e))
                    return False

            if(not self.is_disconnected):
                # We are ready to interact with the client
                return True
            else:
                return False

    async def disconnect(self):
            await self.ble_client.stop_notify(self.INDICATION_CHARACTERISTIC_UUID)

            # Unpairing should cause disconnection
            success = await self.ble_client.unpair()
            print(f"Unpairing was {success}")
            logging.info(f"Unpairing was {success}")
            
            await self.ble_client.disconnect()
            self.is_disconnected = True

    async def pair_device(self, client):
        try:
            await client.pair(0)
        except Exception as e:
            logging.error(f"Error when pairing: {repr(e)}")

    async def start_notify_single(self, client):
        try:
            await client.start_notify(self.INDICATION_CHARACTERISTIC_UUID, self.indication_handler)

        except Exception as e:
            logging.error(f"Failed to start notify: {repr(e)}")

ADDRESS = (
    "D6:93:C4:07:5C:0F" 
)

async def find_device(address):
    device = await BleakScanner.find_device_by_address(address)
    print(device)
    return device

def scanner_loop_async():
    asyncio.run(scanner_loop())

async def scanner_loop():

    device = await find_device(ADDRESS)

    while(True):
        success = await bleClient.connect(device)
        if(not success):
            continue

        await bleClient.disconnect()
        

if __name__ == "__main__":
    bleClient = BleClient()

    if platform.system()=='Windows':
        asyncio.set_event_loop_policy(asyncio.WindowsSelectorEventLoopPolicy())

    threading.Thread(target=lambda: scanner_loop_async()).start()

```

andraskalmar, 2022-03-22:
An update here: with the example above I could reproduce the same problem without using the unpair method, i.e., just using the disconnect function to disconnect. 

Any update/input from your side?

dlech, 2022-03-22:
Constantly connecting and disconnecting without any delay seems like it is asking for trouble in the OS Bluetooth stack. We also don't recommend reusing the same BleakClient object after disconnecting since it will be in an inconsistent state.

andraskalmar, 2022-03-25:
I think I have found the solution for this problem.

The problem was that ```asyncio.CancelledError``` is not a subclass of ```Exception``` but a subclass of ```BaseException```, thus the exception was not handled. I added ```except asyncio.CancelledError as e:``` into the exception handlings and the program continued to work after the exception happened.

I think it could be useful to add this into the Bleak examples.

closed 2022-03-28 by andraskalmar