hbldh/bleak #786: write_gatt_char not working as expected on MacOS

ryan-moreno, 2022-03-16:
* bleak version: 0.14.2
* Python version: 3.7.4
* Operating System: MacOS Monterey 12.3
* BlueZ version (`bluetoothctl -v`) in case of Linux: NA

### Description

I have a python program that interfaces over bluetooth with a Polar H10 device. I receive heart rate data and also pmd data (which we have to request using the pmd control). There is a minimal workable example at the bottom of this report.

My expectation is that I would receive both heart rate and pmd data. This is what occurs when I run the example below on Windows. However, on MacOS I only receive heart rate data. 

I used Bluetility to manually set the pmd control on MacOS, and when I do that this program receives both heart rate and pmd data as expected. If I refresh Bluetility then the program no longer receives pmd data. This leads me to believe that write_gatt_char is not working as expected on MacOS since there is no issue reading the pmd data once the pmd control is set.

This screenshot shows how I set the pmd control using Bluetility (I wrote the hex value 02000001820001010E00 which requests ECG data)

[image]

### Minimum Workable Example

```
import bleak
import asyncio

uuid16_dict = {v: k for k, v in bleak.uuids.uuid16_dict.items()}

# UUID for heart rate
HR_UUID = "0000{0:x}-0000-1000-8000-00805f9b34fb".format(
    uuid16_dict.get("Heart Rate Measurement")
)

# UUID for data stream settings
PMD_CONTROL = "FB005C81-02E7-F387-1CAD-8ACD2D8DF0C8"

# UUID for data stream
PMD_DATA = "FB005C82-02E7-F387-1CAD-8ACD2D8DF0C8"

# UUID Request for ECG Stream
ECG_WRITE = bytearray([0x02, 0x00, 0x00, 0x01, 0x82, 0x00, 0x01, 0x01, 0x0E, 0x00])

def handle_hr_data(sender, data):
    print("Received hr data: " + str(data))

def handle_pmd_data(sender, data):
    print("Received pmd data: " + str(data))

# searches for Polar H10 for up to 10 seconds and returns address if found
async def get_polar_h10_address():
    async with bleak.BleakScanner() as scanner:
        for i in range(10):
            await asyncio.sleep(1.0)
            for d in scanner.discovered_devices:
                if d.name is not None and d.name[:9] == "Polar H10":
                    return d.address

async def main():
    client = None

    try:
        # find and connect to polar H10 device
        ADDRESS = await get_polar_h10_address()
        if ADDRESS is None:
            print("issue finding device")
            return
        client = bleak.BleakClient(ADDRESS)
        if not await client.connect():
            print("issue connecting to device")
            return

        # request ECG data from the client
        await client.write_gatt_char(PMD_CONTROL, ECG_WRITE)

        # start PMD data stream
        await client.start_notify(PMD_DATA, handle_pmd_data)

        # start HR data stream (which is not broadcast through PMD)
        await client.start_notify(HR_UUID, handle_hr_data)

        # sleep for 10 seconds while data is being received
        await asyncio.sleep(10)

        # stop data streams and disconnect from client
        # This stops the service
        await client.stop_notify(PMD_DATA)
        await client.stop_notify(HR_UUID)
        await client.disconnect()

    finally:
        if client is not None:
            await client.disconnect() # need to disconnect or else messes with OS bluetooth cache

if __name__ == "__main__":
    loop = asyncio.new_event_loop()
    asyncio.set_event_loop(loop)
    loop.run_until_complete(main())
```

**Expectation (and what happens on Windows and on MacOS if I use Bluetility to manually set the pmd control):**
```
$ python bleak_issue_mwe.py
Received pmd data: bytearray(b'\x00\x0c\x93\xf3p')
Received hr data: bytearray(b'\x16T\xe2\x02')
Received pmd data: bytearray(b'\x00>\x88i\x92')
Received pmd data: bytearray(b'\x00p}\xdf\xb3')
Received hr data: bytearray(b'\x16U\xdd\x02')
Received pmd data: bytearray(b'\x00\xa2rU\xd5')
Received hr data: bytearray(b'\x16T\xf9\x02\xf9')
Received pmd data: bytearray(b'\x00\xd4g\xcb\xf6')
Received pmd data: bytearray(b'\x00\x06]A\x18')
Received hr data: bytearray(b'\x16T\xf2\x02')
Received pmd data: bytearray(b'\x00Y>\xb79')
Received pmd data: bytearray(b'\x004\xd0,[')
Received hr data: bytearray(b'\x16S\x12\x03')
Received pmd data: bytearray(b'\x00f\xc5\xa2|')
Received pmd data: bytearray(b'\x00\x1ck\x18\x9e')
Received hr data: bytearray(b'\x16R\x12\x03\xfe')
Received pmd data: bytearray(b'\x00\xca\xaf\x8e\xbf')
Received pmd data: bytearray(b'\x00\xfc\xa4\x04\xe1')
Received hr data: bytearray(b'\x16R\xeb\x02')
Received pmd data: bytearray(b'\x00\xd66z\x02')
Received hr data: bytearray(b'\x16R\xda\x02\xe5')
Received pmd data: bytearray(b'\x00*\x18\xf0#')
Received pmd data: bytearray(b'\x00\\\rfE')
Received hr data: bytearray(b'\x16R\xe7\x02')
Received pmd data: bytearray(b'\x00\x8e\x02\xdcf')
Received pmd data: bytearray(b'\x00\xc0\xf7Q\x88')
Received hr data: bytearray(b'\x16R\xee\x02')
Received pmd data: bytearray(b'\x00\xf2\xec\xc7\xa9')
```

**Reality:**
```
python bleak_issue_mwe.py
Received hr data: bytearray(b'\x10W\xca\x02')
Received hr data: bytearray(b'\x10V\xdc\x02\xe6')
Received hr data: bytearray(b'\x10U\xf8\x02\xdf')
Received hr data: bytearray(b'\x10U\xaf\x02')
Received hr data: bytearray(b'\x10Vz\x02o')
Received hr data: bytearray(b'\x10X}\x02')
Received hr data: bytearray(b'\x10Y\xa2\x02')
Received hr data: bytearray(b'\x10Y\xd7\x02\n')
Received hr data: bytearray(b'\x10X!\x03')
Received hr data: bytearray(b'\x10W\x16\x03')
```

ryan-moreno, 2022-03-17:
Good idea. I used packetlogger to capture the packets. I captured a Write Request with the data as expected when I write via Bluetility. There was no write request sent when I wrote via my python program using bleak. I did capture other write requests from python/bleak (namely, from client.start_notify). However, nothing was sent for client.write_gatt_char. Here is what the packet sent from Bluetility looks like 
[image]

dlech, 2022-03-17:
By default, Bleak does write without response, which is a Write Command rather than a Write Request. You could try adding a `response=True` argument to the `write_gatt_char()` call.

ryan-moreno, 2022-03-17:
That fixed it, thank you for the tip. The strange thing is, without "response=True" I didn't see any kind of write command sent in the logger either. Thank you for the info!