hbldh/bleak #787: still connecting to a BLE device

ghazijallouli, 2022-03-16:
* bleak version: latest
* Python version: latest
* Operating System: windows
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

i want to connect to a BLE device just one time and still connected, then i do a FW update

### What I Did

i wrote a class named OtaFwUpadate and has an attribute named client a i wrote a method named connexion, i need to get sure that, when i create a new object: ota=OtaFwUpdate() and do ota.connexion() i'm still connected to the device to do ota.send()

```
from asyncore import loop

from bleak import BleakClient, BleakScanner
from pathlib import Path
import time

class OtaFWUpdate:

    def __init__(self, deviceName, characteristicName, firmwarePath, chunkLength, delay):
        self.deviceName = deviceName
        self.characteristicName = characteristicName
        self.firmwarePath = firmwarePath
        self.chunkLength = chunkLength
        self.delay = delay
        self.address = None
        self.client = None
        self.chrcuuid = None

    def findaddress(self):
        found = False
        tentative = 1
        while not found:
            devices = await BleakScanner.discover(timeout=8)
            i = 0
            for d in devices:
                i = i + 1
                print("device {0} name is  '{1}'".format(i, d.name))
                if (d.name is not None) and (d.name.lower() == self.deviceName.lower()):
                    self.address = d.address
                    found = True
            if not found:
                print("device not found in tentative  {0} \n\n ".format(tentative))
                tentative = tentative + 1

    def connexion(self):
        self.client = BleakClient(self.address, loop=loop)

    def findcharacteristic(self):
        svcs = await self.client.get_services()
        for service in svcs:
            for chrc in service.characteristics:
                if chrc.description.lower() == self.characteristicName.lower():
                    if "write" in chrc.properties:
                        self.chrcuuid = chrc.uuid
                    else:
                        print("there is no 'write' property")

    def send(self):
        i = 0
        y = 1
        file_size = Path(self.firmwarePath).stat().st_size
        if (file_size % self.chunkLength) == 0:
            print(file_size / self.chunkLength)
            print("file size is normal")
            with open(self.firmwarePath, "rb") as firmware:
                data = firmware.read()
                for nb in range(file_size // self.chunkLength):
                    part = slice(i, i + self.chunkLength)
                    packet = data[part]
                    print("sending packet NB '{0}'".format(y))
                    await self.client.write_gatt_char(self.chrcuuid, packet)
                    print("packet '{0}' has been sent".format(y))
                    i = i + self.chunkLength
                    y = y + 1
                print("FW UPDATE is over")

        else:
            print(file_size // self.chunkLength)
            print("complete the file")
            with open(self.firmwarePath, "rb") as firmware:
                data = firmware.read()
                for index in range(self.chunkLength - (file_size % self.chunkLength)):
                    data += bytes((255,))
                for nb in range((file_size // self.chunkLength) + 1):
                    part = slice(i, i + self.chunkLength)
                    packet = data[part]
                    print(type(self.chrcuuid))
                    print(packet)
                    print("sending packet NB '{0}'".format(y))
                    await self.client.write_gatt_char(self.chrcuuid, packet)
                    print("packet '{0}' has been sent".format(y))
                    time.sleep(self.delay)
                    i = i + self.chunkLength
                    y = y + 1
                print("FW UPDATE is over")
```

ghazijallouli, 2022-03-16:
@hbldh

hbldh, 2022-03-16:
I am not sure what the issue is. Moving this to discussion.

You have to somewhere call `connect` and later on `disconnect` as well, but apart from that, it might work.