hbldh/bleak #790: Adding the scan.pattern filter, introduced in Bluez 5.54

ayemiller, 2022-03-22:
* bleak version: 0.14.2
* Python version: 3.9
* Operating System: Debian
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.55

### Description

Would you be open to a PR that would introduce the scan.pattern filter that was introduced in Bluez 5.54. I'm interested in offloading some of the responsibility of MAC address filtering onto Bluez rather than on my Python code.

### What I Did

Using bluetoothctl, by entering the following commands, you can see the type of functionality that I am interested in
```
bluetoothctl
menu scan
pattern AB:12 # only discover devices with a MAC address starting w/ AB:12
back
scan on
```

dlech, 2022-03-22:
I'm pretty sure we can do this already using the `filters` keyword arg.

ayemiller, 2022-03-22:
> I'm pretty sure we can do this already using the `filters` keyword arg.

I don't believe so. The [bluezdbus scanner](https://github.com/hbldh/bleak/blob/deae92b68e6af755056fd6854d88f2cf9bac3041/bleak/backends/bluezdbus/scanner.py#L205) supports filtering on UUIDs, RSSI, DuplicateData, Pathloss and Transport. Unless you're referring to something else?

dlech, 2022-03-22:
I see what you mean. It looks like we should add `Discoverable` as well. Happy to take a PR to add these.

ayemiller, 2022-03-22:
PR up - https://github.com/hbldh/bleak/pull/791

PR #818 (2022-04-29): Release/0.14.3

Changed
-------

* Suppress macOS 12 scanner bug error message for macOS 12.3 and higher. Fixes #720.
* Added filters ``Discoverable`` and ``Pattern`` to BlueZ D-Bus scanner. Fixes #790.

Fixed
-----

* Fixed reading the battery level returns a zero-filled ``bytearray`` on BlueZ >= 5.48. Fixes #750.
* Fixed unpairing does not work on windows with WinRT. Fixes #699
* Fixed leak of ``_disconnect_futures`` in ``CentralManagerDelegate``.
* Fixed callback not removed from ``_disconnect_callbacks`` on disconnect in ``CentralManagerDelegate``.