hbldh/bleak #794: Sometimes suddenly interrupted to receive can not continue to receive data.

allrobot, 2022-03-27:
* bleak version:0.14.2
* Python version:3.7
* Operating System:Windows10

### Description

[image]

Printing is slow, bluetooth output rate of 36000 characters /s(After the ESP32 devices were paired, more than 1,000 samples were sent per second and a sample is 36 characters.), windows10 can only receive a small part of the data, sometimes suddenly interrupted to receive can not continue to receive data.

[image]

How to solve it?

### What I Did

Try to reduce the sampling rate of the device to 500 samples per second, so that the data is collected many times without interruption.

However, this does not meet personal requirements. How can I ensure that the collection will not stop？

dlech, 2022-03-28:
I would suggest logging Bluetooth packets (see Bleak troubleshooting docs) to see if the Bluetooth chip on the computer is actually receiving the info. My guess is that it is not. BLE typically has a connection interval of about 30 to 50 milliseconds, so it can only send 20 to 30 packets per second. HID devices can have the fastest interval of 15 milliseconds (depending on the configuration) which would be 66 packets per second. The maximum MTU is for most devices is 512 bytes, so this would give 66 * 512 = approx 34000 bytes per second, which is slightly less than what you have measured. So perhaps the buffer on the ESP32 is overflowing from trying to send data faster than what is possible.

allrobot, 2022-03-28:
> I would suggest logging Bluetooth packets (see Bleak troubleshooting docs) to see if the Bluetooth chip on the computer is actually receiving the info. My guess is that it is not. BLE typically has [...]

Very your advice, I'm not used to Wireshark capturing Bluetooth data, but thankfully it was my Bluetooth that stopped receiving packets, so I relied on it to find my code problems.

closed 2022-03-28 by allrobot