hbldh/bleak #798: Don't use bluetoothctl

kq98, 2022-04-06:
* bleak version: 0.14.2
* Python version: 3
* Operating System: GNU/Linux
* BlueZ version: 5.60

### Description

Trying to use bleak inside a Flatpak sandbox fails because bluetoothctl, like all other system utils, can't be accessed.

Inside the bluezdbus backend in `__init__.py` a bluetoothctl subprocess is spawned to get the current bluez version. 
I suggest to use the bluez dbus interface directly to get the version.

hbldh, 2022-04-06:
I am pretty sure there is no such thing available in the dbus api for BlueZ. At least there wasn't when Bleak was written.

It is regrettable that bluetoothctl is needed there, and I would like to remove it if possible.

Simplest solution for you would be to modify the init file to have your version hard coded, for this very specific use case.

dlech, 2023-03-18:
This was fixed by #907 by not crashing if `bluetoothctl` is not available.

closed 2023-03-18 by dlech