hbldh/bleak #799: Failure to enable notifications through writing to a descriptor (SensorTag)

OlafSimon, 2022-04-07:
* bleak version: latest
* Python version: 3.8.0
* Operating System: Raspian / Linux
* BlueZ version: 5.50

In order to activate notifications at the Texas Instruments SensorTag it is required to write to the descriptor '2920' of the desired characteristic to let it emit notifications.

I used the key pressed characteristic 0000ffe1-0000-1000-8000-00805f9b34fb (handle 0x4c). Notifications are enabled by writing 0x0100 to the descriptor 00002902-0000-1000-8000-00805f9b34fb (handle 0x4d). Works fine with gatttool and bluetoothctl (see TestBleakSensorTag.py).

I failed to do so by using bleak. The write_gatt_descriptor() method of the BleakClient failed (BleakDBusError: write not permitted). See 'prog2()' within TestBleakSensorTag.py.
[TestBleakSensorTag.txt](https://github.com/hbldh/bleak/files/8444728/TestBleakSensorTag.txt)

The attached ExtBleakClient contains a solution by introducing a new method 'enableNotification' solving that problem (although just for Linux hardware so far).

Please, consider to taking over this feature. If yes, I can support by placing the implementation to the right spots within the library following the hardware independent implementation. 

Olaf

Additional required files are:
[dbusFunction.txt](https://github.com/hbldh/bleak/files/8444726/dbusFunction.txt)
[ExtBleakClient.txt](https://github.com/hbldh/bleak/files/8444727/ExtBleakClient.txt)

dlech, 2022-04-07:
Please see https://bleak.readthedocs.io/en/latest/api.html#bleak.backends.client.BaseBleakClient.start_notify for the method to enable notifications. There is more to it behind the scenes than just writing the descriptor.

OlafSimon, 2022-04-07:
Thanks for the very fast response. I am using start_notify() but that is not sufficient for the sensor tag. start_notify() registers a callback and this is called in case a notification message is incoming. But the sensor tag does not emit notifications unless you explicitly enable them.

dlech, 2022-04-07:
> 2920

It doesn't use the standard 0x2902 "Client Characteristic Configuration" descriptor for this?

The `start_notify()` function should also write to this descriptor. You can log Bluetooth packets to be sure.

OlafSimon, 2022-04-07:
Sorry, this was a typing error. It uses the standard descriptor 0x2902. I'll check if start_notify() writes to that.

OlafSimon, 2022-04-08:
Sorry for making such a scrupulous error. After my trials with gatttool, where I could turn on and off the notifications by directly writing to 0x2902 per handle, it seems I didn't even try not doing this step after using start_notify() that does this step automatically (or made an error during this test).

'bleak' is just working fine with just calling start_notify(). 

Maybe someone is interested in the code to receive key notifications from the Sensor Tag:

```
import asyncio
from bleak import BleakClient

charUUID_Keys = '0000ffe1-0000-1000-8000-00805f9b34fb'

def myKeyNotify(sender, data):
    print("myKeyNotify from {0}: {1}".format(sender, data))

async def main():
    print("SensorTag")
    tag = BleakClient('54:6C:0E:52:C7:84')
    print("Connecting")
    await tag.connect()
    await tag.start_notify(charUUID_Keys, myKeyNotify)
    print("Waiting for notifications. Press user key on SensorTag!")
    await asyncio.sleep(5)
    print("Disconnecting")
    await tag.stop_notify(charUUID_Keys)
    await tag.disconnect()
    
asyncio.run(main())
```

closed 2022-04-08 by OlafSimon

hbldh, 2022-04-08:
@OlafSimon There is also the example file [sensortag.py](https://github.com/hbldh/bleak/blob/develop/examples/sensortag.py) which deals specifically with the TI SensorTag 2650, in case you are interested. It being the test device I used for originally writing Bleak makes it probably the best tested peripheral of all.

OlafSimon, 2022-04-08:
Hello Henrik,

thanks for your notice, I found that excample already, although didn't notice the key feature at the end.

Olaf

------------ Weitergeleitete Nachricht ------------
Datum: Fri, 8 Apr 2022 06:28:05 -0700

https://github.com/hbldh/bleak/issues/799#issuecomment-1092860353

Sent with Dekko - the convergent email app

Sent with Dekko - the convergent email app

OlafSimon, 2022-04-08:
I forgot, many thanks for your perfect support.

Am Thu, 7 Apr 2022 17:34:35 +0000, David Lechner schrieb:
>
>> 2920
>
> It doesn't use the standard 0x2902 "Client Characteristic 
> Configuration" descriptor for this?
>
> The `start_notify()` function should also write to this 
> descriptor. You can log Bluetooth packets to be sure.
>

-- 
Sent with Dekko - the convergent email app