hbldh/bleak #800: For some reason the bleak read_gatt_char command is repeatedly giving 0x01 in return write is working fine but read is giving issues,stuck on this for many days now with no avail

ps2229, 2022-04-08:
* bleak version:0.14.2
* Python version:3.9
* Operating System:Windows & Android
* BlueZ version (`bluetoothctl -v`) in case of Linux:

For some reason the bleak read_gatt_char command is repeatedly giving 0x01 in return write is working fine but read is giving issues,stuck on this for many days now with no avail
### What I Did

 from kivy.app import App
# from kivy.core.window import Window
from kivy.uix.label import Label
from kivy.uix.scrollview import ScrollView

# bind bleak's python logger into kivy's logger before importing python module using logging
from kivy.logger import Logger
import logging

logging.Logger.manager.root = Logger

import asyncio
import bleak

class ExampleApp(App):
    def __init__(self):
        super().__init__()
        self.label = None
        self.running = True

    def build(self):
        self.scrollview = ScrollView(do_scroll_x=False, scroll_type=["bars", "content"])
        self.label = Label(font_size="10sp")
        self.scrollview.add_widget(self.label)
        return self.scrollview

    def line(self, text, empty=False):
        Logger.info("example:" + text)
        if self.label is None:
            return
        text += "\n"
        if empty:
            self.label.text = text
        else:
            self.label.text += text

    def on_stop(self):
        self.running = False

    async def example(self):
        while self.running:
            try:
                self.line("scanning")
                scanned_devices = await bleak.BleakScanner.discover(1)
                self.line("scanned", True)

                if len(scanned_devices) == 0:
                    raise bleak.exc.BleakError("no devices found")

                scanned_devices.sort(key=lambda device: -device.rssi)

                for device in scanned_devices:
                    device1=f"{device.name}"

                    #print(f"{device.name}")
                    if device1=="Bluno":
                        try:
                            async with bleak.BleakClient(device) as client:
                                paired = await client.pair(protection_level=2)
                                print(f"Paired: {paired}")
                                COLOR_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb"

                                uu = '1'
                                #write_value = bytearray([int(uu)])
                                #services = await client.get_services()
                                #await client.write_gatt_char(COLOR_CHARACTERISTIC, write_value)
                                #await asyncio.sleep(3.0)
                                value = await client.read_gatt_char(COLOR_CHARACTERISTIC)
                                print(value)
                                print("I/O Data Read Value: {0}".format(value))

                                w=int.from_bytes(value, byteorder='big')
                                print(w,'www')
                                #value = await client.read_gatt_char(COLOR_CHARACTERISTIC)
                                #int_val = int.from_bytes(value, "big")
                                #print(int_val,'dddd')

                        except bleak.exc.BleakError as e:
                            self.line(f"  error {e}")
                            asyncio.sleep(10)

            except bleak.exc.BleakError as e:
                self.line(f"ERROR {e}")
                await asyncio.sleep(1)

        self.line("example loop terminated", True)

async def main(app):
    await asyncio.gather(app.async_run("asyncio"), app.example())

if __name__ == "__main__":
    Logger.setLevel(logging.DEBUG)
    # app running on one thread with two async coroutines
    app = ExampleApp()
    asyncio.run(main(app))

Dont know what I am doing wrong the write commands work perfectly but with the read they keep giving the wrong output of 0x01 always even though I am not sending 1 anywhere.

dlech, 2022-04-08:
You could try logging Bluetooth packets to see if the device is really sending 1 or something else.

ps2229, 2022-04-09:
Thanks for the help , but I fixed the problem it was related to use_cached command in the read_gatt_char somehow incorrect data was coming when the use_cache was false as soon as it became True the correct data started coming

ps2229, 2022-04-09:
I do have one more problem that in one read_gatt_char command data is fine but reading too many  characters data gets fudged up.

closed 2022-04-30 by ps2229