hbldh/bleak #804: [Funding] Consider joining GitHub Sponsors

leycec, 2022-04-09:
As of this writing, [412 downstream projects](https://github.com/hbldh/bleak/network/dependents?package_id=UGFja2FnZS0yMDk1NTM0ODY%3D) – at least some of which presumably generate revenue for *somebody* – depend on BLEAK. Yet there exists no `.github/FUNDING.yml` file redirecting dependents to funding portals for accelerating development.

In #412, @dlech notes:

> If you know of some large companies that would like to use Bleak and **sponsor multiple full time developers**, I'm sure we could get there much more quickly. :smile:

It's kinda hard to sponsor a project that has yet to publish sponsorship links. I recently joined [GitHub Sponsors myself on a comparable project](https://github.com/sponsors/leycec) and can confirm that the process was unmitigated bureaucratic Hell – but I'm also in a non-trivial tax situation. <sup>*...dual U.S.-Canadian citizen residing in Canada == the Ninth Circle*</sup>

You're probably not me. And I have no doubts: @dlech and his merry band of collaborators can get this done sooner than I did. I regret not doing so earlier. Don't be like me, fellow Pythonistas. Do this to feed the future of BLEAK, yourself, and your cats.

dlech, 2023-03-20:
I have set up sponsors for my personal account if anyone wants to hire me.

closed 2023-03-20 by dlech

leycec, 2023-03-21:
Congratulations, @dlech! So proud of everything Bleak has already accomplished. It takes time, but you'll rapidly gain a cult following of corporate and individual backers. It's just what happens to awesome devs.

Thanks for all the tremendous effort these many years. You surely deserve all the :money_mouth_face: :money_with_wings: :moneybag:  .