hbldh/bleak #805: "Too many open files" error in BlueZ log after a while

nickrbogdanov, 2022-04-14:
I have a script using bleak to occasionally write to a BLE device, like:

```
async with BleakClient(address) as client:
    await client.write_gatt_char(UUID_0, bytearray([0x01]))
    await client.write_gatt_char(UUID_1, bytearray([0xff]))
```

Sometimes this hits an exception because the device has a long polling interval.  My code catches the exception and retries a couple of times, with a backoff period.

Every couple of weeks I see this error in `journalctl -u bluetooth`:

```
Apr 13 21:09:26 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
Apr 13 21:09:30 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
Apr 13 21:09:32 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
Apr 13 21:23:58 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
Apr 13 21:24:02 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
Apr 13 21:24:04 localhost bluetoothd[942102]: ATT bt_io_connect(xx:xx:xx:xx:xx:xx): socket(SEQPACKET, L2CAP): Too many open files (24)
```

The address in the log matches the device that bleak is trying to query.

This error seems to break all BLE connectivity on the machine.  The error (and BLE outage) goes away immediately when I restart the Python script that is using bleak.

Is it possible that bleak is leaking resources, maybe when an operation fails?  If you could advise on how to figure out what resources (e.g. dbus connections) are currently in use by bleak, I can instrument my code to monitor this.

dlech, 2022-04-14:
I wonder if this is similar to #500.

nickrbogdanov, 2022-04-14:
If I run the `async with BleakClient(address) as client:` loop a couple of times over in my bleak app, while monitoring the `/proc/xxx/fd` directory belonging to bluetoothd, I can see that bluetoothd leaks a couple of file descriptors on each iteration.  The pattern I see in `lsof` is that bluetoothd consumes 2 sockets + one file descriptor open to `/dev/urandom`.

This leakage happens even when the requests do not time out.  The requests are occasional, just a few times a day, so that probably explains why it takes a few weeks to hit 1024 fd's.

As a workaround I'm having my bleak app cache and reuse the client object across different requests.  This seems to make the fd leakage go away:

```
        addr = entry["address"]
        if addr in self.clients:
            client = self.clients[addr]
        else:
            client = BleakClient(addr)
            self.clients[addr] = client

        try:
            await client.connect()
        except:
            pass

        await client.write_gatt_char(UUID_0, bytearray([0x01]))
```

dlech, 2022-07-26:
This should be fixed by #902. Can you please test?

commit 5bed883b (2022-07-26): bluezdbus/client: fix leaking socket file descriptors

This works around a bug in the dbus_next library where the socket is
not closed on disconnect.

Fixes #805.
closed 2022-07-29 by dlech