hbldh/bleak #807: client.disconnect causes timeout even with example code

savejeff, 2022-04-16:
* bleak version: 0.14.2
* Python version: 3.8
* Operating System: Windows 10

### Description

I'm unable to close a connection to a client without a timeout.

### What I Did

I Connect to a ESP32 with a BLE Server running the "standard" UART Characteristic. Even just connecting and then trying to disconnect causes this behavior.

I tried several methods, all end in a timeout error.

I've used the example code from this documentation: https://buildmedia.readthedocs.org/media/pdf/bleak/stable/bleak.pdf Page 10

```
address = "94:B9:7E:FE:76:8E"

	async def main(address):
	client = BleakClient(address)
	try:
		await client.connect()

		await client.write_gatt_char(CHARACTERISTIC_UUID_UART_RX, bytes("Hallo, World!", 'ascii'))
		print(f"write done")

	except Exception as e:
		print(e)
	finally:
		await client.disconnect()

asyncio.run(main(address))

```

causes this
```

Traceback (most recent call last):
  File "XXX/BLEHelp.py", line 475, in <module>
    asyncio.run(main(address))
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\asyncio\runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\asyncio\base_events.py", line 616, in run_until_complete
    return future.result()
  File "XXX/BLEHelp.py", line 473, in main
    await client.disconnect()
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\site-packages\bleak\backends\winrt\client.py", line 313, in disconnect
    await asyncio.wait_for(event.wait(), timeout=10)
  File "C:\Users\XXX\AppData\Local\Programs\Python\Python38\lib\asyncio\tasks.py", line 490, in wait_for
    raise exceptions.TimeoutError()
asyncio.exceptions.TimeoutError

```

dlech, 2022-04-16:
I've noticed this recently too. I think we just need to extend the default timeout value to 30 seconds or so. 

For background, Windows doesn't actually let us tell the device to disconnect. We can only say that we aren't using it any more and it is up to the OS to decide when to actually disconnect.

goldwake, 2022-04-19:
for what it's worth, I am often seeing the same issue with disconnection too. Sometimes it works, sometimes it doesn't

    bleak version: 0.14.2
    Python version: 3.9
    Operating System: Windows 10

PR #974 (2022-09-05): backends/winrt/client: extend disconnect timeout
This changes the disconnect timeout on the WinRT backend to 120s.

Sometimes it can take up to a minute for Windows to actually close
the GATT session/disconnect the device. This behavior is also seen
in other apps, like WebBluetooth in Chrome. Furthermore, the device
remains connected even after the app is closed, so clearly this is
an issue with the Windows Bluetooth stack and not something that
can be fixed in userspace code.

Fixes #807.
commit 79259d3e (2022-09-05): backends/winrt/client: extend disconnect timeout

This changes the disconnect timeout on the WinRT backend to 120s.

Sometimes it can take up to a minute for Windows to actually close
the GATT session/disconnect the device. This behavior is also seen
in other apps, like WebBluetooth in Chrome. Furthermore, the device
remains connected even after the app is closed, so clearly this is
an issue with the Windows Bluetooth stack and not something that
can be fixed in userspace code.

Fixes #807.
PR #1006 (2022-09-12): Release/v0.17.0

Added
-----
* ``AdvertisementData`` class now has an attribute ``tx_power``. Merged #987.

Changed
-------
* ``BleakClient`` methods now raise ``BleakError`` if called when not connected in WinRT backend. Merged #973.
* Extended disconnect timeout to 120 seconds in WinRT backend. Fixes #807.
* Changed version check for BlueZ battery workaround to exclude versions >= 5.55. Merged #976.
* Use Poetry for build system and dependencies. Merged #978.
* The BlueZ D-Bus backend implements a services cache between connections to significancy improve reconnect performance.
  To use the  [... cut]
closed 2022-09-05 by dlech