hbldh/bleak #808: How to reduce the number of missed packets received by Bleak?

allrobot, 2022-04-18:
```PYTHON
import asyncio
import logging
import EMG_NUM
import numpy as np
from bleak import BleakClient

logger = logging.getLogger()

def callback_handler(sender,data):
    # Use await asyncio.wait_for(queue.get(), timeout=1.0) if you want a timeout for getting data.
    if data is None:
        logger.info(
            "Quit...."
        )
    else:

        try:
            print(str(data, "utf-8"))
        except:
            pass

async def main(address: str, char_uuid: str):
    async with BleakClient(address) as client:
        await client.start_notify(char_uuid, callback_handler)
        while(True):
             await asyncio.sleep(1.0)

def ble():
    while EMG_NUM.Kill == False:
        try:
            logging.basicConfig(level=logging.INFO)
            asyncio.run(
                main(
                    "7C:9E:BD:48:6A:36",
                    "ca73b3ba-39f6-4ab3-91ae-186dc9577d99",
                )
            )
        except:
            pass
```

Comparing the incoming and outgoing texts, ascend missed several packets.

I find that some library functions are written with async, how can I keep sending values continuously?

With a While loop, there is a gap between the completion of the statement inside the loop and the continuation of the loop, at which point the program is not running and the complete and continuous text cannot be received.

How can I change the code to keep receiving uninterrupted and ensure that packets are not lost?

dlech, 2022-04-21:
Usually, I set up an `asyncio.Event()` that waits for disconnect like in https://github.com/hbldh/bleak/blob/develop/examples/disconnect_callback.py. The notification handler will be called while awaiting the event. The same applies to awaiting sleep - it does not block the rest of the program.

allrobot, 2022-04-21:
> Usually, I set up an `asyncio.Event()` that waits for disconnect like in https://github.com/hbldh/bleak/blob/develop/examples/disconnect_callback.py. The notification handler will be called while awaiting the event. The same applies to awaiting sleep - it does not block the rest of the program.

So it's not it's fault for missing a few packets?😂

dlech, 2022-04-21:
We already discussed this in #794 and #797.

closed 2022-04-22 by allrobot