hbldh/bleak #809: Device Failing to Connect

ccorreia-rhaeos, 2022-04-18:
* bleak version: 0.14.2
* Python version: 3.9
* Operating System: Windows 11 Build: 22000.613
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

I am a first-time bleak user and want to talk to a GATT server with a known address. Connection can be established as shown within Windows Bluetooth settings however it seems like Bleak never recognizes that the device connects and is stuck waiting

### What I Did

async with BleakClient(address) as client:
        print('connected')

Force quitting this yields the following error:

      ----> 1 async with BleakClient(address) as client:
            2         print('connected')
      ~\anaconda3\lib\site-packages\bleak\backends\client.py in __aenter__(self)
           59 
           60     async def __aenter__(self):
      ---> 61         await self.connect()
           62         return self
           63 
      ~\anaconda3\lib\site-packages\bleak\backends\winrt\client.py in connect(self, **kwargs)
          273 
          274         # Obtain services, which also leads to connection being established.
      --> 275         await self.get_services()
          276 
          277         return True
      
      ~\anaconda3\lib\site-packages\bleak\backends\winrt\client.py in get_services(self, **kwargs)
          444             logger.debug("Get Services...")
          445             services: Sequence[GattDeviceService] = _ensure_success(
      --> 446                 await self._requester.get_gatt_services_async(
          447                     BluetoothCacheMode.UNCACHED
          448                 ),
      
      CancelledError: 
      
      
      
      
      
      ```

hientv1999, 2022-04-20:
I'm currently having the same issue. This issue just happens on Windows 11. People here say they can make it work by pairing on another app first. https://stackoverflow.com/questions/70178620/bleak-python-does-not-respond-on-connect
I have tried with Bluetooth LE Explorer from Microsoft but it didn't work still

dlech, 2022-04-21:
I usually solve issues like this by logging Bluetooth packets to see what is going on behind the scenes.

hientv1999, 2022-04-21:
In my case, it stucks at "Get Services..." for an hour.
[image]
Sometimes if it doesn't stuck there, it still got this error.
[image]

angrygorilla, 2022-07-27:
im also having the same issue

nickbedbury, 2022-07-28:
I  think am also having this issue.  Bleak 0.14.3 on Windows 10 Pro.  

Sometimes I see the log message `Services resolved for BleakClientWinRT <BDADDR>` but it does not seem to connect properly.

dlech, 2022-07-28:
Can everyone please provide a minimal reproducible test case and Wireshark logs of the Bluetooth packets?

nickbedbury, 2022-07-28:
I see similar behavior with the `connect_by_bledevice.py` example from this repo, so that might be a minimal reproducible test case.

```
>python connect_by_bledevice.py
Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\asyncio\locks.py", line 226, in wait
    await fut
asyncio.exceptions.CancelledError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\asyncio\tasks.py", line 490, in wait_for
    return fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "C:\Users\user\Desktop\bleak_connect_demo2.py", line 26, in <module>
    asyncio.run(main(sys.argv[1] if len(sys.argv) == 2 else ADDRESS))
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\asyncio\runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\asyncio\base_events.py", line 647, in run_until_complete
    return future.result()
  File "C:\Users\user\Desktop\bleak_connect_demo2.py", line 18, in main
    async with BleakClient(device) as client:
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\site-packages\bleak\backends\client.py", line 61, in __aenter__
    await self.connect()
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\site-packages\bleak\backends\winrt\client.py", line 268, in connect
    await asyncio.wait_for(event.wait(), timeout=timeout)
  File "C:\Users\user\AppData\Local\Programs\Python\Python39-32\lib\asyncio\tasks.py", line 492, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError

```

I am not able to capture Wireshark presently.  I'm not too familiar with doing that for Bluetooth...but it shows no activity when I enabled capture.

dlech, 2022-07-28:
> I am not able to capture Wireshark presently. I'm not too familiar with doing that for Bluetooth...but it shows no activity when I enabled capture.

There are instructions on how to do this at https://bleak.readthedocs.io/en/latest/troubleshooting.html#windows-10

nickbedbury, 2022-08-03:
I updated to `bleak==0.15.1` and am able to connect now.  I haven't done an exhaustive test, but 5 attempts were successful in a row, which seems improved from before.

The hold-up recording Wireshark for me is permission related (work laptop, no admin rights).

I am still unable to write to a CHRC, but perhaps that is some other problem.

dlech, 2024-05-04:
Is this still a problem with Bleak v0.22.0?

hientv1999, 2024-05-04:
> Is this still a problem with Bleak v0.22.0?

It works perfectly fine now. Probably it was a programming error I made in the past. Since mid 2023, I have it worked correctly. My app is currently running on v0.21.1 and has been tested on Windows 11.

dlech, 2024-05-04:
Thanks. We have quite a few similar newer issues with the same timeout when getting services. So I'll go ahead and close this one.

closed 2024-05-04 by dlech