hbldh/bleak #810: Why bleak prompt Event is closed? RuntimeError

allrobot, 2022-04-20:
* bleak version: 0.14.2
* Python version: 3.9
* Operating System: win10

### Description

Run bleak program, it will to prompt:
```BASH
Exception ignored in: Exception ignored in: <function BleakClientWinRT.connect.<locals>.session_status_changed_event_handler at 0x00000283C8FD98B0>
Traceback (most recent call last):
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\site-packages\bleak-0.15.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 248, in session_status_changed_event_handler
Exception ignored in: <function BleakClientWinRT.connect.<locals>.session_status_changed_event_handler at 0x00000283C8FD98B0>
Exception ignored in: <function BleakClientWinRT.connect.<locals>.session_status_changed_event_handler at 0x00000283C8FD98B0>
Traceback (most recent call last):
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\site-packages\bleak-0.15.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 248, in session_status_changed_event_handler
Exception ignored in: <function BleakClientWinRT.connect.<locals>.session_status_changed_event_handler at 0x00000283C8FD98B0>
Traceback (most recent call last):
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\site-packages\bleak-0.15.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 248, in session_status_changed_event_handler
<function BleakClientWinRT.connect.<locals>.session_status_changed_event_handler at 0x00000283C8FD98B0>
Traceback (most recent call last):
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\site-packages\bleak-0.15.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 248, in session_status_changed_event_handler
Traceback (most recent call last):
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\site-packages\bleak-0.15.0a1-py3.9.egg\bleak\backends\winrt\client.py", line 248, in session_status_changed_event_handler
    loop.call_soon_threadsafe(handle_session_status_changed, args)
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\asyncio\base_events.py", line 791, in call_soon_threadsafe
    self._check_closed()
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\asyncio\base_events.py", line 510, in _check_closed
    loop.call_soon_threadsafe(handle_session_status_changed, args)
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\asyncio\base_events.py", line 791, in call_soon_threadsafe
    self._check_closed()
  File "C:\Users\Administrator\.conda\envs\tf-gpu\lib\asyncio\base_events.py", line 510, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
```

### What I Did

Restarting the BT device for several times does not work. I must wait patiently for the connection success message to appear.

dlech, 2022-04-21:
Can you share the program to reproduce this error?

allrobot, 2022-04-21:
> Can you share the program to reproduce this error?

```PYTHON
import asyncio
import logging
import sys
import numpy as np
from bleak import BleakClient

logger = logging.getLogger()

def callback_handler(sender,data):

    if data is None:
        logger.info(
            "Qiuting...."
        )
    else:

        try:
            EMG_NUM.EMG=np.array(str(data, "utf-8").split(','),dtype='int64')

        except:
            pass

async def main(address: str, char_uuid: str):
    async with BleakClient(address) as client:

        
        await client.start_notify(char_uuid, callback_handler)
        
        while True:
            await asyncio.sleep(1.0)

def ble():
    
    
    while EMG_NUM.Kill == False:
        try:
            logging.basicConfig(level=logging.INFO)
            asyncio.run(
                main(
                    "7C:9E:BD:48:6A:36",
                    "ca73b3ba-39f6-4ab3-91ae-186dc9577d99",
                )
            )
        except:
            pass
```