hbldh/bleak #811: Write Requests Not Working

rc14193, 2022-04-21:
* bleak version: 0.14.2
* Python version: 3.8.5, also tried 3.10.4
* Operating System: mac OS monterey 12.3.1, and tried Windows 10 Pro version 10.0.19044
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Describe what you were trying to get done.

I have written a custom bluetooth peripheral, and I am now trying to connect to and communicate to it with bleak. It previously worked fine, but it has now stopped performing write requests, and it seems as if bleak generally is no longer performing write requests.

Tell us what happened, what went wrong, and what you expected to happen.

I am able to discover and connect to the device. I am able to perform read operations, but when I perform write operations the bleak write_gatt_char call never completes. I have tested the peripheral with other centrals that have been coded in Xamarin and using LiteBlue to test writes, and they work without issue. Included in the log statements below, it outputs that it is performing the write, but it just hangs from there.

### What I Did

It is preferable if an issue contains a [Miminal Workable Example](https://stackoverflow.com/help/minimal-reproducible-example).
This will otherwise be one of the first questions you will get as a response.

From my main application, I built a minimal test script that simply tries to perform a single write operation and whose methods model the same process that I perform bleak communication in my application.

```
import sys
import logging
import asyncio
import threading
from bleak import discover, BleakClient, BleakScanner
from bleak.exc import BleakError

SELECT_PAGE_UUID = "d1a7ff07-af78-4449-a34f-4da1afaf51bc"
logger = logging.basicConfig(filename='bletest.log', level=logging.DEBUG)
log = logging.getLogger(__name__)
ch = logging.StreamHandler()
ch.setLevel(logging.DEBUG)
formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')

# add formatter to ch
ch.setFormatter(formatter)

# add ch to logger
log.addHandler(ch)

async def perform_discovery():
    log.debug("trying to perform discover")
    devices = await discover()
    wp_devices = [dev for dev in devices if "wp" in dev.name.lower()]
    log.debug("discover complete, returning found devices")
    return wp_devices

async def connect_spec() -> bool:
    log.debug(f"trying to connect")
    global client
    await client.connect()
    log.debug(f"Connected: {client.is_connected}")
    services = await client.get_services()
    new_line = '\n\n'
    log.debug(f"characteristic properties are {f'{new_line}'.join([str(char.uuid)+str(char.properties) for char in services.characteristics.values()])}")
    return True

async def perform_write():
    log.debug("trying to perform write")
    await attr_write()
    return

async def attr_write():
    global client
    page_ids = bytearray([1,2])
    request = await client.write_gatt_char(SELECT_PAGE_UUID, b"\x00")
    return
[... 84 lines cut ...]

d1a7ff03-af78-4449-a34f-4da1afaf51bc['read', 'write', 'notify']

d1a720b4-af78-4449-a34f-4da1afaf51bc['read', 'write']

d1a7ff04-af78-4449-a34f-4da1afaf51bc['write']

d1a7ff05-af78-4449-a34f-4da1afaf51bc['write']

d1a7ff06-af78-4449-a34f-4da1afaf51bc['read']

d1a7ff07-af78-4449-a34f-4da1afaf51bc['write']

d1a7ff08-af78-4449-a34f-4da1afaf51bc['read', 'notify']

d1a7ff09-af78-4449-a34f-4da1afaf51bc['read', 'notify']

d1a7ff0a-af78-4449-a34f-4da1afaf51bc['read', 'write']
2022-04-21 07:22:41,256 - __main__ - DEBUG - successfully connected. Trying to write now
2022-04-21 07:22:41,256 - __main__ - DEBUG - trying to perform write
2022-04-21 07:22:41,257 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff07-af78-4449-a34f-4da1afaf51bc : b'\x00'

Paste the command(s) you ran and the output.
If there was a crash, please include the traceback here as well.
```

rc14193, 2022-04-21:
[image]

I am not an expert in the low level operations of BLE, but using PacketLogger on Mac I don't see the write request even being sent. The above screenshot is the packet capture of the script with my device. No further info is sent because it hangs and the last information transfer is that HCI event

rc14193, 2022-04-21:
An addition, I wrote a script to try writing all the characteristics that have the write property. This I thought would eliminate me mistyping the UUID since it uses the bleakcharacterisitc object. The log output now only hangs on the last write. The packet capture appears to be the same.

SCRIPT:
```
import sys
import logging
import time
import asyncio
import threading
from bleak import discover, BleakClient, BleakScanner
from bleak.exc import BleakError

logger = logging.basicConfig(filename='bletest.log', level=logging.DEBUG)
log = logging.getLogger(__name__)
ch = logging.StreamHandler()
ch.setLevel(logging.DEBUG)
formatter = logging.Formatter('%(asctime)s - %(name)s - %(levelname)s - %(message)s')

# add formatter to ch
ch.setFormatter(formatter)

# add ch to logger
log.addHandler(ch)

async def perform_discovery():
    log.debug("trying to perform discover")
    devices = await discover()
    dev_info = '\n'.join([str(dev) for dev in devices])
    log.debug(f"found devices with name and addresses:\n{dev_info}")
    wp_devices = [dev for dev in devices if "wp" in dev.name.lower()]
    log.debug("discover complete, returning found devices")
    return wp_devices

async def connect_spec() -> bool:
    log.debug(f"trying to connect")
    global client
    await client.connect()
    log.debug(f"Connected: {client.is_connected}")
    services = await client.get_services()
    new_line = '\n\n'
    log.debug(f"characteristic properties are \n{f'{new_line}'.join([str(char.uuid)+' '+str(char.properties) for char in services.characteristics.values()])}")
    return True

async def perform_write():
    global client
    services = await client.get_services()
    for char in services.characteristics.values():
        try:
            if 'write' in char.properties:
                log.debug(f"char {char.uuid} has write property. Trying to write it")
                request = await client.write_gatt_char(char, b"\x00")
        except Exception as e:
            log.error(f"error while trying to write char of {e}, trying next write")
    return

log.debug("starting test for write characteristics")
log.debug("trying to find wp devices")
loop = asyncio.new_event_loop()
thread = threading.Thread(target=loop.run_forever)
thread.start()
fut = asyncio.run_coroutine_threadsafe(perform_discovery(), loop)
devices = fut.result()
if devices == []:
    log.debug("didn't find any devices. Exiting")
    sys.exit(0)

log.debug(f"found wp devices of {devices}")
device = devices[0]
client = BleakClient(device)
fut = asyncio.run_coroutine_threadsafe(connect_spec(), loop)
[... 77 lines cut ...]

d1a7ff09-af78-4449-a34f-4da1afaf51bc ['read', 'notify']

d1a7ff0a-af78-4449-a34f-4da1afaf51bc ['read', 'write']
2022-04-21 09:05:00,572 - __main__ - DEBUG - successfully connected. Trying to write now
2022-04-21 09:05:00,572 - __main__ - DEBUG - char d1a7ff01-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,573 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff01-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,573 - __main__ - DEBUG - char d1a7ff02-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,573 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff02-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,573 - __main__ - DEBUG - char d1a7ff03-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,573 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff03-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,573 - __main__ - DEBUG - char d1a720b4-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,574 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a720b4-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,574 - __main__ - DEBUG - char d1a7ff04-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,574 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff04-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,574 - __main__ - DEBUG - char d1a7ff05-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,574 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff05-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,574 - __main__ - DEBUG - char d1a7ff07-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,574 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff07-af78-4449-a34f-4da1afaf51bc : b'\x00'
2022-04-21 09:05:00,574 - __main__ - DEBUG - char d1a7ff0a-af78-4449-a34f-4da1afaf51bc has write property. Trying to write it
2022-04-21 09:05:00,574 bleak.backends.corebluetooth.client DEBUG: Write Characteristic d1a7ff0a-af78-4449-a34f-4da1afaf51bc : b'\x00'

```

[image]

rc14193, 2022-04-21:
I am not sure why this changed, but when I added the response = True to the write_gatt_char call it resolved the issue. My peripheral's code has not changed. I'm confused but satisfied it's working. Hopefully this helps someone else in the future. If a contributor has maybe more insight as to the why that would be welcome always.

dlech, 2022-04-21:
Selecting "write with response" or "write without response" depends on the device you are connecting to. Some devices may only support one or the other while some devices may support both. Also see #786.

dlech, 2023-07-20:
Should be fixed by #1236

closed 2023-07-20 by dlech