hbldh/bleak #815: Distinguish Advertisement Data and Scan Response Data

RAlexeev, 2022-04-26:
* bleak version: 0.14.2

### Description

Could you clarify about a limitation of the library, is it possible to distinguish data, received by the first advertisement packet (PDU type: `ADV_IND`) and the subsequent scan response packet (PDU type: `SCAN_RSP`) from a specific device? Right now, launching the example https://github.com/hbldh/bleak/blob/develop/examples/discover.py, I see that it returns both advertisement data and scan response data joined together. I would like to see exactly which data were received through advertisement packet and which one through scan response packet. Is it possible to realize using Bleck library?

dlech, 2022-04-26:
It is only technically possible on Windows, not on other OSes. We made Windows behave like the other OSes in Bleak for consistency and because it caused issues for many users.

RAlexeev, 2022-04-26:
Many thanks for your clarification.

closed 2022-04-26 by RAlexeev