hbldh/bleak #817: bleak giving incorrect values in android

ps2229, 2022-04-29:
* bleak version:0.14.2
* Python version:3.9
* Operating System:android
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Trying to run the same code on windows and android both works perfectly in windows but in android it seems to be giving incorrect values more like only 0x01 only. Does anyone know what to do?

### What I Did

    def scan(self):
        loop = asyncio.get_event_loop()
        loop.run_until_complete(self.begin_scan())

    async def begin_scan(self):
        a = []
        scanned_devices = await bleak.BleakScanner.discover(1)

        if len(scanned_devices) == 0:
            print("NO DEVICES")
        scanned_devices.sort(key=lambda device: -device.rssi)
        for device in scanned_devices:
            device1 = f"{device.name}"
            print(device1)
            if device1 == "Bluno":
                try:
                    async with bleak.BleakClient(device) as client:
                        paired = client.pair(protection_level=2)
                        print(f"Paired: {paired}")
                        COLOR_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb"
                        uu = '1'
                        write_value = bytearray([int(uu)])
                        await client.write_gatt_char(COLOR_CHARACTERISTIC, write_value)
                        await asyncio.sleep(6.0)
                        await asyncio.sleep(1.0)
                        val1 = await client.read_gatt_char(COLOR_CHARACTERISTIC, use_cached=True)
                        print(val1, 'val1')
                      
                except:
                    print("error1")
                    asyncio.sleep(10)
            else:
                print('Device Not Found')

dlech, 2022-04-29:
Does it make a difference if you remove the `use_cached` argument?

ps2229, 2022-04-30:
In windows without using the use_cached it was giving incorrect values like 0x01 after using use_cached I got correct values .
In android tried both no luck.

dlech, 2022-04-30:
I think `use_cached` is a Windows-only option. And if it doesn't work without that option, then there is something wrong with the device. Logging Bluetooth packets will let you see what is actually going on.

ps2229, 2022-05-02:
def scan1(self):
        loop = asyncio.get_event_loop()
        loop.run_until_complete(self.begin_scan1())

    async def begin_scan1(self):
        async def callback(data):
            print('uu')
            print(data,'dataaaaaaa')
           
        a = []
        scanned_devices = await bleak.BleakScanner.discover(1)

        if len(scanned_devices) == 0:
            print("NO DEVICES")
        scanned_devices.sort(key=lambda device: -device.rssi)
        for device in scanned_devices:
            device1 = f"{device.name}"
            print(device1)
            if device1 == "Bluno":
            try:
                async with bleak.BleakClient(device) as client:
                    paired = await client.pair(protection_level=2)
                    print(f"Paired: {paired}")
                    COLOR_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb"
                    uu = '1'
                    write_value = bytearray([int(uu)])
                    await client.write_gatt_char(COLOR_CHARACTERISTIC, write_value)
                    await asyncio.sleep(1.0)
                    print('aa')
                    await client.start_notify(COLOR_CHARACTERISTIC, callback)
                    await asyncio.sleep(10.0)
                    print('bb')
                    await client.stop_notify(COLOR_CHARACTERISTIC)
                    
               except:

                    asyncio.sleep(10)
               
            else:
                    print('NO DEVICE')
    
I changed the code to this but facing alot of errors , can you see anything wrong with the code?

ps2229, 2022-05-03:
> I think `use_cached` is a Windows-only option. And if it doesn't work without that option, then there is something wrong with

Yes I read in the documentation use_cached only works in windows only, any idea how to read data in android and read_gatt_char and start_notify are both giving issues.