hbldh/bleak #822: Improved type annotations

davclark, 2022-05-02:
* bleak version: latest
* Python version: 3.9.11
* Operating System: Linux / CircuitPython
* BlueZ version (`bluetoothctl -v`) in case of Linux: not relevant

### Description

At PyCon 2022, we're sprinting to improve type annotations in the adafruit circuitpython libraries. bleak is one of the foundational libraries in the adafruit bluetooth stack, and so we can't quite get complete typing without bleak *also* having complete typing (or alternatively, stubs).

Is there interest in improved typing? We could produce a PR, but I wanted to check if that would be likely to be accepted first!

### What I Did

- Clone the repo
- Install requirements (including dev)
- Install mypy (which is not currently in either of the above)
- Check types with mypy

```
mypy bleak
```

Found 85 errors in 22 files (let me know if you'd like the full output pasted here)

dlech, 2022-05-02:
Bleak should already have complete type hints. Happy to take a PR to fix mypy errors.

It could be better if we did something like #582 though.

JPHutchins, 2023-12-31:
I've put about 1.5 hours into an initial investigation.

For hacks to make mypy "pass" (but still be useful) I would estimate 40 - 80 hours of work.

Per backend it varies.  The android implementation is particularly worrisome and I'm certain that the static analysis is revealing true runtime bugs.

Generally, the architecture itself, due to supporting multiple backends, is not very strongly typed.  For example, while interfaces are defined, they are not generic and the implementations tend to "add what they need".

What I'd recommend is splitting up the approach:

* sunset python 3.8 support -  it's EOL is less than a year away!
* define statically typed interfaces that are consistent across all backends - careful test coverage!
* improve pyproject.toml "groups" to allow for "cross platform" static analysis - e.g. install windows dependencies on linux, install android dependencies on mac, etc. - dev and CI environments only, of course
* start adding mypy requirements one module at a time; probably the scope would be 15-20 PRs making particular folders pass static analysis and adding to the CI
  * probably these steps would update the interfaces definitions, introduce Generics and TypeVars as needed, etc.
  
Total maybe 120 hours for one person.  For the project, more, managing reviews and test coverage of 30-40 PRs.

For reference, here's the initial mypy error list as of posting:

```
typings/CoreBluetooth/__init__.pyi:3: error: Skipping analyzing ".libdispatch": module is installed, but missing library stubs or py.typed marker  [import-untyped]
typings/CoreBluetooth/__init__.pyi:4: error: Skipping analyzing ".Foundation": module is installed, but missing library stubs or py.typed marker  [import-untyped]
typings/CoreBluetooth/__init__.pyi:55: error: Name "registerForConnectionEventsWithOptions_" already defined on line 52  [no-redef]
bleak/backends/p4android/recipes/bleak/fix_setup.py:1: error: Skipping analyzing "setuptools": module is installed, but missing library stubs or py.typed marker  [import-untyped]
bleak/backends/p4android/recipes/bleak/__init__.py:3: error: Cannot find implementation or library stub for module named "pythonforandroid.recipe"  [import-not-found]
bleak/backends/p4android/recipes/bleak/__init__.py:3: note: See https://mypy.readthedocs.io/en/stable/running_mypy.html#missing-imports
bleak/backends/p4android/recipes/bleak/__init__.py:4: error: Cannot find implementation or library stub for module named "pythonforandroid.toolchain"  [import-not-found]
bleak/backends/p4android/recipes/bleak/__init__.py:5: error: Cannot find implementation or library stub for module named "sh"  [import-not-found]
typings/objc/__init__.py:8: error: Missing return statement  [empty-body]
typings/objc/__init__.py:12: error: Missing return statement  [empty-body]
bleak/backends/corebluetooth/utils.py:40: error: "NSData" has no attribute "initWithBytes_length_"  [attr-defined]
bleak/backends/corebluetooth/__init__.py:11: error: Module has no attribute "options"  [attr-defined]
bleak/backends/winrt/descriptor.py:7: error: Cannot find implementation or library stub for module named "bleak_winrt.windows.devices.bluetooth.genericattributeprofile"  [import-not-found]
bleak/backends/bluezdbus/utils.py:22: error: Argument 1 to "BleakDBusError" has incompatible type "str | ErrorType | None"; expected "str"  [arg-type]
bleak/backends/service.py:98: error: Argument 1 to "get_descriptor" of "BleakGATTServiceCollection" has incompatible type "str | int | UUID"; expected "int"  [arg-type]
bleak/backends/winrt/characteristic.py:12: error: Cannot find implementation or library stub for module named "bleak_winrt.windows.devices.bluetooth.genericattributeprofile"  [import-not-found]
bleak/backends/winrt/characteristic.py:73: error: Need type annotation for "__descriptors" (hint: "__descriptors: List[<type>] = ...")  [var-annotated]
bleak/backends/corebluetooth/characteristic.py:99: error: Incompatible return value type (got "list[BleakGATTDescriptorCoreBluetooth]", expected "list[BleakGATTDescriptor]")  [return-value]
bleak/backends/corebluetooth/characteristic.py:99: note: "List" is invariant -- see https://mypy.readthedocs.io/en/stable/common_issues.html#variance
bleak/backends/corebluetooth/characteristic.py:99: note: Consider using "Sequence" instead, which is covariant
bleak/backends/corebluetooth/characteristic.py:118: error: Argument 1 to "append" of "list" has incompatible type "BleakGATTDescriptor"; expected "BleakGATTDescriptorCoreBluetooth"  [arg-type]
bleak/backends/bluezdbus/characteristic.py:44: error: Need type annotation for "__descriptors" (hint: "__descriptors: List[<type>] = ...")  [var-annotated]
docs/conf.py:217: error: Need type annotation for "latex_elements" (hint: "latex_elements: Dict[<type>, <type>] = ...")  [var-annotated]
bleak/backends/winrt/service.py:9: error: Cannot find implementation or library stub for module named "bleak_winrt.windows.devices.bluetooth.genericattributeprofile"  [import-not-found]
bleak/backends/winrt/service.py:22: error: Need type annotation for "__characteristics" (hint: "__characteristics: List[<type>] = ...")  [var-annotated]
bleak/backends/winrt/service.py:33: error: Signature of "characteristics" incompatible with supertype "BleakGATTService"  [override]
bleak/backends/winrt/service.py:33: note:      Superclass:
bleak/backends/winrt/service.py:33: note:          list[BleakGATTCharacteristic]
bleak/backends/winrt/service.py:33: note:      Subclass:
[... cut ...]
examples/kivy/main.py:4: error: Cannot find implementation or library stub for module named "kivy.app"  [import-not-found]
examples/kivy/main.py:7: error: Cannot find implementation or library stub for module named "kivy.uix.label"  [import-not-found]
examples/kivy/main.py:8: error: Cannot find implementation or library stub for module named "kivy.uix.scrollview"  [import-not-found]
examples/kivy/main.py:11: error: Cannot find implementation or library stub for module named "kivy.logger"  [import-not-found]
Found 289 errors in 37 files (checked 74 source files)
```

JPHutchins, 2023-12-31:
Then again, there is some low-hanging fruit.  For example, users are frequently allowed to use int, str, or uuid to specify BLE resources.  This is counterproductive - the user should be forced to put their int, str, or uuid into a type that marshals the value into the required type.  This converts a runtime error into a "squiggle" in the code editor, all while relieving pressure on the documentation.

That said, it must be approached without changing the existing API.  Maybe not so easy.

dlech, 2025-12-16:
I think we can close this one. We now have `pyright` and `mypy` checks passing and running in CI. The Android backend still needs work and is currently ignored by the type checkers.

closed 2025-12-16 by dlech