hbldh/bleak #823: I'm trying to connect to a BLE device via bleak - it works fine when I use client.start_notify on Windows 11 and andriod , but script fails , My logcat error: bleak.exc.BleakError: Descriptor None was not found!

NewAge12, 2022-05-03:
def scan1(self):
        
        loop = asyncio.get_event_loop()
        loop.run_until_complete(self.begin_scan1())

    async def begin_scan1(self):

        def callback2(sender, data):
            print(sender, data, 'llllll')
            self.val1 = data
            
        a = []
        scanned_devices = await bleak.BleakScanner.discover(1)

        if len(scanned_devices) == 0:
            print("NO DEVICES")
        scanned_devices.sort(key=lambda device: -device.rssi)
        for device in scanned_devices:
            device1 = f"{device.name}"
            print(device1)
            if device1 == "Bluno":
                # try:

                async with bleak.BleakClient(device) as client:
                    print('rr')
                    paired = client.pair(protection_level=2)
                    print(f"Paired: {paired}")
                    COLOR_CHARACTERISTIC = "0000dfb1-0000-1000-8000-00805f9b34fb"
                     

                    uu = '1'
                    write_value = bytearray([int(uu)])
                    await client.write_gatt_char(COLOR_CHARACTERISTIC, write_value)
                    await asyncio.sleep(4.0)
                    print('www')
                    await client.start_notify(COLOR_CHARACTERISTIC, callback2)
                    print('wwee')

                    await asyncio.sleep(10.0)
                    print('rrrq')
                    await client.stop_notify(COLOR_CHARACTERISTIC)

NewAge12, 2022-05-03:
bleak.exc.BleakError: Descriptor None was not found!

dlech, 2022-05-03:
Can out share the output of https://github.com/hbldh/bleak/blob/develop/examples/service_explorer.py when connected to this device?

NewAge12, 2022-05-04:
@dlech @hbldh 

INFO:bleak.backends.winrt.client:Services resolved for BleakClientWinRT (C4:BE:84:20:1B:59)
INFO:__main__:Connected: True
INFO:__main__:[Service] 00001800-0000-1000-8000-00805f9b34fb (Handle: 1): Generic Access Profile
INFO:__main__:    [Characteristic] 00002a00-0000-1000-8000-00805f9b34fb (Handle: 2):  (read), Value: b'Bluno'
INFO:__main__:    [Characteristic] 00002a01-0000-1000-8000-00805f9b34fb (Handle: 4):  (read), Value: b'\x00\x00'
INFO:__main__:    [Characteristic] 00002a02-0000-1000-8000-00805f9b34fb (Handle: 6):  (read), Value: b'\x00'
INFO:__main__:    [Characteristic] 00002a03-0000-1000-8000-00805f9b34fb (Handle: 8):  (write), Value: None
INFO:__main__:    [Characteristic] 00002a04-0000-1000-8000-00805f9b34fb (Handle: 10):  (read), Value: b'P\x00\xa0\x00\x00\x00\xe8\x03'
INFO:__main__:[Service] 00001801-0000-1000-8000-00805f9b34fb (Handle: 12): Generic Attribute Profile
INFO:__main__:    [Characteristic] 00002a05-0000-1000-8000-00805f9b34fb (Handle: 13):  (indicate), Value: None
INFO:__main__:        [Descriptor] 00002902-0000-1000-8000-00805f9b34fb (Handle: 15): Client Characteristic Configuration) | Value: b'\x02\x00'
INFO:__main__:[Service] 0000180a-0000-1000-8000-00805f9b34fb (Handle: 16): Device Information
INFO:__main__:    [Characteristic] 00002a23-0000-1000-8000-00805f9b34fb (Handle: 17):  (read), Value: b'Y\x1b \x00\x00\x84\xbe\xc4'
INFO:__main__:    [Characteristic] 00002a24-0000-1000-8000-00805f9b34fb (Handle: 19):  (read), Value: b'DF Bluno'
INFO:__main__:    [Characteristic] 00002a25-0000-1000-8000-00805f9b34fb (Handle: 21):  (read), Value: b'0123456789'
INFO:__main__:    [Characteristic] 00002a26-0000-1000-8000-00805f9b34fb (Handle: 23):  (read), Value: b'FW V1.97'
INFO:__main__:    [Characteristic] 00002a27-0000-1000-8000-00805f9b34fb (Handle: 25):  (read), Value: b'HW V1.7'
INFO:__main__:    [Characteristic] 00002a28-0000-1000-8000-00805f9b34fb (Handle: 27):  (read), Value: b'SW V1.97'
INFO:__main__:    [Characteristic] 00002a29-0000-1000-8000-00805f9b34fb (Handle: 29):  (read), Value: b'DFRobot'
INFO:__main__:    [Characteristic] 00002a2a-0000-1000-8000-00805f9b34fb (Handle: 31):  (read), Value: b'\xfe\x00experimental'
INFO:__main__:    [Characteristic] 00002a50-0000-1000-8000-00805f9b34fb (Handle: 33):  (read), Value: b'\x01\r\x00\x00\x00\x10\x01'
INFO:__main__:[Service] 0000dfb0-0000-1000-8000-00805f9b34fb (Handle: 35): Vendor specific
INFO:__main__:    [Characteristic] 0000dfb1-0000-1000-8000-00805f9b34fb (Handle: 36): Serial (read,write-without-response,write,notify), Value: b'\x01'
INFO:__main__:        [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 38): Characteristic User Description) | Value: b'Serial'
INFO:__main__:    [Characteristic] 0000dfb2-0000-1000-8000-00805f9b34fb (Handle: 39): AT Ctrl (read,write-without-response,write,notify), Value: b'\x02'
INFO:__main__:        [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 41): Characteristic User Description) | Value: b'AT Ctrl'
Process finished with exit code 0

dlech, 2022-05-04:
> INFO:**main**: [Characteristic] 0000dfb1-0000-1000-8000-00805f9b34fb (Handle: 36): Serial (read,write-without-response,write,notify), Value: b'\x01'
> INFO:**main**: [Descriptor] 00002901-0000-1000-8000-00805f9b34fb (Handle: 38): Characteristic User Description) | Value: b'Serial'

This is the "COLOR" characteristic. It does not have a "Client Characteristic Configuration" descriptor, so that explains the error seen on Android. I'm not sure how it works on Windows. You could capture Bluetooth packets on Windows to get a better idea of what is going on under the hood on Windows.

Perhaps we just need to surround https://github.com/hbldh/bleak/blob/46c828b2f856b717394414957258fb77f6fb678a/bleak/backends/p4android/client.py#L497-L500 with `if characteristic.notification_descriptor:`.

NewAge12, 2022-05-05:
@dlech @hbldh 
 The "COLOR" characteristic is just a variable for the UUID . As seen from the get_services list only the '0000dfb1-0000-1000-8000-00805f9b34fb' UUID has the capabilities to read and write data from device to hardware. We did try surrounding 
 if characteristic.notification_descriptor:
          await self.write_gatt_descriptor( 
               characteristic.notification_descriptor, 
               defs.BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE, 
           ) 
still no solution

> This is the "COLOR" characteristic. It does not have a "Client Characteristic Configuration" descriptor, so that explains the error seen on Android. I'm not sure how it works on Windows. You could [...]

@dlech @hbldh 
 The "COLOR" characteristic is just a variable for the UUID . As seen from the get_services list only the '0000dfb1-0000-1000-8000-00805f9b34fb' UUID has the capabilities to read and write data from device to hardware. We did try surrounding 
 if characteristic.notification_descriptor:
          await self.write_gatt_descriptor( 
               characteristic.notification_descriptor, 
               defs.BluetoothGattDescriptor.ENABLE_NOTIFICATION_VALUE, 
           ) 
able to write data with this UUID but for reading there is no function neither read_gatt_char works for start_notify!!!

NewAge12, 2022-05-11:
@dlech @hbldh 
Any update on the issue?

apacodes, 2023-10-06:
I ran into the same thing, my fix was to pin bleak to 0.20.2 which resolves the "error" and makes the call to start_notify() work as expected.

I haven't checked if the pinning causes other packets to be downgraded (since the problem is currently time sensitive) but i do as of now assume that this was caused by a change in bleak.

dlech, 2023-10-06:
@apacodes it sounds like there is a regression between 0.20.2 and 0.21.0? If so, can you please start a new issue with more detail about the problem?

apacodes, 2023-10-06:
Yes.

I will investigate an do so in the coming days.

ccwang65, 2024-01-23:
This issue may because of files in: bleak/backends/p4android
defs.py last line #89:

CLIENT_CHARACTERISTIC_CONFIGURATION_UUID = "2902"

characteristic.py #90

        if descriptor.uuid == defs.CLIENT_CHARACTERISTIC_CONFIGURATION_UUID:
            self.__notification_descriptor = descriptor

Maybe we need to use back the full string:

CLIENT_CHARACTERISTIC_CONFIGURATION_UUID = "00002902-0000-1000-8000-00805f9b34fb"

skywodd, 2024-03-28:
Hello,

I'm currently having the same error (Descriptor None was not found).

I've solved the issue by monkey patching the CLIENT_CHARACTERISTIC_CONFIGURATION_UUID constant at startup in my app using :
```
from bleak.backends.p4android import defs
defs.CLIENT_CHARACTERISTIC_CONFIGURATION_UUID = "00002902-0000-1000-8000-00805f9b34fb"
```

I think `bleak/backends/p4android/characteristic.py` should check for both the full and short uuid constant to avoid regression.

(Bleak version 0.21.1 with Python-for-Android, Kivy and Buildozer, running on Android 13.1)

dlech, 2024-03-28:
Thanks for confirming. I would take a pull request for this. I think the short version is wrong and we don't need to have both.

narodnik, 2024-04-06:
Confirming this as well. I tried downgrading to 0.20.2 but it did not work either.

```
04-06 14:56:18.361 13037 13062 I python  :  Traceback (most recent call last):
04-06 14:56:18.361 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/app/main.py", line 101, in <module>
04-06 14:56:18.362 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/other_builds/python3/arm64-v8a__ndk_target_21/python3/Lib/asyncio/base_events.py", line 653, in run_until_complete
04-06 14:56:18.363 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/app/main.py", line 67, in monitor_hr
04-06 14:56:18.364 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/app/main.py", line 76, in run_sensor
04-06 14:56:18.364 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/app/polarh10.py", line 295, in start_hr_stream
04-06 14:56:18.365 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/ecgapp/arm64-v8a/bleak/__init__.py", line 834, in start_notify
04-06 14:56:18.366 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/ecgapp/arm64-v8a/bleak/backends/p4android/client.py", line 450, in start_notify
04-06 14:56:18.366 13037 13062 I python  :    File "/home/narodnik/src/pump/ecgapp/.buildozer/android/platform/build-arm64-v8a_armeabi-v7a/build/python-installs/ecgapp/arm64-v8a/bleak/backends/p4android/client.py", line 418, in write_gatt_descriptor
04-06 14:56:18.367 13037 13062 I python  :  bleak.exc.BleakError: Descriptor None was not found!

```

commit d45ec90d (2024-05-23): Android: fix 'Descriptor None was not found!' (#1580)

The keys in the descriptor dict are the normalized 128-bit UUID string, so we need to use this to correctly find the descriptor.

Fixes #823
PR #1587 (2024-06-01): Release/v0.22.2

Changed
-------
* Retrieve the BLE address required by ``BleakClientWinRT`` from scan response if advertising is None (WinRT).
* Changed type hint for ``adv`` attribute of ``bleak.backends.winrt.scanner._RawAdvData``.
* ``BleakGATTCharacteristic.max_write_without_response_size`` is now dynamic.

Fixed
-----
* Fixed ``discovered_devices_and_advertisement_data`` returning devices that should
  be filtered out by service UUIDs. Fixes #1576.
* Fixed a ``Descriptor None was not found!`` exception occurring in ``start_notify()`` on Android. Fixes #823.
* Fixed exception raised when  [... cut]
closed 2024-05-23 by dlech