hbldh/bleak #825: Could not get GATT services: Unreachable

akyupely, 2022-05-04:
* bleak version: 0.14.2
* Python version: 3.10
* Operating System: Windows 10
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Hello. I am trying to connect a specific ble device in a group of devices to my PC. I am able to scan and find my device, however, when I try to connect I get the error above. Also, the device name changes from the name associated with the address to a different device name with that same address.

### What I Did

I combined the scanning example with the service discovering example as a test to connect to a device to my PC.

```
async def bleConnect(device_addr):
    print(device_addr)
    device = await BleakScanner.find_device_by_address(device_addr, timeout=20.0)
    print (device)
    if not device:
        raise BleakError(f"A device with address {device_addr} could not be found.")

    async with BleakClient(device) as client: 
        print(client)
        svcs = await client.get_services()
        print("Services:")
        for service in svcs:
            print(service)
```

dlech, 2022-07-13:
Could you log Bluetooth packets using wireshark so we can see what the device is actually doing? Also #695  may help.

dlech, 2022-07-13:
duplicate of #740