hbldh/bleak #826: Please add an example of using Bleak from a non async function?

zapta, 2022-05-07:
Bleaks API is based on async functions which makes it difficult to use from non async function.   Please add an example where a non async main reads and print a value from a device.

In my case, the non async caller is a callback from a GUI framework which must be non async.

dhalbert, 2022-05-07:
I did this several years ago when encountering the same problem:
https://github.com/adafruit/Adafruit_Blinka_bleio/blob/a7d037244956a2d77e99250f76c11f9660d46386/_bleio/common.py#L136-L140. I am not sure it is the right way to do it these days.

zapta, 2022-05-07:
Thanks. I also got this https://stackoverflow.com/questions/72155012/how-to-call-an-async-python-function-from-a-non-async-function.

I think it worth having an explicit example with the recommended way to call Bleak from sync function.

dlech, 2022-10-03:
Calling async code from non-async is outside of the scope of bleak and not a use case we want to spend time supporting. If you don't want to use asyncio, you should probably consider using a different library, for example, https://github.com/OpenBluetoothToolbox/SimpleBLE looks promising.

closed 2022-10-03 by dlech

lidorshimoni, 2024-03-26:
> I did this several years ago when encountering the same problem: https://github.com/adafruit/Adafruit_Blinka_bleio/blob/a7d037244956a2d77e99250f76c11f9660d46386/_bleio/common.py#L136-L140. I am not sure it is the right way to do it these days.

This worked for me!