hbldh/bleak #827: programmatically accept pairing request in python without dialog interaction

mitea1, 2022-05-09:
* bleak version: 0.14.3
* Python version: 3.7.3
* Operating System: Win10 (and Linux)
* BlueZ version (`bluetoothctl -v`) in case of Linux: 5.50

### Description

Looking for a way to accept pairing request during integration tests without dialog action in windows. also want to understand what would be needed to do the same in a linux environment

### What I Did

used BleakClient.pair(). works if I wait and confirm pairing in dialog. but I'm looking for a way to do that by code.

hbldh, 2022-05-09:
In Windows, the pairing is already designed to be "Just Works" without any user interaction: 
https://github.com/hbldh/bleak/blob/develop/bleak/backends/winrt/client.py#L367-L368

A pairing request is sent and the method in the link above should just accept the pairing without any further processing needed.

If a dialog is brought up, then that is a change in how Windows handles pairing. I did not get a dialog when I paired the last time.

From the [Microsoft UWP docs concering the ConfirmOnly solution](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.devicepairingkinds?view=winrt-22000):

> The application must confirm they wish to perform the pairing action. You can present an optional confirmation dialog to the user. With a value of ConfirmOnly, call [Accept](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.devicepairingrequestedeventargs.accept?view=winrt-22000#windows-devices-enumeration-devicepairingrequestedeventargs-accept) from the event args of the [PairingRequested](https://docs.microsoft.com/en-us/uwp/api/windows.devices.enumeration.deviceinformationcustompairing.pairingrequested?view=winrt-22000) event handler if you want the pairing to complete.

For integration tests, just monkeypatch the `pair` function on the client to this

```python
async def mock_pair_success(client, protection_level: int = None, **kwargs) -> bool:
    return True

client.pair = mock_pair_success
```
You will never be able to test pairing without performing the actual pairing with Bleak, since it uses the OS native BLE APIs. You could possible monkeypatch the CustomPairing class from the `winrt` package, but I do not think you will gain anything by doing that...

mitea1, 2022-05-09:
> If a dialog is brought up, then that is a change in how Windows handles pairing. I did not get a dialog when I paired the last time.

using Windows version 10.0.19042 Build 19042. it's not latest and greatest

> You will never be able to test pairing without performing the actual pairing with Bleak, since it uses the OS native BLE APIs. You could possible monkeypatch the CustomPairing class from the winrt package, but I do not think you will gain anything by doing that...

I don't want to prevent pairing at all I just want to do it in a way that I don't need to process any user dialog. just processing it in code or automatically if possible.

should just works (like you describe for windows above) also work in a linux environment?

dlech, 2022-05-09:
This is not technically possible on macOS. For Windows and Linux, there is a pull request open for this, but it has been stalled. #640

sfoley-gpqa, 2022-05-19:
> This is not technically possible on macOS. For Windows and Linux, there is a pull request open for this, but it has been stalled. #640

This is a problem for my team's work in MacOS 12+ right now. :(

sfoley-gpqa, 2025-03-25:
> > This is not technically possible on macOS. For Windows and Linux, there is a pull request open for this, but it has been stalled. [#640](https://github.com/hbldh/bleak/pull/640)
> 
> This is a problem for my team's work in MacOS 12+ right now. :(

Update:
My automation team uses the Applescript below to automatically click on the "Connect" button in the Bluetooth connection dialog in macOS:

```osascript
#!/usr/bin/osascript
--   This script searches for the "Connect" button in the OSD that appears when connecting to a new Bluetooth device
--   and automagically clicks it

tell application "System Events"
    repeat
        try
            repeat with win in (windows of application process "BluetoothUIServer")
                repeat with btn in (buttons of win)
                    set btnName to name of btn
                    if btnName is "Connect" then
                        log "[click_connect.applescript] Connect button found. Clicking it now..."
                        click btn
                        return
                    end if
                end repeat
            end repeat
        on error errMsg number errNum
            -- Ignore this error:
            --     Error: System Events got an error: Can’t get application process "BluetoothUIServer". (Error number: -1728)
            if errNum is not -1728 then
                log "Error: " & errMsg & " (Error number: " & errNum & ")"
            end if
        end try
        delay 0.1
    end repeat
end tell
```