hbldh/bleak #828: Any way to start_notify if the UUID characteristic has  no Descriptor

ps2229, 2022-05-11:
* bleak version:0.14.3
* Python version:3.9
* Operating System:Android
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Trying to use start_notify function but getting errors like no descriptor, and after checking the bluetooth module there is no descriptor in the UUID characteristic , is there any way to still use start_notify without the descriptor.Using bluno bettle ble module

### What I Did
 On some forums I found solutions like using 'gatt.setCharacteristicNotification' but really confused as to call it where in the code.? Any idea how to go about it.

hbldh, 2022-05-11:
This is not really a Bleak issue. If `start_notify` delivers error that no descriptor is present, then it seems like you really need to add these in your peripheral to enable notifications.

I have no idea what is needed to be done on your a
Arduino, but Bleak cannot alleviate problems with incorrectly implemented peripherals.

NewAge12, 2022-05-11:
Having the same issue with the error using bluno bettle ble  --->> bleak.exc.BleakError: Descriptor None was not found!  
It does not contain the CLIENT_CHARACTERISTIC_CONFIGURATION_UUID='"00002902-0000-1000-8000-00805f9b34fb" , So the start_notify is giving errors, any way to bypass this and read the values using start_notify.??

NewAge12, 2022-05-16:
d

Egidio92, 2022-06-07:
I've the same problem too

NewAge12, 2022-06-08:
In the p4android folder change the client.py file to this and it will work.
# -*- coding: utf-8 -*-
"""
BLE Client for python-for-android
"""
import asyncio
import logging
import uuid
import warnings
from typing import Callable, Optional, Union

from bleak.backends.device import BLEDevice
from bleak.backends.service import BleakGATTServiceCollection
from bleak.exc import BleakError
from bleak.backends.client import BaseBleakClient
from bleak.backends.p4android.service import BleakGATTServiceP4Android
from bleak.backends.p4android.characteristic import BleakGATTCharacteristicP4Android
from bleak.backends.p4android.descriptor import BleakGATTDescriptorP4Android

from android.broadcast import BroadcastReceiver
from jnius import java_method

from . import defs
from . import utils

logger = logging.getLogger(__name__)

class BleakClientP4Android(BaseBleakClient):
    """A python-for-android Bleak Client

    Args:
        address_or_ble_device (`BLEDevice` or str): The Bluetooth address of the BLE peripheral to connect to or the `BLEDevice` object representing it.

    Keyword Args:
        disconnected_callback (callable): Callback that will be scheduled in the
            event loop when the client is disconnected. The callable must take one
            argument, which will be this client object.
        adapter (str): Bluetooth adapter to use for discovery. [unused]
    """

    def __init__(self, address_or_ble_device: Union[BLEDevice, str], **kwargs):
        super(BleakClientP4Android, self).__init__(address_or_ble_device, **kwargs)
        # kwarg "device" is for backwards compatibility
        self.__adapter = kwargs.get("adapter", kwargs.get("device", None))
        self.__gatt = None
        self.__mtu = None

    def __del__(self):
        if self.__gatt is not None:
            self.__gatt.close()
            self.__gatt = None

    # Connectivity methods

    async def connect(self, **kwargs) -> bool:
        """Connect to the specified GATT server.

        Returns:
            Boolean representing connection status.

        """
        loop = asyncio.get_event_loop()

        self.__adapter = defs.BluetoothAdapter.getDefaultAdapter()
        if self.__adapter is None:
            raise BleakError("Bluetooth is not supported on this hardware platform")
        if self.__adapter.getState() != defs.BluetoothAdapter.STATE_ON:
            raise BleakError("Bluetooth is not turned on")

        self.__device = self.__adapter.getRemoteDevice(self.address)
[... 498 lines cut ...]
        self.result_state(status, "onServicesDiscovered")

    @java_method("(I[B)V")
    def onCharacteristicChanged(self, handle, value):
        self._loop.call_soon_threadsafe(
            self._client._subscriptions[handle], handle, bytearray(value.tolist())
        )

    @java_method("(II[B)V")
    def onCharacteristicRead(self, handle, status, value):
        self.result_state(
            status, ("onCharacteristicRead", handle), bytes(value.tolist())
        )

    @java_method("(II)V")
    def onCharacteristicWrite(self, handle, status):
        self.result_state(status, ("onCharacteristicWrite", handle))

    @java_method("(Ljava/lang/String;I[B)V")
    def onDescriptorRead(self, uuid, status, value):
        self.result_state(status, ("onDescriptorRead", uuid), bytes(value.tolist()))

    @java_method("(Ljava/lang/String;I)V")
    def onDescriptorWrite(self, uuid, status):
        self.result_state(status, ("onDescriptorWrite", uuid))

Egidio92, 2022-06-08:
I'm sorry but it doesn't work, thanks anyway ... I'll explain my problem better. I have to read the weight from a Mi Smart Scale 2 scale via Mac. 

The weight is encoded in a BLE characteristic. However, sometimes it reads the weight, other times it doesn't. 

I'm using the code found in the Bleak repository examples and this is my code:

import sys
import asyncio
import platform

from bleak import BleakClient

CHARACTERISTIC_UUID = "00002a9d-0000-1000-8000-00805f9b34fb"
ADDRESS = (
    "70:87:9e:0f:f8:7d"
    if platform.system() != "Darwin"
    else "775A199B-A238-4AB7-86EB-4EFF8C404F9A"
)

def notification_handler(sender, data):
    """Simple notification handler which prints the data received."""
    data1 = list(data)
    data2 = ((data1[1]+data1[2]*256)*0.005)
    print("{0}: {1}".format(sender, data2))

async def main(address, char_uuid):
    async with BleakClient(address) as client:
        print(f"Connected: {client.is_connected}")

       
        await client.start_notify(char_uuid, notification_handler)
        await asyncio.sleep(30.0)
        await client.stop_notify(char_uuid)
        

if __name__ == "__main__":
    asyncio.run(
        main(
            sys.argv[1] if len(sys.argv) > 1 else ADDRESS,
            sys.argv[2] if len(sys.argv) > 2 else CHARACTERISTIC_UUID,
        )
    )

dlech, 2022-06-08:
> have to read the weight from a Mi Smart Scale 2 scale via Mac.

This issue is about Android. If you have a problem on macOS, please start a new issue.

dlech, 2022-07-13:
Possibly related to #883.

dlech, 2024-05-23:
Going to call this one a duplicate of #823 and hope it is fixed now.

closed 2024-05-23 by dlech