hbldh/bleak #830: Any way of using bleak in a non blocking way?

NewAge12, 2022-05-16:
* bleak version:0.14.3
* Python version:3.9
* Operating System:Android
* BlueZ version (`bluetoothctl -v`) in case of Linux:

### Description

Bleak commands are working perfectly but while it is in process it blocks the whole app, any way to use bleak in a non blocking way .Tried threading at first but didnt work . any other way it can be done?

NewAge12, 2022-05-17:
All right so basically what is happening is when the bluetooth is scanning I wanted to show a gif during that time but while doing that the whole program gets stopped including the gif and since I was doing in android cant figure out how to use threading to make things simpler. Asyncio doesnt do tasks in parallel so it does not solve my problem dont really want a gif after the scan is done.

dlech, 2022-05-17:
> Asyncio doesnt do tasks in parallel

On the contrary, this is exactly what asynio is designed to do.

Can you share your code?

NewAge12, 2022-05-18:
import asyncio
import sqlite3
from datetime import date, datetime
from kivy.clock import Clock
from kivy.uix.image import Image
from kivy.uix.popup import Popup
from kivymd.uix.floatlayout import MDFloatLayout
from kivymd.uix.label import MDLabel

import bleak
from kivymd.toast import toast

now = datetime.now()
from kivy.uix.screenmanager import Screen

from kivymd_extensions.sweetalert import SweetAlert

class TestScreen(Screen):
    aa=''
    def on_enter(self, *args):
        self.ids.b1.disabled = False

   
   async  def scanPOP(self):
        box = MDFloatLayout()
        i = Image(source='2.gif',size_hint=(.8, .8), pos_hint={'x': 0, 'y': 0.1})

        box.add_widget(i)

        self.main_pop = Popup(title=' ', content=box,separator_height=0,
                              size_hint=(.6, .7), auto_dismiss=False, title_size=30,background_color=(0,0,0,0))

        self.main_pop.open()

    def scan1(self):
         loop1 = asyncio.get_event_loop()
        loop1.run_until_complete(self.scanPOP())
        loop = asyncio.get_event_loop()
        loop.run_until_complete(self.begin_scan1())
   
    async def begin_scan1(self):
        def handler(sender, data):
            print(data)
            self.val=data
        hr_uuid = "0000dfb1-0000-1000-8000-00805f9b34fb"
        ADDRESS = "C4:BE:84:20:19:3F"
        a = []
        self.ids.p_fat.text=' '
        scanned_devices = await bleak.BleakScanner.discover(1)
         
        if len(scanned_devices) == 0:
            print("NO DEVICES")
            self.main_pop.dismiss()
        scanned_devices.sort(key=lambda device: -device.rssi)
        for device in scanned_devices:
            device1 = f"{device.name}"
            print(device1)
            
            if device1 == "Bluno":
                try:           
                    async with bleak.BleakClient(device) as client:    
                        uu = '1'
                        write_value = bytearray([int(uu)])
                        await client.write_gatt_char(hr_uuid, write_value)
                        await asyncio.sleep(3.0)
                
                        await client.start_notify(hr_uuid, handler)
                        await asyncio.sleep(10.0)
                        await client.stop_notify(hr_uuid)
                         
                        
                except:
                    await asyncio.sleep(3.0)
                    toast('Error 101')
            else:
                toast('No device',duration=1)
                self.main_pop.dismiss()
   
       
The scan1() is called on the press of a button. What happens is the scan is done and after that the popup is called which is wrong. Cant really figure out a way to do both these things simultaneously. Is there anything wrong in the code??

dlech, 2022-05-18:
To start, there should be one top-level `asyncio.run()` to run the entire app as in https://github.com/hbldh/bleak/blob/develop/examples/kivy/main.py. There should be no other calls that start a new event loop. Instead, there are functions like `asyncio.create_task()` that can start running an async coroutine from a non-async function.

Since the popup code is not awaitabile, it doesn't need to be in an `async` function.

Maybe something like this?

```python
import asyncio
from kivy.uix.image import Image
from kivy.uix.popup import Popup
from kivymd.uix.floatlayout import MDFloatLayout
from kivymd.uix.label import MDLabel
import bleak
from kivy.uix.screenmanager import Screen

class TestScreen(Screen):
    aa=''
    def on_enter(self, *args):
        self.ids.b1.disabled = False

    def create_popup(self):
        box = MDFloatLayout()
        i = Image(source='2.gif',size_hint=(.8, .8), pos_hint={'x': 0, 'y': 0.1})

        box.add_widget(i)

        popup = Popup(title=' ', content=box,separator_height=0,
                              size_hint=(.6, .7), auto_dismiss=False, title_size=30,background_color=(0,0,0,0))

        popup.open()

        return popup

    def on_button_press(self):
        asyncio.create_task(self.scan())
   
    async def scan(self):
        def handler(sender, data):
            print(data)
            self.val=data

        hr_uuid = "0000dfb1-0000-1000-8000-00805f9b34fb"
        ADDRESS = "C4:BE:84:20:19:3F"
        a = []
        self.ids.p_fat.text=' '

        popup = self.create_popup()

        try:
            scanned_devices = await bleak.BleakScanner.discover(1)
         
            if len(scanned_devices) == 0:
                print("NO DEVICES")
                self.main_pop.dismiss()
            scanned_devices.sort(key=lambda device: -device.rssi)
            for device in scanned_devices:
                device1 = f"{device.name}"
                print(device1)
            
                if device1 == "Bluno":
                    try:           
                        async with bleak.BleakClient(device) as client:    
                            uu = '1'
                            write_value = bytearray([int(uu)])
                            await client.write_gatt_char(hr_uuid, write_value)
                            await asyncio.sleep(3.0)
                
                            await client.start_notify(hr_uuid, handler)
                            await asyncio.sleep(10.0)
                            await client.stop_notify(hr_uuid)
                         
                        
                    except:
                        await asyncio.sleep(3.0)
                        toast('Error 101')
                else:
                    toast('No device',duration=1)
        finally:
            popup.dismiss()

class ExampleApp(App):
    ...

async def main():
    app = ExampleApp()
    await app.async_run("asyncio")

if __name__ == "__main__":
    asyncio.run(main())
```